Koenkk/zigbee2mqtt #19573: Sensors for certain Zigbee devices are not created after the 2023.11.1 update

Murianwind, 2023-11-05:
### What happened?

After the 2023.11.1 update, sensors for certain Zigbee devices are not created. In Z2M, the sensor operates normally, but in HA, the sensor is not created and does not operate.

[image]

[image]

When I registered an issue with HA's Core, I received the response below.

Needs to be fixed in z2m. Sensor entities are no longer allowed to be set as config category. https://github.com/home-assistant/core/pull/101471

### Zigbee2MQTT version

1.33.2-1

### Adapter firmware version

20230507

### Adapter

SMARTLIGHT CC2652P (SLZB-02)

### Debug log

Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:279
Integration: MQTT (documentation, issues)
First occurred: 05:51:40 (1 occurrences)
Last logged: 05:51:40

Error 'Entity category `config` is invalid' when processing MQTT discovery message topic: 'homeassistant/binary_sensor/0x8cf681fffe3e2688/card/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/작업 기록/availability'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0x8cf681fffe3e2688'], 'manufacturer': 'Dawon DNS', 'model': 'IOT Card holder (KB-HD100-ZB)', 'name': '작업 기록'}, 'entity_category': 'config', 'icon': 'mdi:clipboard-check', 'json_attributes_topic': 'zigbee2mqtt/작업 기록', 'name': 'Card', 'object_id': '작업_기록_card', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.33.2', 'support_url': 'https://www.zigbee2mqtt.io'}, 'payload_off': False, 'payload_on': True, 'state_topic': 'zigbee2mqtt/작업 기록', 'unique_id': '0x8cf681fffe3e2688_card_zigbee2mqtt', 'value_template': '{{ value_json.card }}', 'platform': 'mqtt'}'

KennethLavrsen, 2023-11-05:
I see this too for Aqara FP1 sensors

Error 'Entity category `config` is invalid' when processing MQTT discovery message topic

roastcom, 2023-11-09:
I second that multiple times. 

I have 2 devices (different 1-gang wall switches) that 'lost' on/off in HA after I updated the docker image last week. 
Many others while didn't completely lose anything changed sensor names to some generic stuff like ''MQTT Sensor".

For example here is temperature and humidity sensor TS0601 _TZE200_vvmbj46n :
HA:
[image]

MQTT
[image]

Or here is one that 'lost' on/off (TS0001 _TZ3000_ckorfokt):
HA:
[image]

MQTT:
[image]

jorioux, 2023-11-10:
I have the same issue.

* Zigbee2mqtt version [1.33.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.0) commit: [7ee207f](https://github.com/Koenkk/zigbee2mqtt/commit/7ee207f)
* HA version 2023.11.2

> 2023-11-10 13:02:26.553 ERROR (MainThread) [homeassistant.components.mqtt.mixins] Error 'Entity category `config` is invalid' when processing MQTT discovery message topic: 'homeassistant/sensor/0x003c84fffefac315/power_on_behavior/config', message: '{'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'device': {'identifiers': ['zigbee2mqtt_0x003c84fffefac315'], 'manufacturer': 'IKEA', 'model': 'TRADFRI bulb E26/E27 CWS 800/806 lumen, dimmable, color, opal white (LED1924G9)', 'name': 'Piano Light', 'sw_version': '1.0.021'}, 'enabled_by_default': False, 'entity_category': 'config', 'icon': 'mdi:power-settings', 'name': 'Power-on behavior', 'object_id': 'piano_light_power_on_behavior', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.33.0', 'support_url': 'https://www.zigbee2mqtt.io'}, 'state_topic': 'zigbee2mqtt/0x003c84fffefac315', 'unique_id': '0x003c84fffefac315_power_on_behavior_zigbee2mqtt', 'value_template': '{{ value_json.power_on_behavior }}', 'platform': 'mqtt'}'

nagydavid, 2023-11-11:
Same issue here: with Tradfri and Heimgard after update.

schumi2004, 2023-11-14:
Having the same issue but only in zigbee2mqtt Edge version: 1.30.4-dev commit: [a404f9f]

Multiple error messages like these:

2023-11-14 23:01:24.227 ERROR (MainThread) [homeassistant.components.mqtt.mixins] Error 'Entity category `config` is invalid' when processing MQTT discovery message topic: 'homeassistant/sensor/0xe0798dfffec792e0/power_outage_memory/config', message: '{'availability': [{'topic': 'zigbee2mqtt-edge/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt-edge/Vaatwasser/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0xe0798dfffec792e0'], 'manufacturer': 'TuYa', 'model': 'Smart plug (with power monitoring by polling) (TS011F_plug_3)', 'name': 'Vaatwasser'}, 'enabled_by_default': False, 'entity_category': 'config', 'icon': 'mdi:power-settings', 'name': 'Vaatwasser power outage memory', 'state_topic': 'zigbee2mqtt-edge/Vaatwasser', 'unique_id': '0xe0798dfffec792e0_power_outage_memory_zigbee2mqtt-edge', 'value_template': '{{ value_json.power_outage_memory }}', 'platform': 'mqtt'}'

I will see if there is an update and report back.

roastcom, 2023-11-15:
It seems that an update to the HA (the latest docker image) has fixed the issues I was having with weird sensor names and disappearing on/off states.

closed 2024-06-13 by github-actions[bot]