Koenkk/zigbee2mqtt #19576: Network error when launching the zigbee suite

Kirjy, 2023-11-05:
### What happened?

Hi,
I have the same problem as specified in there : https://github.com/Koenkk/zigbee2mqtt/issues/18090
When I'm starting the docker, I get the following error. 
It used to work correctly on my raspberry pi 4 under raspberry pi OS 64 bits. But now I'm trying to migrate to a mini pc with an Intel N100 running on Ubuntu server. It seems that the problem in the first post was also on this mini pc. That's why I'm specifying it. 

```
Attaching to z2m-ha, z2m-mosquitto, z2m-zigbee2mqtt
z2m-mosquitto    | chown: /mosquitto/config/mosquitto.conf: Operation not permittedz2m-mosquitto    | 1699186615: mosquitto version 2.0.18 starting
z2m-mosquitto    | 1699186615: Config loaded from /mosquitto/config/mosquitto.conf.z2m-mosquitto    | 1699186615: Opening ipv4 listen socket on port 1883.
z2m-mosquitto    | 1699186615: Opening ipv6 listen socket on port 1883.
z2m-mosquitto    | 1699186615: Opening websockets listen socket on port 9001.
z2m-mosquitto    | 1699186615: mosquitto version 2.0.18 running
z2m-ha           | s6-rc: info: service s6rc-oneshot-runner: starting
z2m-ha           | s6-rc: info: service s6rc-oneshot-runner successfully started
z2m-ha           | s6-rc: info: service fix-attrs: starting
z2m-ha           | s6-rc: info: service fix-attrs successfully started
z2m-ha           | s6-rc: info: service legacy-cont-init: starting
z2m-ha           | s6-rc: info: service legacy-cont-init successfully started
z2m-ha           | s6-rc: info: service legacy-services: starting
z2m-ha           | services-up: info: copying legacy longrun home-assistant (no readiness notification)
z2m-zigbee2mqtt  | Using '/app/data' as data directory
z2m-ha           | s6-rc: info: service legacy-services successfully started
z2m-zigbee2mqtt  | Zigbee2MQTT:info  2023-11-05 13:16:56: Logging to console and directory: '/app/data/log/2023-11-05.13-16-56' filename: log.txt
z2m-zigbee2mqtt  | Zigbee2MQTT:info  2023-11-05 13:16:56: Starting Zigbee2MQTT version 1.33.2 (commit #9996c93)
z2m-zigbee2mqtt  | Zigbee2MQTT:info  2023-11-05 13:16:56: Starting zigbee-herdsman (0.21.0)
z2m-zigbee2mqtt  | Zigbee2MQTT:info  2023-11-05 13:16:59: zigbee-herdsman started (resumed)
z2m-zigbee2mqtt  | Zigbee2MQTT:info  2023-11-05 13:16:59: Coordinator firmware version: 'undefined'
z2m-zigbee2mqtt  | Zigbee2MQTT:error 2023-11-05 13:17:09: Failed to start zigbee
z2m-zigbee2mqtt  | Zigbee2MQTT:error 2023-11-05 13:17:09: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
z2m-zigbee2mqtt  | Zigbee2MQTT:error 2023-11-05 13:17:09: Exiting...
z2m-zigbee2mqtt  | Zigbee2MQTT:error 2023-11-05 13:17:09: Error: get network parameters Error:TIMEOUT
z2m-zigbee2mqtt  |     at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1002:35)
z2m-zigbee2mqtt  |     at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:360:44)
z2m-zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:113:62)
z2m-zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:101:27)
z2m-zigbee2mqtt  |     at start (/app/index.js:107:5)
```

### How to reproduce it (minimal and precise)

Launching my docker compose. It can be found here : https://github.com/Kirjy/DockerizedRaspberryPi/tree/main/z2m

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

deCONZ_ConBeeII_0x26780700

### Adapter

Conbee 2

YewTreeWeb, 2023-11-23:
I'm also having this issue on a mini PC running a Intel N100 CPU. Z2M also ran perfectly fine on my Raspberry Pi 4. However I'm running a Proxmox VM. My error message is:
```
[20:59:39] INFO: Preparing to start...
[20:59:39] INFO: Socat not enabled
[20:59:39] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2023-11-23 21:00:11: Failed to start zigbee
Zigbee2MQTT:error 2023-11-23 21:00:11: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-11-23 21:00:11: Exiting...
Zigbee2MQTT:error 2023-11-23 21:00:11: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1002:35)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:360:44)
    at Zigbee.start (/app/lib/zigbee.ts:113:62)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

Kirjy, 2023-11-23:
As I couldn't make it work, I tried a few things : 
1 - running deconz instead of zigbee2mqtt works
2 - running zha also works
3 - running zigbee2mqtt with a sonoff key works too
4 - running it on a pi 4 works with zigbee2mqtt and a conbee 2 key

That makes the reason even more weird to me.

anthosz, 2023-11-25:
Same issue here:
- Latest version of zigbee2mqtt docker image
- Key well found via lsusb: Bus 001 Device 006: ID 1cf1:0030 Dresden Elektronik ZigBee gateway [ConBee II]
- On my previous mini pc, all was ok (Debian + Docker) -> switched to N100 (like @YewTreeWeb ) + Ubuntu + Docker *> didn't works anymore

anthosz, 2023-11-26:
Small update:
- after few restart of the docker container,  "Coordinator firmware version" is well found but "Error: get network parameters Error:TIMEOUT" continue.
- I kept the docker compose up all the night (so basically, auto restart of container once stopped) -> issue was "fixed" during the night and z2mqtt was finally up... but with several "empty" interval during the night (like 1h without any state).
After a restart of the container, the issue remain.

anthosz, 2023-12-04:
Finally it worked during 1 week (until now)... I tried to add a new device & boom, error is back since 1h

x3ntr0, 2023-12-15:
The solution is trivial :) Service is already running, stop it and start it again.
VM: zigbee2mqtt 
$: sudo systemctl stop zigbee2mqtt
$: npm start

Kirjy, 2023-12-15:
> The solution is trivial :) Service is already running, stop it and start it again. VM: zigbee2mqtt $: sudo systemctl stop zigbee2mqtt $: npm start

Just like in another thread : sounds good, doesn't work.

anthosz, 2023-12-15:
> The solution is trivial :) Service is already running, stop it and start it again. VM: zigbee2mqtt $: sudo systemctl stop zigbee2mqtt $: npm start

It's not the same error. In my case, npm is not running at all so docker restart...

anthosz, 2023-12-15:
In addition, like mentioned 2 weeks ago.

All worked well after few hours of restart, z2mqtt was stable during few days.
Tried to allow pairing:
```
Zigbee2MQTT:error 2023-12-15 21:13:33: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'APS TIMEOUT'
Zigbee2MQTT:debug 2023-12-15 21:13:34: Error: APS TIMEOUT
    at Driver.processApsBusyQueue (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:723:28)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:162:56)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
```

10 sec later:
```
UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "TIMEOUT".
```

10 sec later, it's back:
```
Zigbee2MQTT:debug 2023-12-15 21:13:55: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2023-12-15 21:13:55: Logging to console and directory: '/app/data/log/2023-12-15.21-13-55' filename: log.txt
Zigbee2MQTT:debug 2023-12-15 21:13:55: Removing old log directory '/app/data/log/2023-12-09.10-51-26'
Zigbee2MQTT:info  2023-12-15 21:13:55: Starting Zigbee2MQTT version 1.34.0 (commit #aae7312)
Zigbee2MQTT:info  2023-12-15 21:13:55: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-15 21:13:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2023-12-15 21:13:58: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-12-15 21:13:58: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":120,"product":0,"revision":"0x26780700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:error 2023-12-15 21:14:08: Failed to start zigbee
Zigbee2MQTT:error 2023-12-15 21:14:08: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-12-15 21:14:08: Exiting...
Zigbee2MQTT:error 2023-12-15 21:14:08: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1017:35)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:360:44)
    at Zigbee.start (/app/lib/zigbee.ts:113:62)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

Now, just need to wait few hours and z2mqtt will be back.. But no way to pair new devices :(

anthosz, 2023-12-22:
Tried also on different USB port (with or without cable): same issue
Tried with a DB reset (reset the z2mqtt conf): same issue
Tried to check on bios what goes wrong: no idea (only cpu stuff can be modified)
Do you have any idea @Koenkk ? It seems we are several peoples impacted and majority switch to another solution :/

I'm available for any test/debug but at this time, I'm out of ideas

Koenkk, 2023-12-23:
Looks like some bug in the Deconz integration, @chrishae could you take a look at this?

anthosz, 2023-12-23:
Workaround at this time: switched zigbee2mqtt to raspberry (docker) and all works in addition of the N100 mini computer (HA/mosquitto)...

thomaszub, 2023-12-23:
Hi,

I have the same issue with an Intel N100 and ConBee2 (works fine on RPi 4). Here are debug logs with zigbee-herdsman included:

```
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Zigbee2MQTT:debug 2023-12-23 14:04:55: Loaded state from file /app/data/state.json
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-23 14:04:55: Logging to console and directory: '/app/data/log/2023-12-23.14-04-55' filename: log.txt
zigbee2mqtt  | Zigbee2MQTT:debug 2023-12-23 14:04:55: Removing old log directory '/app/data/log/2023-12-23.13-59-51'
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-23 14:04:55: Starting Zigbee2MQTT version 1.34.0 (commit #aae7312)
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-23 14:04:55: Starting zigbee-herdsman (0.25.0)
zigbee2mqtt  | Zigbee2MQTT:debug 2023-12-23 14:04:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
zigbee2mqtt  | 2023-12-23T13:04:55.628Z zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'DeconzAdapter'
zigbee2mqtt  | 2023-12-23T13:04:55.629Z zigbee-herdsman:deconz:driver Set delay to 0
zigbee2mqtt  | 2023-12-23T13:04:55.630Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
zigbee2mqtt  | 2023-12-23T13:04:55.630Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
zigbee2mqtt  | 2023-12-23T13:04:55.633Z zigbee-herdsman:deconz:driver Serialport opened
zigbee2mqtt  | 2023-12-23T13:04:55.633Z zigbee-herdsman:controller:log Started with result 'resumed'
zigbee2mqtt  | 2023-12-23T13:04:55.633Z zigbee-herdsman:controller:log Injected database: true, adapter: true
zigbee2mqtt  | 2023-12-23T13:04:55.634Z zigbee-herdsman:controller:device:log Request Queue (0x00212effff06baf0): default expiration timeout set to 0
zigbee2mqtt  | 2023-12-23T13:04:55.635Z zigbee-herdsman:controller:database:log Writing database to '/app/data/database.db'
zigbee2mqtt  | 2023-12-23T13:04:55.637Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
zigbee2mqtt  | 2023-12-23T13:04:55.657Z zigbee-herdsman:deconz:frameParser MAC: 0x00212effff06baf0
zigbee2mqtt  | 2023-12-23T13:04:55.657Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 2 paramId: 7
zigbee2mqtt  | 2023-12-23T13:04:58.630Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
zigbee2mqtt  | 2023-12-23T13:04:58.634Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 3 request id: 1
zigbee2mqtt  | 2023-12-23T13:04:58.636Z zigbee-herdsman:deconz:frameParser NWKADDR: 0
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-23 14:04:58: zigbee-herdsman started (resumed)
zigbee2mqtt  | 2023-12-23T13:04:58.639Z zigbee-herdsman:deconz:driver send read firmware version request from queue. seqNr: 4
zigbee2mqtt  | 2023-12-23T13:04:58.642Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 1 status: 6
zigbee2mqtt  | 2023-12-23T13:04:58.642Z zigbee-herdsman:deconz:driver networkstate: 0 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 0
zigbee2mqtt  | 2023-12-23T13:05:08.640Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xd seqNr: 4
zigbee2mqtt  | 2023-12-23T13:05:08.641Z zigbee-herdsman:deconz:adapter Get coordinator version Error: TIMEOUT
zigbee2mqtt  | Zigbee2MQTT:info  2023-12-23 14:05:08: Coordinator firmware version: 'undefined'
zigbee2mqtt  | 2023-12-23T13:05:08.647Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 5
zigbee2mqtt  | 2023-12-23T13:05:08.649Z zigbee-herdsman:deconz:frameParser read firmware version response - version: 0,7,102,38
zigbee2mqtt  | 2023-12-23T13:05:18.651Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 5
zigbee2mqtt  | 2023-12-23T13:05:18.653Z zigbee-herdsman:deconz:driver too many timeouts - restart serial connecion
zigbee2mqtt  | 2023-12-23T13:05:18.654Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
zigbee2mqtt  | 2023-12-23T13:05:18.656Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 6 paramId: 5
zigbee2mqtt  | 2023-12-23T13:05:18.659Z zigbee-herdsman:deconz:driver Serialport opened
zigbee2mqtt  | 2023-12-23T13:05:18.661Z zigbee-herdsman:deconz:frameParser PANID: 1a62
zigbee2mqtt  | 2023-12-23T13:05:18.662Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 7
zigbee2mqtt  | 2023-12-23T13:05:18.680Z zigbee-herdsman:deconz:frameParser device state: 10000000
zigbee2mqtt  | 2023-12-23T13:05:18.680Z zigbee-herdsman:deconz:driver networkstate: 0 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 0
zigbee2mqtt  | 2023-12-23T13:05:18.681Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 8 paramId: 11
[... cut ...]
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
zigbee2mqtt  |     at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:360:44)
zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:113:62)
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:101:27)
zigbee2mqtt  |     at start (/app/index.js:107:5)
zigbee2mqtt  | 2023-12-23T13:05:28.741Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 9
```

thomaszub, 2023-12-23:
Hi,

same issue here with an Intel N100 and ConBee 2 (works in RPi 4). The issue

`Zigbee2MQTT:info  2023-11-05 13:16:59: Coordinator firmware version: 'undefined'`

vanished for me with updating the ConBee 2 firmware to the latest version, but the Timeout remains. If you need more logs or some testing, you can just ping me.

ChrisHae, 2024-01-02:
Does the problem only occur in Docker or Proxmox? Does deCONZ also run in Docker when it works and does it run on a Raspberry Pi natively or also in Docker?

Can anybody share a log of `sudo dmesg`?

Docker has a problem with reconnecting USB devices that causes them to not be passed through correctly.

Haerteleric, 2024-01-02:
like [ChrisHae](https://github.com/ChrisHae) said we need a gonna need a dmesg log of the Host, that handles the USB-Connections to pin down the issue.

I also discovered in your docker-compose, that the tty-handle of the ConBee is passed through 2 different Containers:

https://github.com/Kirjy/DockerizedRaspberryPi/blob/main/z2m/compose.yml :18 and :57 afaik this could cause issues, when both containers try to interact with the Device.

Kirjy, 2024-01-02:
Hello.
Thanks for your answer. As I migrated to sonoff I don't have logs anymore. Can someone who has the same problem provide them?
I have indeed 2 containers linking the key but I also tried with only the good one. The second is when I forgot to remove this link when trying to integrate it through zha directly without using z2mqtt.

thomaszub, 2024-01-06:
Happy new year :)

I get the following dmesg logs (running zigbee2mqtt in a container):
```
[  154.126479] usb 1-1: new full-speed USB device number 4 using xhci_hcd
[  154.268553] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  154.268571] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  154.268578] usb 1-1: Product: ConBee II
[  154.268582] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  154.268586] usb 1-1: SerialNumber: DE2410227
[  154.285997] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  154.286036] usbcore: registered new interface driver cdc_acm
[  154.286039] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  157.762328] usb 1-1: USB disconnect, device number 4
[  158.129750] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[  158.271783] usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  158.271802] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  158.271808] usb 1-1: Product: ConBee II
[  158.271812] usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  158.271816] usb 1-1: SerialNumber: DE2410227
[  158.276989] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  223.043574] br-578da946cd5e: port 1(vethb19dde4) entered blocking state
[  223.043581] br-578da946cd5e: port 1(vethb19dde4) entered disabled state
[  223.043591] vethb19dde4: entered allmulticast mode
[  223.043930] vethb19dde4: entered promiscuous mode
[  223.045173] br-578da946cd5e: port 1(vethb19dde4) entered blocking state
[  223.045182] br-578da946cd5e: port 1(vethb19dde4) entered forwarding state
[  223.046369] br-578da946cd5e: port 1(vethb19dde4) entered disabled state
[  223.240202] eth0: renamed from vethf615533
[  223.260656] br-578da946cd5e: port 1(vethb19dde4) entered blocking state
[  223.260673] br-578da946cd5e: port 1(vethb19dde4) entered forwarding state
[  223.311773] br-578da946cd5e: port 2(vethb9a8449) entered blocking state
[  223.311791] br-578da946cd5e: port 2(vethb9a8449) entered disabled state
[  223.311805] vethb9a8449: entered allmulticast mode
[  223.311998] vethb9a8449: entered promiscuous mode
[  223.312129] br-578da946cd5e: port 2(vethb9a8449) entered blocking state
[  223.312134] br-578da946cd5e: port 2(vethb9a8449) entered forwarding state
[  223.333370] br-578da946cd5e: port 3(vethabd9ebe) entered blocking state
[  223.333715] br-578da946cd5e: port 3(vethabd9ebe) entered disabled state
[  223.333737] vethabd9ebe: entered allmulticast mode
[  223.333815] vethabd9ebe: entered promiscuous mode
[  223.333969] br-578da946cd5e: port 3(vethabd9ebe) entered blocking state
[  223.333974] br-578da946cd5e: port 3(vethabd9ebe) entered forwarding state
```

infamousdimsum, 2024-01-08:
Same error as YewTreeWeb. Network timeout.  Fresh install of HA ,restored backup from my Pi4 onto my new Kamrui N95 Mini PC. 

Noticed when I did this write access test for the Mini PC,  -w /dev/ttyACM0 && echo success || echo failure  . It got the failure message.  When I entered the same command on the Pi4, it said success.  

Also tried these two methods, but HAOS with the advanced terminal addon, doesn't recognize these commands.  ZHA and deconz also works, I did uninstall the 2 before attempting Z2M. I'll stick with the Pi4 for a bit longer.

Method 1: Give your user permissions on every reboot.

You can create a 'udev' rule to give your user permissions after every reboot:

    udevadm info -a -n /dev/ttyACM0 | grep 'serial' get the serial to your device YOURSERIAL

    Create the rule file with: sudo nano /etc/udev/rules.d/99-usb-serial.rules

    add this line: SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", ATTRS{serial}=="YOURSERIAL", SYMLINK="ttyUSB.CC2531-01", OWNER="pi"

    modify your Zigbee2MQTT config to adjust new SYMLINK name: nano /opt/zigbee2mqtt/data/configuration.yaml

… serial: port: /dev/ttyUSB.CC2531-01 …

After reboot your device will have the right permissions and always the same name.
Method 2: Add your user to specific groups

As mentioned on https://github.com/esp8266/source-code-examples/issues/26 , depending on your linux installation, various groups could have ownership of the device.

Add your user to the uucp , tty , dialout groups:

anthosz, 2024-05-30:
Hello,

Do we have any workaround excluding rpi for this issue?

Kirjy, 2024-05-31:
> Hello,
> 
> Do we have any workaround excluding rpi for this issue?

Hello,
My workaround was to buy a sonoff dongle. I've got no problem with this one on a N100 computer.
I couldn't figure how to make the conbee 2 work.

Grigorek, 2024-07-24:
Hi,
same issue here with proxmox and HassOS installation on zigbee2mqtt 1.39 installed via addon store.
I have installed earlier on my QNAP Nas on docker and it was working fine and now that I wanted move it to mini pc with Intel N100 I am facing this issue. It can recognize firmware but it get Timeout anyway and on another attempt it gets firmware undefined and there was also one attempt when if succesfully connected to Conbee but couldn't connect to any zigbee device.

`[2024-07-24 13:01:06] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-24 13:01:06] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #unknown)
[2024-07-24 13:01:06] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-24 13:01:06] info: 	z2m: zigbee-herdsman started (resumed)
[2024-07-24 13:01:09] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":120,"product":0,"revision":"0x26780700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[2024-07-24 13:01:19] error: 	z2m: Failed to start zigbee
[2024-07-24 13:01:19] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-24 13:01:19] error: 	z2m: Exiting...
[2024-07-24 13:01:19] error: 	z2m: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1015:35)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:379:44)
    at Zigbee.start (/app/lib/zigbee.ts:125:62)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)
[13:08:47] INFO: Preparing to start...
[13:08:47] INFO: Socat not enabled
[13:08:47] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-07-24 13:08:49] info: 	z2m: Logging to console, file (filename: log.log)
[2024-07-24 13:08:49] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #unknown)
[2024-07-24 13:08:49] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-07-24 13:08:52] info: 	z2m: zigbee-herdsman started (resumed)
[2024-07-24 13:08:52] info: 	z2m: Coordinator firmware version: 'undefined'
[2024-07-24 13:09:02] error: 	z2m: Failed to start zigbee
[2024-07-24 13:09:02] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-07-24 13:09:02] error: 	z2m: Exiting...
[2024-07-24 13:09:02] error: 	z2m: Error: get network parameters Error:TIMEOUT
    at DeconzAdapter.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1015:35)
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
    at Controller.getNetworkParameters (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:379:44)
    at Zigbee.start (/app/lib/zigbee.ts:125:62)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)`

I can provide logs etc.

anthosz, 2025-01-03:
Up

anthosz, 2025-05-07:
Up

closed 2026-01-14 by github-actions[bot]