Koenkk/zigbee2mqtt #19579: Gledopto GL-C-103P is not working proberly

Skixo01, 2023-11-05:
### What happened?

i am trying to set up my first network of devices (all Lights and Switches) on a raspberry pi 4 using home assistant an a sonoff usb 3.0 coordinator and have been very succesfull so far.
In my bedroom and living room i bought 2 Gledopto GL-C-103P controllers that should controll BFT-Lighting  LED Strips.

The one in the livingroom cant find any network route wich is strange cause there are 3 Spotlights from Paulmann right next to it wich act as routers in the network.

The one in the bedroom connects to the network and the colours can be controlled. It also can be bound to other devices. But im facing a problem with turning of effects. 
It is always blinking or breathing and never just a coloured light.

I am still very unexperienced and found most of the automations and stuff like that from github or youtube. 

### What did you expect to happen?

Both Controllers to connect properly and be able to controll them with zigbee2mqtt. 

### How to reproduce it (minimal and precise)

first one: try connect the Controller and it will pop up on the devices list. there will be a error pop up when trying to do anything, for example turn on or off, that says no network route.

second one: turn on the LED Strip using the Details page on the Device. try using stop_effects in this page but it changes nothing. Cant even swith the effects on this page. It only changes by pressing the modes button on the controller. 

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB 

### Debug log

Publish 'set' 'state' to 'Living_Room_Light_Strip' failed: 'Error: Command 0xa4c13821e9a7b9b3/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

Skixo01, 2023-11-05:
[image]
[image]

those are my devices and the map if anything of that is important :) 
its Bedroom_Light_Strip and Living_Room_Light_Strip thats giving me Problem

closed 2023-11-09 by Skixo01

vukitoso, 2023-11-18:
@Skixo01, Did you manage to solve the problem? If so, how?

Skixo01, 2023-11-19:
Sadly, no.Bought a new controller and a new LED Strip and used them. Some problem with those Gledopto ones. Every time i use one of them it only makes problems. Freundliche GrüßeJohnny Moore

10der, 2024-11-30:
the same issues here. it doesn't work at all.
more than now i can't turn on-off the device :(