Koenkk/zigbee2mqtt #19580: Lixee ZiPulses not report values

rty3108, 2023-11-05:
### What happened?

When the device receives a pulse, it sends the following message to Z2M:

Nov 05 15:40:57 myHub npm[24814]: Zigbee2MQTT:debug 2023-11-05 15:40:57: Received Zigbee message from '0x00158d0005d2a052', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,26]}' from endpoint 1 with groupID null

and does not report the number of steps

Any ideas?

### Zigbee2MQTT version

1.33.3

### Adapter firmware version

0x26730700

### Adapter

ConBee2/RaspBee2

rty3108, 2023-11-06:
Thanks for the reply.
Indeed it does not work because not all the configuration values ​​had been assigned.

In the release 1.33.2 of the production branch it doesn't work.
In the current developer branch the problem has already been solved by setting the UnitOfMeasure value.

Let's hope the new release is released soon.

rty3108, 2023-11-07:
Thanks, I can wait!

florianbelhomme, 2023-11-20:
You can find the unitOfMesure here : https://github.com/fairecasoimeme/ZiPulses#tableau-dunit%C3%A9-de-mesure

But as you say, HA just have "a number" without the unit of mesure.
A basic way to fix this is to add a sensor through the template, something like that :
```template:
  - sensor:
      - name: "Water consumption"
        unit_of_measurement: "m³"
        state: >
          {{ states('sensor.zipulses_agua_general_energy') }}
```

Then you will be able to add it to the energy dashboard.

ortigosano, 2023-11-20:
> ```yaml
> "Water consumption
> ```

Thanks!
The sensor works but still not recognized as sensor for the water consumption dashboard:

[image]
[image]
[image]

Am I doing something wrong?

Thans in advance!!

florianbelhomme, 2023-11-20:
I've updated my exemple, i've added the device_class that should make it work.

ortigosano, 2023-11-20:
> I've updated my exemple, i've added the device_class that should make it work.

unfortunately it doesn't for me. Thanks anyway!

ortigosano, 2023-11-20:
> I've updated my exemple, i've added the device_class that should make it work.

Sorry @florianbelhomme , I copied another sample wich was on my clipboard 🤦‍♂️(no comments) 

If I copy yours:

```
template:
  - sensor:
    - name: "Water consumption"
      unit_of_measurement: "L"
      device_class: "water"
      state_class: "total_increasing"
      state: {{ states('sensor.zipulses_agua_general_energy') | float }}
      availability: "{{ states('sensor.zipulses_agua_general_energy') not in ['none', 'unknown', 'unavailable'] }}"
```

I get the following error:

```
Configuration invalid!
Error loading /config/configuration.yaml: invalid key: "{"states('sensor.zipulses_agua_general_energy') | float": None}"
in "/config/configuration.yaml", line 21, column 0
```

[image]

I'm using the latest version of HA OS:

Core
2023.11.2
Supervisor
2023.11.3
Operating System
11.1
Frontend
20231030.2
 
Merci!!

florianbelhomme, 2023-11-21:
It was missing double quotes around the state parameter value

ortigosano, 2023-11-21:
> It was missing double quotes around the "state" parameter value
Thanks a lot! Much appreciated it! 
I finally think I have succeeded.!!

Do we need to add an utility meter like this? 

```
  daily_water:
    source: sensor.watermeter_total_water_usage
    name: Water Usage Daily Utility Config
    cycle: daily
```

closed 2024-06-19 by github-actions[bot]