Koenkk/zigbee2mqtt #19582: OTA update Ikea Symfonisk sound controller (IKEA E1744) failed

yvos, 2023-11-05:
### What happened?

When updating my Ikea Symfonisk sound controller (IKEA E1744) 
from firmware version: 2.3.080 to the latest the debug log shows: Error: Device didn't respond to OTA request

### What did you expect to happen?

Succefull update the firmware

### How to reproduce it (minimal and precise)

Register a Ikea Symfonisk sound controller (IKEA E1744) in zigbee2mqtt with firmware version 2.3.080 and press update firmware in the OTA tab.

After that check the log (set to debug first)

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

7.3.1.0 build 0

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

debug 2023-11-05 16:12:45: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Ikea Symfonisk sound controller","transaction":"ekc4p-1"}'
info  2023-11-05 16:12:45: Updating 'Ikea Symfonisk sound controller' to latest firmware
debug 2023-11-05 16:13:05: Updating to latest '0x50325ffffe6daeb0' (SYMFONISK Sound Controller)
debug 2023-11-05 16:13:05: Using endpoint '1'
debug 2023-11-05 16:14:05: Update of 'Ikea Symfonisk sound controller' failed (Error: Device didn't respond to OTA request)
info  2023-11-05 16:14:05: MQTT publish: topic 'zigbee2mqtt/Ikea Symfonisk sound controller', payload '{"action":null,"battery":100,"brightness":245,"linkquality":255,"rate":195,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":null}'
info  2023-11-05 16:14:05: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Ikea Symfonisk sound controller"},"error":"Update of 'Ikea Symfonisk sound controller' failed (Device didn't respond to OTA request)","status":"error","transaction":"ekc4p-1"}'
error 2023-11-05 16:14:05: Update of 'Ikea Symfonisk sound controller' failed (Device didn't respond to OTA request)
debug 2023-11-05 16:14:05: Error: Device didn't respond to OTA request
    at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:202:15)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:314:21)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:268:41)

VerodinDev, 2023-11-30:
Having the same issue. Recently got into updating my Ikea devices and this one got dusted off for an update. But allas, I too got "Update of '0xccccccfffebd899e' failed (Device didn't respond to OTA request)".

Trying to go from v2.1.022 to the latest update.

Eistee82, 2024-02-14:
My E1744 is on firmware 2.3.075 and update is not success. It will upload the firmware and then the device did a reboot but old firmware is shown after that.

```javascript
{
    "device": {
        "applicationVersion": 33,
        "dateCode": "20210526",
        "friendlyName": "0xccccccfffea8c6b2",
        "hardwareVersion": 1,
        "ieeeAddr": "0xccccccfffea8c6b2",
        "manufacturerID": 4476,
        "manufacturerName": "IKEA of Sweden",
        "model": "E1744",
        "networkAddress": 24254,
        "powerSource": "Battery",
        "softwareBuildID": "2.3.075",
        "stackVersion": 98,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "last_seen": "2024-02-14T18:44:05+01:00",
    "linkquality": 153,
    "update": {
        "installed_version": 587683377,
        "latest_version": 604241925,
        "state": "available"
    },
    "battery": 60,
    "update_available": null,
    "brightness": 255,
    "action": "brightness_stop",
    "action_rate": 195
}
``` 
(Battery is not the problem I also try with a new battery.)

YanisKyr, 2024-06-10:
I have 2 of these SYMFONISK / E1744, both on v2.1.022 and the OTA fails.

Is there anything I can share to help investigate this further?

mboarman, 2024-06-29:
I also have a couple of these devices and they will not update to 2.3.080.  I get the follow error in my logs:

Update of 'Volume-SittingRoom' failed (OTA: Device didn't respond to OTA request)

They both have new batteries.

closed 2025-01-11 by github-actions[bot]

drsound, 2025-05-19:
I was able to resolve it by ensuring the device remained active at the start of the update process. I continuously rotate the knob when initiating the update. Once I confirmed the update had actually begun (e.g., by observing the Zigbee2MQTT logs indicating firmware transfer activity), I stopped rotating the knob and left the device alone. The update then proceeded and completed without any further issues.

Nine-sfj, 2026-10-25:
Thanks a lot for that tip @drsound!! I can confirmed that it worked for me as well 🙌