Koenkk/zigbee2mqtt #19586: SR-ZG9101SAC-HP-SWITCH-2CH ('Vesternet Zigbee 2 channel switch (VES-ZB-SWI-015)') wrong configuration

devildant, 2023-11-05:
### What happened?

hello, when i try to add the device SR-ZG9101SAC-HP-SWITCH-2CH i take an error for haElectricalMeasurement and seMetering
inside the documentation of the product i see that the electrical Measurment and simple metering are present on endpoint 11 not 1
[image]

but instide zigbee-herdsman-converters it's map on endpoint 1
[image]

i try to put external_converters without success, the internal config are systematically load

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['VES-ZB-SWI-015'],
    model: 'VES-ZB-SWI-015',
    vendor: 'Vesternet',
    description: 'Zigbee 2 channel switch test',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.power_on_behavior, fz.ignore_genOta],
    toZigbee: [tz.on_off, tz.power_on_behavior], // <-- add here all clusters to send commands to zigbee
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage(), e.energy(), e.power_on_behavior(['off', 'on', 'previous'])],
	whiteLabel: [{vendor: 'Sunricher', model: 'SR-ZG9101SAC-HP-SWITCH-2CH'}],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(2);
            const endpoint11 = device.getEndpoint(11);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(endpoint11, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint1);
            await reporting.onOff(endpoint2);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint11);
            await reporting.activePower(endpoint11);
            await reporting.rmsCurrent(endpoint11, {min: 10, change: 10});
            await reporting.rmsVoltage(endpoint11, {min: 10});
            await reporting.readMeteringMultiplierDivisor(endpoint11);
            await reporting.currentSummDelivered(endpoint11);
        },
};

module.exports = definition;
```

### What did you expect to happen?

[image]

have power, energy.....

### Zigbee2MQTT version

1.33.2-1

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Debug log

Debug 2023-11-05 20:04:55Received Zigbee message from '0x385b44fffe0f2780', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 11 with groupID 0
Debug 2023-11-05 20:04:55Received Zigbee message from '0x385b44fffe0f2780', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 11 with groupID 0
Info 2023-11-05 20:04:55Successfully interviewed '0x385b44fffe0f2780', device has successfully been paired
Info 2023-11-05 20:04:55Device '0x385b44fffe0f2780' is supported, identified as: Vesternet Zigbee 2 channel switch (VES-ZB-SWI-015)
Info 2023-11-05 20:04:56MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee 2 channel switch","exposes":[{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","label":"State","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","label":"State","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Controls the behavior when the device is powered on after power loss","label":"Power-on behavior","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","on","previous"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"VES-ZB-SWI-015","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","label":"Current calibration","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device.","label":"Current precision","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","label":"Voltage calibration","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device.","label":"Voltage precision","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device.","label":"Energy precision","name":"energy_precision","property":"energy_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the energy value (percentual offset), takes into effect on next report of device.","label":"Energy calibration","name":"energy_calibration","property":"energy_calibration","type":"numeric"}],"supports_ota":false,"vendor":"Vesternet"},"friendly_name":"0x385b44fffe0f2780","ieee_address":"0x385b44fffe0f2780","status":"successful","supported":true},"type":"device_interview"}'
Info 2023-11-05 20:04:56Configuring '0x385b44fffe0f2780'
Debug 2023-11-05 20:04:58Received Zigbee message from '0x385b44fffe0f2780', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Error 2023-11-05 20:04:58Failed to configure '0x385b44fffe0f2780', attempt 3 (Error: Read 0x385b44fffe0f2780/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:22) at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:29:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/vesternet.ts:125:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))

devildant, 2023-11-05:
when i bind the endpoint 11 to coordinator with haElectricalMeasurement and setMetering i see the power, current and voltage, but with wrong value, voltage for exemple show 2404v instead of 240.2v, it's same for power and current, but i don't know the correct value

[image]

[image]

devildant, 2023-11-05:
i found, just add finger print inside file :p

also i create a pull request :
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6411

closed 2023-11-07 by devildant

chris-1243, 2023-12-12:
I took me some times but your converter (and your correction) did the trick....

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'ON/OFF(2CH)', softwareBuildID: '2.9.2_r3'}],
    model: 'VES-ZB-SWI-015',
    vendor: 'Vesternet',
    description: 'Zigbee 2 channel switch',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.power_on_behavior, fz.ignore_genOta],
    toZigbee: [tz.on_off, tz.power_on_behavior], // <-- add here all clusters to send commands to zigbee
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage(), e.energy(), e.power_on_behavior(['off', 'on', 'previous'])],
	whiteLabel: [{vendor: 'Sunricher', model: 'SR-ZG9101SAC-HP-SWITCH-2CH'}],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2};
        },
        meta: {multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(2);
            const endpoint11 = device.getEndpoint(11);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await reporting.bind(endpoint11, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint1);
            await reporting.onOff(endpoint2);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint11);
            await reporting.activePower(endpoint11);
            await reporting.rmsCurrent(endpoint11, {min: 10, change: 10});
            await reporting.rmsVoltage(endpoint11, {min: 10});
            await reporting.readMeteringMultiplierDivisor(endpoint11);
            await reporting.currentSummDelivered(endpoint11);
        },
};

module.exports = definition;
`

devildant, 2023-12-12:
hello, I still use my converter, because my pull request was rejected because it causes problems on old versions of the product. and I don't have much time at the moment to look

chris-1243, 2023-12-12:
I read the different threads and this why I finally went the same way as you.

I wish I could help but my programming knowledges are really not good. 

There is exactly the same coding problem in the sunricher converter for this particular model. I just had a look...

devildant, 2023-12-12:
the implementation that I made is good, it's just that you have to split the configuration by version, but it's not really planned for what I saw in z2m so you'd have to rename the device or something of the kind