Koenkk/zigbee2mqtt #19588: EAKCB-T-M-Z Increase Overvoltage Limit

lucasimons, 2023-11-05:
### What happened?

I use this DIN circuit breaker but the range of overvoltage threshold would be edited ... Now is 220 to 260V but the default value is 265V so we need to ingrease the max value to 265V
 

```
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:194
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 20:25:45 (13 occurrences)
Last logged: 20:35:55

Invalid value for number.ingresso_enel_inverter_2_over_voltage_threshold: 265 (range 220.0 - 260.0)

```

[image]

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20210120

### Adapter

zStack3x0

Koenkk, 2023-11-05:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-11-05 by Koenkk

lucasimons, 2023-11-05:
When this fix is available in the normal add on not the edge version?

Koenkk, 2023-11-06:
Next release is on 1 December.