Koenkk/zigbee2mqtt #19591: TS0601 _TZE204_dvosyycn 8 Chanel switch

gvlad01, 2023-11-05:
### Link

https://de.aliexpress.com/item/1005006132345060.html?spm=a2g0o.store_pc_allProduct.8148356.27.58973f3bJVxxt6&pdp_npi=4%40dis%21EUR%21%E2%82%AC%200%2C94%21%E2%82%AC%200%2C66%21%21%217.18%215.03%21%402103890116992174001185725e3478%2112000035903085154%21sh%21DE%21179425550%21

### Database entry

Router, 0xa4c138636d03a45d

### Comments

Hello, are there any plans to integrate this module into zigbee2MQTT?
This module has 8 inputs and 8 outputs, I believe this is from the same manufacturer as the 24 channel module https://www.zigbee2mqtt.io/devices/ZYXH.html#tuya-zyxh
I can't convert myself, I don't know
Thank you very much

DennisDeKoster, 2023-12-14:
Same here. Got one today, not knowing how easy or difficult it would be haha. I'll keep searching how to implement is. If I find it before anyone else can help us, I'll share it here.

gvlad01, 2023-12-14:
Great, thank you.

DennisDeKoster, 2023-12-16:
Hi gvlad01,

I got mine working using parts of code from the converter for the 5-way switch. I found out it's not a really a separate 8 input + 8 output module out of the box. The inputs and outputs are linked together. So when input 1 is activated, output 1 will be so too. I couldn't find a way to change any settings on it (I don't have a Tuya gateway). So my plan is to use an input to signal from an external device and leave its corresponding output not connected. And use a separate output for something else and leave its input not connected.

Here's how I got mine working in Home Assistant:

- Create a converter file, something like "TS0601_8_way_switch.js" with the code below and place the file in `homeassistant/config/zigbee2mqtt/`
- Add these lines to your `/homeassistant/zigbee2mqtt/configuration.yaml`
```
external_converters:
  - TS0601_8_way_switch.js
```
- Restart Zigbee2mqtt and add the device when Z2M is running again.

I've added unused datapoints with null, so those will be ignored. I don't know if that is really necessary though.
Let me know if yours is working.

Kind regards, 
Dennis

Converter code:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_dvosyycn',
        },
    ],
    model: 'TS0601_8_way_switch',
    vendor: 'TuYa',
    description: '8 way switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        ...Array.from({length: 8}, (_, i) => tuya.exposes.switch().withEndpoint(`l${i + 1}`))
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1};
    },
    meta: {
        // All datapoints go in here
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [0x65, 'state_l7', tuya.valueConverter.onOff],
            [0x66, 'state_l8', tuya.valueConverter.onOff],
            [0x67, 'null', null],
            [0x68, 'null', null],
            [0x69, 'null', null],
            [0x6A, 'null', null],
            [0x6B, 'null', null],
            [0x6C, 'null', null],
            [0x6D, 'null', null],
            [0x6E, 'null', null],
            [0xB6, 'null', null],           //Datapoint 182?
            [0xB7, 'null', null],           //Datapoint 183?
            [0xB8, 'null', null],           //Datapoint 184?
            [0xB9, 'null', null],           //Datapoint 185?
            [0xBA, 'null', null],           //Datapoint 186?
            [0xBB, 'null', null],           //Datapoint 187?
            [0xBC, 'null', null],           //Datapoint 188?
            [0xBD, 'null', null],           //Datapoint 189?
        ],
    },
};

module.exports = definition;

```

Koenkk, 2023-12-17:
I've added the switch!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

gvlad01, 2023-12-29:
So, I added the device, 
everything works, 
thank you again 🤝

DennisDeKoster, 2023-12-30:
No problem 😊

closed 2023-12-30 by Koenkk

iN-etlaM, 2024-02-23:
I got a 8 gang Tuya switch as well.  
In my case only Switch (l8 endpoint) works. from 1-7 i get this message for example: "No converter available for 'state_l7' ("ON")" 
Can you help me? :) Thanks in advance

clumsy-stefan, 2024-03-04:
I have a similar device which also reports as TZE204_dvosyycn. However it seems to have 8 analog input's (0-5.5V/VCC) and 8 digital outputs from [aliexpress here](https://www.aliexpress.com/item/1005006132345060.html?spm=a2g0o.productlist.main.3.7ee64187dPk1oR&algo_pvid=972360b7-d035-4f0d-b799-1e56112933ea&algo_exp_id=972360b7-d035-4f0d-b799-1e56112933ea-1&pdp_npi=4%40dis%21CHF%2111.55%2111.55%21%21%2192.25%2192.25%21%402103010f17094630296717609e2bfe%2112000035903085153%21sea%21CH%21134377239%21&curPageLogUid=BsgdmBOIwgOr&utparam-url=scene%3Asearch%7Cquery_from%3A) 

I managed to amend the definitions, so that I can read the analog input's and the digital outputs. By default inputs and outputs are linked together, so if an input goes to high (~ >4V) the respective output also changes. However according to the description it's possible to decouple the input's from the output's with a config setting and then use tehm sepparately (which is what I'm trying to acheive). Unfortunately I don't succeed in configuring, as I don't know which cluster/setting has to be changed.

Also I don't exactly know what the DP 0x67-0x6E really do. They only get transferred rarely..

Any help would be appreciated.

To read all the exposed datapoints from this board I ammended the config as follows:
```
    {
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_vmcgja59' },
            { modelID: 'TS0601', manufacturerName: '_TZE204_dvosyycn' },
        ],
        model: 'TS0601_switch_8',
        vendor: 'TuYa',
        description: 'ZYXH 8 gang switch',
        fromZigbee: [tuya.fz.datapoints,fzLocal.TS110E_switch_type],
        toZigbee: [tuya.tz.datapoints,fzLocal.TS110E_switch_type],
        configure: tuya.configureMagicPacket,
        exposes: [
            ...Array.from({ length: 8 }, (_, i) => tuya.exposes.switch().withEndpoint(`l${i + 1}`)),
            ...Array.from({ length: 8 }, (_, i) => e.binary(`set`, ea.STATE, true, false).withEndpoint(`a${i + 1}`).withDescription(`Bool a${i + 1}`)),
            ...Array.from({ length: 8 }, (_, i) => e.voltage().withEndpoint(`a${i + 1}`).withDescription(`Voltage ${i + 1}`)),
        ],
        endpoint: (device) => {
            return { 'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1,
                    'a1': 1, 'a2': 1, 'a3': 1, 'a4': 1,'a5': 1, 'a6': 1,'a7': 1, 'a8': 1,
            };
        },
        meta: {
            multiEndpoint: true,
            tuyaDatapoints: [
                [1, 'state_l1', tuya.valueConverter.onOff],
                [2, 'state_l2', tuya.valueConverter.onOff],
                [3, 'state_l3', tuya.valueConverter.onOff],
                [4, 'state_l4', tuya.valueConverter.onOff],
                [5, 'state_l5', tuya.valueConverter.onOff],
                [6, 'state_l6', tuya.valueConverter.onOff],
                [0x65, 'state_l7', tuya.valueConverter.onOff],
                [0x66, 'state_l8', tuya.valueConverter.onOff],
                [0x67, 'set_a1', tuya.valueConverter.trueFalse],
                [0x68, 'set_a2', tuya.valueConverter.trueFalse],
                [0x69, 'set_a3', tuya.valueConverter.trueFalse],
                [0x6A, 'set_a4', tuya.valueConverter.trueFalse],
                [0x6B, 'set_a5', tuya.valueConverter.trueFalse],
                [0x6C, 'set_a6', tuya.valueConverter.trueFalse],
                [0x6D, 'set_a7', tuya.valueConverter.trueFalse],
                [0x6E, 'set_a8', tuya.valueConverter.trueFalse],
                [0xB6, 'voltage_a1', tuya.valueConverter.divideBy10],
                [0xB7, 'voltage_a2', tuya.valueConverter.divideBy10],
                [0xB8, 'voltage_a3', tuya.valueConverter.divideBy10],
                [0xB9, 'voltage_a4', tuya.valueConverter.divideBy10],
                [0xBA, 'voltage_a5', tuya.valueConverter.divideBy10],
                [0xBB, 'voltage_a6', tuya.valueConverter.divideBy10],
                [0xBC, 'voltage_a7', tuya.valueConverter.divideBy10],
                [0xBD, 'voltage_a8', tuya.valueConverter.divideBy10],
            ],
        },
    },
```

pirracas77, 2024-04-09:
> I have a similar device which also reports as TZE204_dvosyycn. However it seems to have 8 analog input's (0-5.5V/VCC) and 8 digital outputs from [aliexpress [...]

I have a similar device https://www.zigbee2mqtt.io/devices/TY-04Z.html#zyxh-ty-04z.

I would like also to  decouple the input's from the output's and use the input as external digital input sensor.
Did you manage to set up this configuration?
Also I opened this issue, as these kind of relays have other options not yet exposed in zigbee2mqtt.
https://github.com/Koenkk/zigbee2mqtt/issues/22145

clumsy-stefan, 2024-04-09:
Yes I could decouple it successfully, see #7232 for a complete template to include. Not sure though if it's possible with your board.
I'm preparing a pull request to be included in z2m, but currently I'm a bit short on time. However you can still include the template from #7232 locally in your installation and use all functions of the board.

pirracas77, 2024-04-09:
> Yes I could decouple it successfully, see #7232 for a complete template to include. Not sure though if it's possible with your board. I'm preparing a pull request to be included in z2m, but [...]

Thanks for your quick reply.

Are you sure I is the 7232 ?

clumsy-stefan, 2024-04-09:
Sorry, its 7232 but in the zigbee-herdsman-converters repository, see [here](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7232)

pirracas77, 2024-04-09:
> Sorry, its 7232 but in the zigbee-herdsman-converters repository, see [here](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7232)
> 
> EDIT: it's a bit strange how you need to change it though: first set "channel select" (value 1) to the channel you want to decouple, then set the "channel mode" (value 2) to 0 for decoupled, 2 for direct linked, 3 for blinking, etc...

thanks, I will take a look

pirracas77, 2024-06-12:
> Sorry, its 7232 but in the zigbee-herdsman-converters repository, see [here](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7232)
>
> EDIT: it's a bit strange how you need to change it [...]

Hello again,

I have discovered how decoupling, inching and interlock is managed in my device by Tuya app using IOT Tuya web.

- DP191 = 108 is for set up interlock
- DP191 = 131 is for set up JOG (inching)
- DP191 = 105 is for decoupled, direct linked etc
- DP191 = 106 is for type of signal (???)
- DP192 is for select number of channel
- DP193 depends on if you are in JOG,INCHING or decoupled settings.

Looking in to your  converter, I think they are the same as yours.  Aren't they?

On the other hand when decoupled is selected, How the values of the imputs are reported to Tuya? I cannot see them.

Thanks in advance

clumsy-stefan, 2024-06-12:
I have no clue what DP190 and DP191 do, I just found that when setting 192 to a certain channel and then setting 193 to the desired channel mode it seems to work...
So, yes, it looks the same.

Values are reported in voltage_a1 through voltage_a8

pirracas77, 2024-06-12:
> I have no clue what DP190 and DP191 do, I just found that when setting 192 to a certain channel and then setting 193 to the desired channel mode it seems to work... So, yes, it looks the same.
> 
> Values are reported in voltage_a1 through voltage_a8

I realize taht DP190 have to be set to 10 in order to start operating as decoupled. In fact, the app also changes and start to show the imputs (previously this section was not there)

[image]

So now I'm sure that I can use you converter.

clumsy-stefan, 2024-06-12:
Ok, great! Let me know if you ammend the converter, so I can also update it!

pirracas77, 2024-06-13:
> DP190

Sadly I don't have any idea how to add the DPs to the already existing converter. I tried to follow the documentation but it is not clear for me.  I do not know what I'm doing LOL. I hope someone in the original [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6226) could add them

pirracas77, 2024-06-13:
I have tried to include some datapoint into the converter. I have restarted zigbe2mqtt but the new DP doesn't appear. What I'm doing wrong? 

`    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_aqnazj70', '_TZE200_di3tfv5b', '_TZE200_mexisfik', '_TZE204_6wi2mope', '_TZE204_iik0pquw',
            '_TZE204_aagrxlbd']),
        model: 'TS0601_switch_4_gang_1',
        vendor: 'TuYa',
        description: '4 gang switch',
        exposes: [
            tuya.exposes.switch().withEndpoint('l1'),
            tuya.exposes.switch().withEndpoint('l2'),
            tuya.exposes.switch().withEndpoint('l3'),
            tuya.exposes.switch().withEndpoint('l4'),
            ...Array.from({ length: 4 }, (_, i) => e.binary(`input`, ea.STATE, true, false).withEndpoint(`a${i + 1}`).withDescription(`Bool a${i + 1}`)),
        ],
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        whiteLabel: [
            tuya.whitelabel('ZYXH', 'TY-04Z', '4 gang switch', ['_TZE204_iik0pquw']),
            { vendor: 'Norklmes', model: 'MKS-CM-W5' },
            { vendor: 'Somgoms', model: 'ZSQB-SMB-ZB' },
            { vendor: 'Moes', model: 'WS-EUB1-ZG' },
            { vendor: 'AVATTO', model: 'ZGB-WS-EU' },
        ],
        meta: {
            multiEndpoint: true,
            tuyaDatapoints: [
                [1, 'state_l1', tuya.valueConverter.onOff],
                [2, 'state_l2', tuya.valueConverter.onOff],
                [3, 'state_l3', tuya.valueConverter.onOff],
                [4, 'state_l4', tuya.valueConverter.onOff],
                [189, 'input_a1', tuya.valueConverter.trueFalse],
                [188, 'input_a2', tuya.valueConverter.trueFalse],
                [187, 'input_a3', tuya.valueConverter.trueFalse],
                [186, 'input_a4', tuya.valueConverter.trueFalse],
            ],
        },
        endpoint: (device) => {
            return { 'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'a1': 1, 'a2': 1, 'a3': 1, 'a4': 1 };
        },
    }`

pirracas77, 2024-06-14:
Ok, I've started to understand how it works. It is necessary to use a external converter. The file tuya.js cannot be eddited directly.

Here you are my first testing. I don't understand why sometimes my changes appears en the zigbee2mqtt webpage and sometimes don't.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', [
        '_TZE200_aqnazj70', '_TZE200_di3tfv5b', '_TZE200_mexisfik',
        '_TZE204_6wi2mope', '_TZE204_iik0pquw', '_TZE204_aagrxlbd'
    ]),
    model: 'TS0601_switch_4_gang_1',
    vendor: 'TuYa',
    description: 'Interruptor de 4 canales',
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.switch().withEndpoint('l3'),
        tuya.exposes.switch().withEndpoint('l4'),
        e.enum('mode', ea.STATE_SET, ['10', '115']).withDescription('Mode'),
    ],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints, {
        key: ['mode'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 190; // Define the datapoint
            const val = value === '10' ? 10 : 115; // Convert value to appropriate numeric value
            await tuya.sendDataPointValue(entity, dp, val);
            return { state: { mode: value } };
        },
    }],
    configure: tuya.configureMagicPacket,
    whiteLabel: [
        tuya.whitelabel('ZYXH', 'TY-04Z', 'Interruptor de 4 canales', ['_TZE204_iik0pquw']),
        { vendor: 'Norklmes', model: 'MKS-CM-W5' },
        { vendor: 'Somgoms', model: 'ZSQB-SMB-ZB' },
        { vendor: 'Moes', model: 'WS-EUB1-ZG' },
        { vendor: 'AVATTO', model: 'ZGB-WS-EU' },
    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [190, 'mode', tuya.valueConverterBasic.lookup({ '10': 10, '115': 115 })],
        ],
    },
    extend: modernExtend.switch(),
};

```

eduseiden, 2025-03-28:
Hi everyone,

I'm using Zigbee2MQTT version 1.41 and I have an 8-gang Tuya wall switch that works perfectly with the external converter below. However, I haven’t been able to disable the backlight.

I’ve tried several datapoints (like 0x67, 0x68, etc.) with no success.

Has anyone figured out how to turn off the backlight on this device? If you know the correct datapoint or have a working example, I’d really appreciate your help.

Here’s the external converter I’m using:

***********************

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_nvxorhcj',
        },
    ],
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_nvxorhcj',
    description: '8 gang wall touch switch board',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
    exposes: [
        ...Array.from({length: 8}, (_, i) => 
            tuya.exposes.switch()
                .withEndpoint(`l${i + 1}`)
                .withDescription(`Switch ${i + 1}`)
        ),
        // Backlight
        e.binary('backlight_mode', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Mode of the backlight. ON = backlight on when switch is off, OFF = backlight off when switch is off'),
    ],
    endpoint: (device) => {
        return Object.fromEntries(
            Array.from({length: 8}, (_, i) => [`l${i + 1}`, 1])
        );
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [0x65, 'state_l7', tuya.valueConverter.onOff],
            [0x66, 'state_l8', tuya.valueConverter.onOff],
            // Add datapoint - backlight_mode
            [0x67, 'backlight_mode', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;

***********************

Thanks in advance.

Eduardo

eduseiden, 2025-03-31:
Update: backlight working fine. The only issue is if you want to control a specific endpoint in Z2M group.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

// Conversor fromZigbee personalizado para DP 14 e 15
const fzSpecialDPs = {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp;
        const data = msg.data.data;
        const result = {};

        // Light mode (DP 15)
        if (dp === 15) {
            const lookup = {0: 'none', 1: 'relay', 2: 'pos'};
            const value = tuya.getDataValue(data);
            result.light_mode = lookup[value] || 'none';
        }
        // Restart status (DP 14)
        else if (dp === 14) {
            const lookup = {0: 'power_off', 1: 'power_on', 2: 'last'};
            const value = tuya.getDataValue(data);
            result.restart_status = lookup[value] || 'power_off';
        }

        return result;
    }
};

// Conversor toZigbee personalizado para DP 14 e 15
const tzSpecialDPs = {
    key: ['light_mode', 'restart_status'],
    convertSet: async (entity, key, value, meta) => {
        if (key === 'light_mode') {
            const lookup = {'none': 0, 'relay': 1, 'pos': 2};
            const modeValue = lookup[value];
            
            if (modeValue !== undefined) {
                await tuya.sendDataPointEnum(entity, 15, modeValue);
                return {light_mode: value};
            }
        }
        else if (key === 'restart_status') {
            const lookup = {'power_off': 0, 'power_on': 1, 'last': 2};
            const modeValue = lookup[value];
            
            if (modeValue !== undefined) {
                await tuya.sendDataPointEnum(entity, 14, modeValue);
                return {restart_status: value};
            }
        }
    },
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_nvxorhcj',
        },
    ],
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_nvxorhcj',
    description: '8 gang wall touch switch board',
    // Combinando os dois métodos
    fromZigbee: [fzSpecialDPs, tuya.fz.datapoints],
    toZigbee: [tzSpecialDPs, tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
    exposes: [
        ...Array.from({length: 8}, (_, i) => 
            tuya.exposes.switch()
                .withEndpoint(`l${i + 1}`)
                .withDescription(`Switch ${i + 1}`)
        ),
        exposes.enum('light_mode', ea.STATE_SET, ['none', 'relay', 'pos'])
            .withDescription('Control the light mode (backlight). Options: none, relay, pos'),
        exposes.enum('restart_status', ea.STATE_SET, ['power_off', 'power_on', 'last'])
            .withDescription('Control the restart status. Options: power_off, power_on, last'),
    ],
    endpoint: (device) => {
        return Object.fromEntries(
            Array.from({length: 8}, (_, i) => [`l${i + 1}`, 1])
        );
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [0x65, 'state_l7', tuya.valueConverter.onOff],
            [0x66, 'state_l8', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;