Koenkk/zigbee2mqtt #19592: State change does not work when switch bound to a light group

Diaoul, 2023-11-05:
### What happened?

I'm using a group binding to control a Zigbee light ([Paul Neuhaus NLG-TW](https://www.zigbee2mqtt.io/devices/NLG-TW_light.html)) from a zigbee switch ([Legrand 067773](https://www.zigbee2mqtt.io/devices/067773.html)) instead of direct binding as described in the [binding documentation](https://www.zigbee2mqtt.io/guide/usage/binding.html#state-changes).

- When I press the button of the switch, the state is not updated in zigbee2mqtt
- When I force-refresh the state from the UI, I get the updated value correctly

> If your devices do not support reporting put the device in a group and bind the remote to the group instead of directly to the device. This will make Zigbee2MQTT poll the device for updates when the bound remote controls the device. To minimize traffic this has not been enabled for all devices. If this does not work please create an issue for it [here](https://github.com/Koenkk/zigbee2mqtt/issues)

In my case, I don't think the polling works: the states is not updated

### What did you expect to happen?

I expected that the state would be updated

### Zigbee2MQTT version

1.33.2-dev commit: be48d54

### Adapter firmware version

20230507

### Adapter

tube_zb_gw_cc2652p2_poe

### Debug log

When I press the button on the switch:

```
debug 2023-11-05 22:59:54Received Zigbee message from 'Living Room Entrance Light Switch', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
debug 2023-11-05 22:59:54Received Zigbee message from 'Living Room Entrance Light Switch', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 8
info 2023-11-05 22:59:54MQTT publish: topic 'zigbee2mqtt/Living Room Entrance Light Switch', payload '{"action":"toggle","action_group":8,"battery":80,"brightness":0,"linkquality":54,"rate":255,"update":{"installed_version":2769411,"latest_version":4473347,"state":"available"},"update_available":null,"voltage":2900}'
info 2023-11-05 22:59:54MQTT publish: topic 'zigbee2mqtt/Living Room Entrance Light Switch/action', payload 'toggle'
debug 2023-11-05 22:59:54Received Zigbee message from 'Living Room Entrance Light Switch', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2023-11-05 22:59:54Device 'Living Room Entrance Light Switch' announced itself
info 2023-11-05 22:59:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living Room Entrance Light Switch","ieee_address":"0x0004740000c67791"},"type":"device_announce"}'
info 2023-11-05 22:59:54MQTT publish: topic 'zigbee2mqtt/Living Room Entrance Light Switch', payload '{"action":"toggle","battery":80,"brightness":0,"linkquality":105,"rate":255,"update":{"installed_version":2769411,"latest_version":4473347,"state":"available"},"update_available":null,"voltage":2900}'
info 2023-11-05 22:59:54MQTT publish: topic 'zigbee2mqtt/Living Room Entrance Light Switch/action', payload 'toggle'
debug 2023-11-05 22:59:55Received Zigbee message from 'Living Room Entrance Light Switch', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
info 2023-11-05 22:59:55MQTT publish: topic 'zigbee2mqtt/Living Room Entrance Light Switch', payload '{"battery":100,"brightness":0,"linkquality":102,"rate":255,"update":{"installed_version":2769411,"latest_version":4473347,"state":"available"},"update_available":null,"voltage":3000}'
debug 2023-11-05 22:59:56Retrieving state of 'Living Room Entrance Light Switch' after reconnect
```

miezie, 2024-03-06:
Can confirm this behavior with a Philips Hue Wall Switch module and a Paul Neuhaus NLG-TW light

Diaoul, 2024-03-06:
It seems to not affect Paul Neuhaus lights with a more recent firmware. However, I would love to have this fixed as there is currently no way to update the firmware on those lights...

Diaoul, 2024-09-03:
This is still a problem. While there may be some hardware related or spec compliance issue on some lights, I think it would be great to be able to force trigger a refresh on commands impacting affected lights since this seem to work fine.

Diaoul, 2025-01-03:
Not stale

Diaoul, 2025-03-06:
I'd love to get this sorted out, this is still an issue.

For now the lights change state OK but the reporting of its state is broken. Hitting the refresh arrows on the brightness in expose tab for example refreshes the state alright.

Maybe it's just a matter of doing that automatically whenever we send a state-changing command to the light?

Diaoul, 2025-05-07:
Still getting the issue, I'd love to find a software solution to this and not have to replace all my lights.

@Koenkk I'm happy to debug and provide a fix PR myself but I'm not sure where to start looking, do you hav any pointers?

iamkarlson, 2025-05-31:
This is very annoying problem. I would prefer direct binding over HA automation but this bug is frustrating

iamkarlson, 2025-08-04:
Not stale

Diaoul, 2025-10-04:
@Nerivec do you think this can be an issue with the firmware on the coordinator?

Nerivec, 2025-10-04:
From early comments, this affects specific device firmware, so, not something that Z2M can fix.
As a workaround, maybe the converter could do a read after a write, but I'm not familiar with these devices at all.

Diaoul, 2025-10-05:
Apparently the reporting of the device is broken entirely. It just never reports state back when the light is changing following a bound remote action.

My options seem to be:
1. Do aggressive polling like [this one](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/gledopto.ts#L344-L344)
2. Use home-assistant automations to trigger the state change such that I don't rely on device reporting
3. A combination of the 2 (e.g. on-off through automations but brightness in direct control)
4. Find a way to update device's hardware or order a new led driver from [Anway](http://www.anwayelec.com/product.php) (seems to be the manufacturer based on the labels of the PCBs)
5. Change light's hardware entirely from Aliexpress (already did on one, its standard CCT led)

Happy to have your thoughts 🙏

closed 2025-12-12 by Diaoul