Koenkk/zigbee2mqtt #19593: [Feature request]: Philips Effects Support for "Glisten", "Opal", and "Prism"

thk-socal, 2023-11-06:
### Is your feature request related to a problem? Please describe

With the new Philips Festavia String Lights, there are three new effects. Those effects are Glisten, Opal, and Prism. These are in addition to the existing "Candle", "Fireplace", and "Sparkle".

### Describe the solution you'd like

I would like to see the same options as the existing Philips Hue Effects.

### Describe alternatives you've considered

None

### Additional context

None

thk-socal, 2023-11-06:
Actually I now see that Glisten and Opal are added. Please change to just Prism as the additional.

baloo2, 2024-05-29:
As a LOT OF HUE lights received update to support new effects (firmware 1.116.3 from May 2024) I woul like to ask for Z2M update as well. Thanks.

deviantintegral, 2024-05-29:
I'm able to trigger the new effects (opal, glisten, and sparkle) on my [9290022266A](https://www.zigbee2mqtt.io/devices/9290022266A.html#philips-9290022266a) bulbs just by sending those strings as the effect value. So, I think this is just updating the definition at https://github.com/Koenkk/zigbee-herdsman-converters/blob/b876f557a623be8c1cd2a9cce114be58e134de83/src/lib/philips.ts#L66-L67 to support them.

merlinschumacher, 2024-07-01:
I have the following lamps: 
- 929002966501
- 929002471601

they both support opal, glisten, sparkle and sunrise. But neither supports prism.

closed 2025-01-11 by github-actions[bot]