Koenkk/zigbee2mqtt #19597: Pull 19542 introduces lots of errors in log file -  Erroneous JSON: {"in_progress": False}

derekoharrow, 2023-11-06:
### What happened?

The fix applied by https://github.com/Koenkk/zigbee2mqtt/pull/19542 introduces lots of repeating errors in the HA log file:

```
2023-11-06 14:11:03.086 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:11:10.083 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:11:28.603 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:11:51.664 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:23.620 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:46.709 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:47.098 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:49.691 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:50.096 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:51.424 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:54.406 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
2023-11-06 14:12:55.642 WARNING (MainThread) [homeassistant.components.mqtt.mixins] Erroneous JSON: {"in_progress": False}
```

### What did you expect to happen?

No errors.

### How to reproduce it (minimal and precise)

Update to latest dev version 

### Zigbee2MQTT version

1.33.2-dev commit: 1fd1686

### Adapter firmware version

20221226

### Adapter

cc2652

Koenkk, 2023-11-06:
@Laxilef could you look into this?

Laxilef, 2023-11-07:
@Koenkk looks like I was wrong again. Could you please check this? I can't test this locally right now. Thank you and sorry for my mistake.
```
json_attributes_template: `{"in_progress": {{ iif(value_json['update']['state'] == 'updating', 1, 0) }}}`,
```

Koenkk, 2023-11-08:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-11-08 by Koenkk

xripper, 2023-11-09:
Tested just now, unfortunatelly OTA update still stalling in HA updates.

pbranly, 2023-11-10:
Hi I use the very last dev version and I still receive hundreds of those messages in Home Assistant 
Phil

mundschenk-at, 2023-11-10:
@Koenkk The `to_json` filter needs an Python object as its input (and it is spelled `to_json`, not `tojson`), so this will not work. 

Changing the `json_attribute_template` to `{'in_progress': {{ iif(value_json['update']['state'] == 'updating', 'true', 'false') }} }` should do the trick.

Koenkk, 2023-11-10:
@mundschenk-at many thanks for the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

mundschenk-at, 2023-11-11:
@Koenkk I am sorry, but of course I managed to pick the wrong quotes for JSON. I'll make new PR to fix it ASAP.

Thessa, 2023-11-19:
In the addon...

Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:553
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 11:33:42 (1 occurrences)
Last logged: 11:33:42
Erroneous JSON: {'in_progress': false }

Laxilef, 2023-11-20:
@Thessa please use the Dev version

Thessa, 2024-01-15:
Hi,
Still have ths message
Erroneous JSON: {'in_progress': false }message first occurred at 21:18:15 and shows up 11 times
Greetings, met een vriendelijke groet,
 Ben de Boer,

ATTENTION:

The information in this electronic mail message is private and confidential, and only intended for the addressee. Should you receive this message by mistake, you are hereby notified that any disclosure, reproduction, distribution or use of this message is strictly prohibited. Please inform the sender by reply transmission and delete the message without copying or opening it. 

Messages and attachments are scanned for all viruses known. If this message contains password-protected attachments, the files have NOT been scanned for viruses. Always scan attachments before opening them. 

 
 That's what I'm using...