Koenkk/zigbee2mqtt #19598: [New device support]: Tuya ZigBee ZMB9-Z-4P 3-phase circuit breaker/power meter [TS0601 _TZE200_abatw3kj]

xardas-eu, 2023-11-06:
### Link

https://allegro.pl/oferta/bezpiecznik-nadpradowy-zigbee-4p-3-fazy-tuya-1-63a-12422973030

### Database entry

{"id":50,"type":"Router","ieeeAddr":"0xa4c138d758167c4f","nwkAddr":52651,"manufId":4417,"manufName":"_TZE200_abatw3kj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���,i���,i���,i���,i���,i���,i\b��,i","65506":54,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_abatw3kj","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1699287930807,"defaultSendRequestWhen":"immediate"}

### Comments

There was a ticket about this device already but it got no traction :( https://github.com/Koenkk/zigbee2mqtt/issues/14859

Creating this issue hoping that somebody has figured the device out (especially how to read per-phase power readings, both instant (W) and total (kWh) 

The converter I use for testing produces this debug output:

2023-11-06 17:33:09zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":61187,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0]}}]} VALUE = ETU1-IOT-Zigbee
2023-11-06 17:33:09zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":61187,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0]}}]} VALUE = ETU1-IOT-Zigbee
2023-11-06 17:33:27zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":61955,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:27zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":61955,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:27zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":61955,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:27zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":61955,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:27zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":62467,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,100,3,1,1,19,4,1,0,175]}}]} VALUE = d�
2023-11-06 17:33:27zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":62467,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,100,3,1,1,19,4,1,0,175]}}]} VALUE = d�
2023-11-06 17:33:28zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":62723,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0]}}]} VALUE = ETU1-IOT-Zigbee
2023-11-06 17:33:28zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":62723,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0]}}]} VALUE = ETU1-IOT-Zigbee
2023-11-06 17:33:45zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":63491,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:45zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":63491,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:45zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":63491,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:45zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":63491,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:33:45zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":64003,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,100,3,1,1,19,4,1,0,175]}}]} VALUE = d�
2023-11-06 17:33:45zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":64003,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,100,3,1,1,19,4,1,0,175]}}]} VALUE = d�
2023-11-06 17:33:46zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":64259,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0]}}]} VALUE = ETU1-IOT-Zigbee
2023-11-06 17:34:03zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":65027,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:34:03zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":65027,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:34:03zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":65027,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:34:03zigbee-herdsman-converters:: NOT RECOGNIZED DP #9 with data {"seq":65027,"dpValues":[{"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}]} VALUE = 0
2023-11-06 17:34:03zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":4,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,100,3,1,1,19,4,1,0,175]}}]} VALUE = d�
2023-11-06 17:34:03zigbee-herdsman-converters:: NOT RECOGNIZED DP #18 with data {"seq":4,"dpValues":[{"dp":18,"datatype":0,"data":{"type":"Buffer","data":[1,1,0,100,3,1,1,19,4,1,0,175]}}]} VALUE = d�
2023-11-06 17:34:04zigbee-herdsman-converters:: NOT RECOGNIZED DP #19 with data {"seq":260,"dpValues":[{"dp":19,"datatype":3,"data":{"type":"Buffer","data":[69,84,85,49,45,73,79,84,45,90,105,103,98,101,101,0]}}]} VALUE = ETU1-IOT-Zigbee

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const fzLocal = {
    tuya_dinrail_switch: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = legacy.getDataValue(dpValue);
                const dp = dpValue.dp
                meta.logger.info(`RECEIVED DP #${dp} -- VALUE = ${value}`);

                switch (dp) {
                    case 16: // DPID that we added to common
                        return {
                            state: value ? 'ON' : 'OFF'
                        };
                    case 1:
[... 34 lines cut ...]
        },
    },

};

const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE200_abatw3kj'
    }],
    model: 'ZMB9-Z-4P',
    vendor: 'GOMEDIA',
    extend: extend.switch(),
    description: 'Circuit breaker 4P with meter',
    fromZigbee: [fzLocal.tuya_dinrail_switch, ],
    toZigbee: [tzLocal.state],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
};

module.exports = definition;
```

clumsy-stefan, 2024-03-06:
Anyone made any progress on this? I have a similar device which identifies as `_TZE200_wbhaespm`.

Any help would be gratly appreciated!

xardas-eu, 2024-03-10:
Yeah I figured it out but I don't have good news. This device is just not
capable of metering the used energy, nor does it report the electric current

clumsy-stefan, 2024-03-10:
I did some more debugging and received the manufacturer documentation (in chinese, machine translated to english). I succeeded in creating a template, with all the DP's, see details [here](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7181)

xardas-eu, 2024-03-10:
I was able to do the similar thing for my device but the reported data
seemed trash and there was no incremental count anywhere. The "current" was
forever stuck at 1A and only voltage on the 1st phase seemed to be correct.
Does your device produce more believable data?

clumsy-stefan, 2024-03-10:
I guess so yes, but it's only sent about once an hour for the three phases and once a day for the temperature... only DP's 1,9,16,19 are sent regularly (Energy, Alarm, Switch State, Serial-Nr).

I'm waiting for a tuya GW to connect the breaker to and sniff the data/change the reporting interval (which according to the documentation should be possible)...

xardas-eu, 2024-03-10:
Hmm, your device may be more capable than mine. Care to share a link where
you got it?

clumsy-stefan, 2024-03-10:
Got it from Aliexpress: https://de.aliexpress.com/item/1005005749883633.html
 I have three 2P and one 4P. Only the 4P in use currently...

xardas-eu, 2024-03-10:
Sweet! Looks like yours is indeed more capable. The one I got does not
mention power metering specifically, I just assumed it would have it when I
bought it :D

I'll be getting that unit, many thanks!

clumsy-stefan, 2024-03-21:
I made some progress on mine, especially on the fault-results and some of the functions. Still trying to make the alarm-settings available in z2m, but some more debugging needed. Let me know if you need the latest version...

xardas-eu, 2024-03-22:
@clumsy-stefan  how does your energy usage data look like? is it believable? my only purpose for this device is to track the energy and I held off of buying it yet because I wanna make sure I don't waste my money. thanks!

clumsy-stefan, 2024-03-22:
Currently yes, I think so, but I have little to nothing attached behind currently as I use it as a switch and leakage protection for special purposes. For this I still need some time to verify the readings...

I also have the [`TS011F_ TZ3000_lepzuhto`](https://de.aliexpress.com/item/1005006029725329.html) in use (single phase) which works very well for exact this purpose (measurement and switching)...

However I made some progress on the RCBO again last night, by now I can set all the different triggers as in the original tuya app and get all values/alarms back. See [here](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7181).

xardas-eu, 2024-03-22:
Thanks for quick response! I pulled the trigger and ordered it, should get it early April :) Could you share your latest quirk? I am a developer (not too familiar with z2m ecosystem though) so I may be able to contribute in some way when I get the device.

clumsy-stefan, 2024-03-22:
see [above mentioned issue 7181](https://github.com/Koenkk/zigbee-herdsman-converters/issues/7181), the complete running template is in there, just save it to a .js file and include it in configuration.yaml!

xardas-eu, 2024-04-10:
@clumsy-stefan  I got the device and connected it. Happy to report that it (and your converted) works as expected, although on  z2m 1.33.0 I had to replace tuya with legacy.sendDataPointRaw

and on more recent version (1.36.0) it does not pick up the converter at all. do you know why might that be?

still, many thanks!

clumsy-stefan, 2024-04-10:
I'm running 1.36.0 commit: 2a53b8e5 without issues... can't say why...

xardas-eu, 2024-04-10:
Weird..  is the version posted here - https://github.com/Koenkk/zigbee-herdsman-converters/issues/7181 - the most recent or were there any changes since?

closed 2024-11-07 by github-actions[bot]