Koenkk/zigbee2mqtt #19599: Samotech SM323 v2 needs power metering

triblondon, 2023-11-06:
### What happened?

I bought a Samotech SM323 v2 dimmer switch, hooked it up to my zigbee network and it was [correctly recognised by Z2M](https://www.zigbee2mqtt.io/devices/SM323.html).  The exposes tab lists `state`, `brightness`, `power_on_behavior`, `effect`, and `linkquality`.   In HA, these map to the following entities:

[image]

### What did you expect to happen?

The [Samotech product page](https://www.samotech.co.uk/products/zigbee-dimmer-switch-hue-compatible/) for the new v2 ("new for 2024") version states that it has power metering, and when I asked them for support they referred me to a screenshot of a test Home assistant instance where the device was added with ZHA, where it does indeed appear to expose power metering entities:

[image]

So I expected to see power metering on the 'exposes' tab of Z2M, and I expected to see power metering entities on the HA device screen.

### How to reproduce it (minimal and precise)

Buy an SM323 v2, connect to Z2M.  Observe the 'Exposes' tab.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20210120

### Adapter

https://slae.sh/projects/cc2652/

djkemp, 2023-11-23:
Very interested to see this implemented also as about to pull the trigger and purchase a few of these.

Koenkk, 2023-11-23:
could you provide the data/database.db entry of this device?

triblondon, 2023-11-27:
@Koenkk can you refer me to instructions for how to do that?

Koenkk, 2023-11-27:
you can find it next to your z2m configuration.yaml

Koenkk, 2023-12-13:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/0d46ad10a98424c8c839094fadcd0e7e): 

- make sure you are using z2m 1.34.0
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

mpartington, 2023-12-13:
Failed to restart, maybe I've done something wrong?

SyntaxError: Identifier 'tuya' has already been declared
    at new Script (node:vm:100:7)
    at createScript (node:vm:265:10)
    at Object.runInNewContext (node:vm:306:10)
    at loadModuleFromText (/app/lib/util/utils.ts:154:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:161:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:171:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

Koenkk, 2023-12-13:
whoops, fixed the ext converter.

mpartington, 2023-12-13:
some different errors

[19:33:50] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:21
    extend: [light({configureReporting: true}), electricityMeter()],
            ^

ReferenceError: light is not defined
    at /app/data/extension/externally-loaded.js:21:13
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:154:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:161:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:171:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

Koenkk, 2023-12-13:
are you using z2m 1.34.0?

mpartington, 2023-12-13:
[image]
I think so, this is from the add on section

Koenkk, 2023-12-13:
Updated https://gist.github.com/Koenkk/0d46ad10a98424c8c839094fadcd0e7e, can you try again?

mpartington, 2023-12-13:
Looks good this time thanks, although it probably doesn't need the colour setting

[image]
[image]

mpartington, 2023-12-14:
Working well, but am getting these errors. Imagine it's something simple, but outside my knowledge how to fix
[image]

Koenkk, 2023-12-15:
> Another observation is the energy field doesn't seem correct (I haven't used 11kWh today on a 20W light)

This is the total consumption of that plug, not of today.

mpartington, 2023-12-16:
Thanks. It was a brand new switch, just installed, so not sure where it's got the consumption from. Not a major issue and can work around it

Koenkk, 2023-12-17:
Great, I've integrated the changes.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-17 by Koenkk

viharm, 2024-01-08:
Hi @Koenkk, thanks. I am using this device in HA (via the Z2M add-on) through the external converter you have linked above. Since this has been merged, can we remove the relevant external converter configuration if this is part of the release?

Also, would these power/energy related entities be exposed and available within HA?

Koenkk, 2024-01-09:
You can remove the external converter, electrical measurements entities will be available in HA

james-fry, 2024-01-13:
I have ~10 of the old version of the SM323 without power measurement capability.
I'm seeing errors in log about haElectricalMeasurement now:
`Zigbee2MQTT:error 2024-01-13 23:41:48: Failed to configure 'Kitchen Island Plinth Strips', attempt 1 (Error: Device 0xf4b3b1fffef5f8d6 has no input cluster haElectricalMeasurement
    at getEndpointsWithInputCluster (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:18:15)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:196:36)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:31)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))`
Is this related?

viharm, 2024-01-14:
> You can remove the external converter, electrical measurements entities will be available in HA

I can't seem to find these entities/attributes in HA. I've checked the developer tools and these are not exposed. Am I doing something wrong? I'm happy to open a new issues, but trying to prevent additional admin if it's not needed.

mpartington, 2024-01-14:
@viharm, you will need to enable them in the MQTT device. They seem to be turned off by default.

[image]

viharm, 2024-01-14:
> @viharm, you will need to enable them in the MQTT device. They seem to be turned off by default.
> 
> 

Thanks @mpartington. However the diagnostic panel is merely read-only, and there are no configurable settings to "enable" reading the power metrics.

Fig.1 MQTT Device screen
[image]

Fig. 2 Z2M device exposed metrics
[image]

I may have to unpair and repair the device, which I was trying to avoid, as this will require opening the wall face plate.

Are there any other alternatives?

mpartington, 2024-01-14:
Maybe ive got the wrong end of the stick, but I just opened up the diagnostic entity and 'enabled' it. This created an entity that can be monitored for me. I just enabled voltage and current this morning seeing this.
[image]

mpartington, 2024-01-14:
[image]

[image]

mpartington, 2024-01-14:
[image]

mpartington, 2024-01-14:
[image]

mpartington, 2024-01-14:
[image]

Then
[image]

viharm, 2024-01-14:
Yes! That's it! I'm sorry I'm still quite new to HA, so didn't realise that clicking on the disable entities from the dashboard view and drilling through to the "cog" icon allowed enabling. Now I realise what you meant in your [earlier comment](https://github.com/Koenkk/zigbee2mqtt/issues/19599#issuecomment-1890872754) when you said they are "turned off" by default. Thanks for your patience with me.

viharm, 2024-01-14:
I've enabled it, and am waiting for the values to come through over the next few minutes.

viharm, 2024-01-14:
Success! Thanks so much.

[image]