Koenkk/zigbee2mqtt #19603: Frient/Develco SIRZB-110 Sirene can't turn off after triggered

kasperhoejbjerg, 2023-11-06:
### What happened?

Frient/Develco SIRZB-110 Sirene do not turn off after started from the Z2M "exposes" menu. 

### What did you expect to happen?

It is expected to work acc. to the documentation: [https://www.zigbee2mqtt.io/devices/SIRZB-110.html](url)
_"Can be set by publishing to zigbee2mqtt/FRIENDLY_NAME/set with payload {"alarm": "START"} and {"alarm": "OFF"}"_

### How to reproduce it (minimal and precise)

1. Follow the menu path Zigbee2MQTT --> Devices 
2. Select Friendly name "SIRZB-110 Sirene" 
3. Select the tab "exposes"
4. scroll to the Alarm (Manual Start of Siren) section
5. Click "START"
6. Click "OFF"

### Zigbee2MQTT version

1.33.2-1

### Adapter firmware version

20231030.1

### Adapter

Nabu Casa SKYconnect

### Debug log

[image]

Aeliassen, 2024-01-23:
I'm experiencing this problem on my SIRZB-110 with firmware 1.9.3 - what firmware are you running on your device?

kasperhoejbjerg, 2024-01-24:
I'm on firmware 1.9.3 as well.
 
[image]

Aeliassen, 2024-01-24:
I have the firmware file for the frient SIRZB-110 version 1.7.1 - does anyone know how to downgrade this with Z2M? If so it might be a temporary fix to downgrade the firmware and then be able to use the device until this is fixed.

kasperhoejbjerg, 2024-01-24:
I think it is possible to downgrade it. You might want to look at this: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
Could you share the firmware file? :-)

Aeliassen, 2024-01-24:
[Router - Siren - 1.7.1.zip](https://github.com/Koenkk/zigbee2mqtt/files/14036476/Router.-.Siren.-.1.7.1.zip)
This is the file i recived - I will not be able to test it for a couple of days, so please don't hold me responsible if it brakes something ;) 

Please let me know if you get it downgraded

Aeliassen, 2024-01-25:
> I think it is possible to downgrade it. You might want to look at this: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files Could you share the firmware file? :-)

Did you try to install the old firmware? I have not had time to look into this yet.

LMQTLV, 2024-03-16:
Hello,
I have exactly the same problem with a Frient one also with firmware 1.9.3 :-(
Does anyone know a way to downgrade it to 1.7.1 please ?

martinjaeger95, 2024-07-07:
Hi,
I'm having the same issue. Did anyone find a solution yet?

closed 2025-01-23 by github-actions[bot]

CannyGent, 2025-05-09:
Hi - did anybody find a way to downgrade this firmware? I have been trying using the firmware posted above on zigbee2mqtt but no luck so far