Koenkk/zigbee2mqtt #19605: [New device support]: Roller Shade Motor ZM25R1

gregolin, 2023-11-06:
### Link

https://www.zemismart.com/products/zm25r1

### Database entry

{"id":47,"type":"Router","ieeeAddr":"0x142d41fffe1b39d1","nwkAddr":41770,"manufId":4098,"manufName":"_TZE200_1vxgqfba","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_1vxgqfba","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1699306161967,"defaultSendRequestWhen":"immediate"}

### Comments

I've tried to use another external converter (for ex: https://www.zigbee2mqtt.io/devices/ZM25RX-08_30.html), but no success yet (I don't have experience with new external converters, so it's possible I'm doing some mistake).

### External converter

```shell
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
        modelID: 'TS0601',
        manufacturerName: '_TZE200_1vxgqfba'
        },
    ],
    model: 'ZM25R1/2',
    vendor: 'Zemismart',
    description: 'Tubular Motor',
    fromZigbee: [legacy.fromZigbee.tuya_cover],
    toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
};

module.exports = definition;
```

vinta7, 2023-12-03:
Hi.

After googling around I created this working external converter, hope you can use it.

Regards

```
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'upper_stroke_limit', 'middle_stroke_limit', 'lower_stroke_limit']
}

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE200_1vxgqfba'
},
],
model: 'ZM25R1',
vendor: 'Zemismart',
description: 'Tubular Motor',
fromZigbee: [legacy.fromZigbee.tuya_cover],
toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options, tzDatapoints],
exposes: [e.cover_position().setAccess('position', ea.STATE_SET),
          e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the upper stroke limit'),
          e.enum('middle__stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the middle stroke limit'),
          e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the lower stroke limit'),
],

    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
            [104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
            [105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})]
        ],
    }
};

module.exports = definition;
```

roverell, 2024-01-05:
This worked for me! You're a legend, thanks @vinta7, I'd all but given up getting it working.

Koenkk, 2024-01-05:
@vinta7 would you mind making a pull request for out-of-the-box support?

vinta7, 2024-01-05:
Thanks for the positive feedback @roverell , there was a small typo "e.enum('middle__stroke_limit'", below is the edited version, I need some help getting this code into the next version of zigbee2mqtt @Koenkk , still a newbie regarding github :-(

```
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'upper_stroke_limit', 'middle_stroke_limit', 'lower_stroke_limit']
}

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE200_1vxgqfba'
},
],
model: 'ZM25R1',
vendor: 'Zemismart',
description: 'Tubular Motor',
fromZigbee: [legacy.fromZigbee.tuya_cover],
toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options, tzDatapoints],
exposes: [e.cover_position().setAccess('position', ea.STATE_SET),
          e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the upper stroke limit'),
          e.enum('middle_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the middle stroke limit'),
          e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the lower stroke limit'),
],

    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
            [104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
            [105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})]
        ],
    }
};

module.exports = definition;
``` 

[image]

closed 2024-01-05 by Koenkk

kdober, 2024-08-01:
Hi
Probably of topic, but it is the only reference I found on google for this device in Github. 
Is there any way this code can be used to make a 'quirk' of this device on ZHA? If any can point out what to read on how to do this will be highly appreciated. Thanks