Koenkk/zigbee2mqtt #19607: Philips Hue White and Color Ambiance GU10 8718696485880 genOnOff not mapped to OnOff

Davst, 2023-11-06:
### What happened?

Trying to set reporting for a few bulbs.

the OnOff selection in the UI fails with a timeout.

The bulb reports OnOff to be available under genOnOff

"genOnOff":{"attributes":{"onOff":1}}"

### What did you expect to happen?

Bulb accepting the reporting setting for OnOff similar to other bulbs

### How to reproduce it (minimal and precise)

Go to reporting, select endpoint 11, OnOff, OnOff and any setting

### Zigbee2MQTT version

1.33.0 commit: 7ee207f

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

### Debug log

Error 2023-11-07 00:15:27Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x0017880104a2b7a4/11 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 35998 - 11 - 234 - 6 - 7 after 10000ms)'

Davst, 2023-11-07:
So.. two things, firstly.. wouldn't it be interesting to have the OnOff support anyways? 

Secondly.. for what I'm trying to achieve:
That's a shame. I wonder how deCONZ solved this.. recently moved to z2m from deCONZ and this is one of the main differences, that the system isn't aware of changes in lights.

I don't doubt you.. just curious how deCONZ managed to track this.

Koenkk, 2023-11-07:
> So.. two things, firstly.. wouldn't it be interesting to have the OnOff support anyways?

What do you mean with this? OnOff can be controlled but reporting isn't supported. I guess Deconz does polling.

closed 2024-06-05 by github-actions[bot]