Koenkk/zigbee2mqtt #19610: [New device support]:

Ajhomepn, 2023-11-07:
### Link

https://www.leroymerlin.pt/produtos/iluminacao/focos-de-iluminacao/focos-led/focos-de-superficie/painel-led-inspire-gdansk-enki-5400lm-84845509.html?src=clk

### Database entry

Where can i get it ? 

### Comments

Hi, 

My deeply apologies, i'm aware that i'm not following the right path asking for support where. 

I have a light panel from Adeo , the ZBEK-29. I can connect it to zigbee2mqtt but it shows as "Not supported".
[image]

I have a similar product , the Adeo ZBEK-27 and it work's perfectly.

I understand if you can't support due to the fact that I have not done the initial step's of the https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html; there is any video to support the guys that are not soo tech savvy ? 

Thank you for your time,

Koenkk, 2023-11-07:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-11-07 by Koenkk

Ajhomepn, 2023-11-07:
😮😮 so fast ?? Amaizing !!

will this be available on the standard version on the future ?

Koenkk, 2023-11-08:
yes, nest release is on 1 December