Koenkk/zigbee2mqtt #19618: Philips 915005997001 - Update failed

eikowagenknecht, 2023-11-07:
### What happened?

Today a new firmware version showed up. Currently the device is on [1.104.2](https://www.philips-hue.com/en-us/support/release-notes/lamps). Clicking the update button results in the following message almost immediately:

`Error 2023-11-07 20:45:54Update of 'EG.Essen Hue Decke' failed (Update failed with reason: 'aborted by device')`

This happens for multiple devices of this type.

Other Philips Hue device types (eg [9290012573A](https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a) ) that also received new firmware versions today seem to upgrade fine.

### What did you expect to happen?

Update runs through without errors

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

 20230507

### Adapter

Sonoff ZBDongle-P

shanelord01, 2023-11-07:
I’ve got the same issue across a few hue devices. Some definitely have an update available (as their peers already updated a few weeks ago).

AbeltjeNL, 2023-11-07:
Same behavior here. Update available and after a view minutes the update was available again but unable to install. After a restart of Z2M there is no update available anymore. Has it been revoked?

thk-socal, 2023-11-08:
If you run a "Check All" the update should disappear. My bet is that it has been pulled and is no longer available.

eikowagenknecht, 2023-11-08:
The updates seem to have been retracted for some more devices... [8718699703424](https://www.zigbee2mqtt.io/devices/8718699703424.html#philips-8718699703424) also was available until ~2 hours ago, failed and is now not available any more.

eikowagenknecht, 2023-11-10:
This is getting more confusing:

Upon using 'Check all' today, I noticed that some updates disappeared from the list. But here's the twist: updates show up for some devices of the same model but not for others. "Check all" and restarting doesn't change this.

Oddly enough, if I try updating one device (and it fails), *then* hit 'Check all', it seems to catch the update on the second round. Makes me wonder, are the missing updates a glitch, or is a partial update tricking the system into thinking it's already got the latest version?

Here's what it looks like for two identical devices – one with an update and one without:

[image]

Is there a reason 'Check all' isn't consistent, or am I missing something?

u20p17, 2023-11-21:
+1 
i also get the update available merssage but update is not working... any solutioin for this?

thk-socal, 2023-12-25:
Just installed some MR16 lights that are running a newer version of the firmware than is available from OTA.

Here are the details:

    "update": {
        "installed_version": 16785170,
        "latest_version": 16785162,
        "state": "idle"
     }

As you can see, the installed version is newer than what can be currently download.

closed 2024-07-23 by github-actions[bot]