Koenkk/zigbee2mqtt #19624: [New device support]: TRV602 but in zigbee2mqtt TRV601

stef1725, 2023-11-08:
### Link

https://de.aliexpress.com/item/1005004732463260.html?spm=a2g0o.productlist.main.21.1a87qR9mqR9moP&algo_pvid=bfd23009-be6d-43d8-a580-24af823eb127&algo_exp_id=bfd23009-be6d-43d8-a580-24af823eb127-10&pdp_npi=4%40dis%21CHF%2168.21%2125.92%21%21%2174.28%21%21%402103244816994615613127468e2a4a%2112000030272377604%21sea%21CH%21909578594%21&curPageLogUid=gzMa7NBxVBcR

### Database entry

How do I reach this path?

### Comments

I bought a smart radio valve and now wanted to integrate it into my zigbee2mqtt. But I'm confused because 1. it doesn't work because the valve is not known and on the other hand the valve is recognized as a TRV601, but it is a TRV602. How can I help or what am I doing wrong? Why isn't it recognized correctly?

gvlad01, 2023-11-09:
Hello, I have the same valves and they are also recognized as TRV601, but the valves work for me

stef1725, 2023-11-12:
@gvlad01 
I have set the valve to pairing mode as prescribed and the following message is then displayed:
 
Device '0xa4c13805b4c66944' with Zigbee model 'TS0601' and manufacturer name '_TZE200_rtrmfadk' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

But this is not correct as it is a TRV602.

gvlad01, 2023-11-12:
All settings are recognized and working for me

Gerätename
CR HR
Beschreibung
Zuletzt gesehen
N/A
Verfügbarkeit
[Deaktiviert](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Geräte-Typ
EndDevice
Zigbee-Modell
TS0601
Zigbee-Hersteller
_TZE200_rtrmfadk
Beschreibung
Thermostatic radiator valve
Unterstützungsstatus
Unterstützt

IEEE-Addresse
0xa4c138f4d9c55680
Netzwerk-Adresse
0xBFD2
Hersteller
[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Modell
[TS0601_thermostat_1](https://www.zigbee2mqtt.io/devices/TS0601_thermostat_1.html#tuya-ts0601_thermostat_1)
Spannungsversorgung
Batterie 100%
Interview erfolgreich
Trifft zu

stef1725, 2023-11-12:
This is what I want to achieve, I just don't know how to convince the system to recognize the correct type?
Do I have to enter this in the TUYA converter?

It is not listed in the list of supported devices.

gvlad01, 2023-11-12:
Try resetting the valve, TS0601 and TS0602 are the same devices, they just have different displays, I didn't enter anything in the converter, the valves were immediately recognized by zigbee2mqtt

stef1725, 2023-11-14:
No matter how I try to integrate the part, it is always recognized with the type TS0601 and _TZE200_rtrmfadk.

gvlad01, 2023-11-14:
Then I don't know, sometimes you have to reset devices several times until they are recognized, or restart Homeassistant

mrslvd, 2023-12-04:
I have the same device. It is recognized as  "TS0601_thermostat_1".
Unfortunately the device is not reporting the valve position or it is reported as NULL.
The other problem I have is that by default (according to device user manual that comes with it) the device is acting as simple ON/OFF device ( "Energy-saving mode" as stated in user manual ). To change the mode to use PID control ( "Comfort mode" as stated in user manual) I have to use TuYa gateway and android App. Obviously I don't use TuYa gatway and App. I don't find a way to change this using z2m. It will be great to find a way solving this with z2m.

mrslvd, 2023-12-05:
> So I ordered some more, apparently there are different software for TRV602, the _TZE200_rtrmfadk really doesn't show the valve position, the _TZE200_z1tyspqw shows the valve position

What about the default mode? Are _TZE200_z1tyspqw by default ON/OFF or PID. And do you know what is the resolution of valve position ( 5%, 25% or other ) of _TZE200_z1tyspqw.

gvlad01, 2023-12-05:
_TZE200_z1tyspqw has PID and the resolution of the valve is in 1% steps
[image]

mrslvd, 2023-12-06:
Thanks for your answer. 
I was not clear enough with my question, I guess :(
_TZE200_rtrmfadk has PID but it is not enabled by default. To enable it you should use TuYa Gateway and App. I am using z2m and I don't have Tuya gateway.
Regarding the valve precision, I mean the step size of regulating. For example ON/OFF mode means from 0%-100% in one step.
When PID is enabled it should be something like 5%. I believe 1% step for PID TRV is not pointless.
If  _TZE200_z1tyspqw has PID enabled by default and the step size of PID is 5% (or something like this) could you share where I can buy  _TZE200_z1tyspqw from.

gvlad01, 2023-12-06:
_TZE200_z1tyspqw makes valve in 1% steps, see picture.
[image]

I bought it from the seller, but at the moment he only has WiFi thermostats:
https://vi.aliexpress.com/item/1005006166148746.html?spm=a2g0o.order_list.order_list_main.100.67eb1802esgv9o&gatewayAdapt=glo2vnm

I'll try the _TZE200_rtrmfadk on the Tuya Gateway later

mrslvd, 2023-12-21:
Hi,
I did some experiments with following this:

https://github.com/Koenkk/zigbee-herdsman-converters/issues/6482

    https://github.com/Koenkk/zigbee2mqtt/issues/18892
    I made some mix between posted converters and now I have both position and PID ( I still can't confirm it is PID )
    I can post my file but I am not sure whether I made things in the right way....

Position is not exposed on datapoint 102 but 108. So you have to change datapoint to 108.
Comfort/Eco modes are not implemented in standard converter so, external converter has to be used while this gets implemented in standard. It is not hard to make external converter following the steps from the links above. Personally I did combine two posted converters from the links. As a base I was using:
https://gist.github.com/Koenkk/3993885d51baf82ed334afe0781c7d55
and just modify datapoint position and add datapoint for comfort/eco mode as in:
https://github.com/Koenkk/zigbee2mqtt/issues/18892

closed 2024-07-19 by github-actions[bot]