Koenkk/zigbee2mqtt #19626: Frequency report Sonoff  SNZB-02D

evirci, 2023-11-08:
### What happened?

The device reporting the value with the long interval, to much long.
the value non change with 0,2°C.

### What did you expect to happen?

Refresh temperature and humidity value in a half our and with 0,1°C 

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20231030.1

### Adapter

Sonoff zigbee dongle plus 3.0

alexh3o, 2023-11-12:
Hi,
Did you try changing the reporting. I have Sonoff older version of this sensor and setting the reporting to desire works. In my screenshot reporting for temperature and humidity is done at a rate of 10 minutes if temperatures changes less than 0.2°C or 2 minutes otherwise.
[image]

KevinMckk, 2024-01-10:
Hello, same issue for me:

**What happened?**
The device reporting the value with long interval and don't respect the maximum_report_interval
But I don't know if the issue come from Zigbee2MQTT or from the Sonoff SNZB-02D firmware..

We can see happen on this graphic
[image]

**What did you expect to happen?**
Take in account the configuration `"maximum_report_interval": 1800,"`

**Zigbee2MQTT version**
1.35.1

**Adapter firmware version**
6.10.3.0 build 297

**Adapter**
ZBDongle-E (EFR32MG21)

**Debug log**

state.json
> 
        "0xf4b3b1fffe44ad58": {
            "date_code": "20230105",
            "definition": {
                "description": "Temperature and humidity sensor with screen",
                ...
                ...
                ...
                    "configured_reportings": [
                        {
                            "attribute": "measuredValue",
                            "cluster": "msTemperatureMeasurement",
                            "maximum_report_interval": 1800,
                            "minimum_report_interval": 5,
                            "reportable_change": 20
                        },
                ...
                ...
                ...
            "friendly_name": "Sensor Salon",
            "ieee_address": "0xf4b3b1fffe44ad58",
            "interview_completed": true,
            "interviewing": false,
            "manufacturer": "SONOFF",
            "model_id": "SNZB-02D",
            "network_address": 61200,
            "power_source": "Battery",
            "software_build_id": "1.0.9",
            "supported": true,
            "type": "EndDevice"
        },

>

czerepsky, 2024-01-22:
Same issue for me. Even when I changed reporting setting it dosent work properly

HWiese1980, 2024-07-11:
Same here. The settings seem to be ignored. There's also no feedback whatsoever if they have been accepted or not.

a1ex-ak, 2024-07-28:
I faced the same problem but I see the error

optimistic-optician, 2024-08-22:
I am facing the same issue with the SNZB-02D. It seems like it doesn't want to report any changes smaller than 0.2°C, even when I set "min_rep_change" to 10. Default is 100 (1°C), but it doesn't go below 0.2° Intervals. Maybe thats a limit that Sonoff set to save battery more agressively.

a1ex-ak, 2024-08-22:
> I am facing the same issue with the SNZB-02D. It seems like it doesn't want to report any changes smaller than 0.2°C, even when I set "min_rep_change" to 10. Default is 100 (1°C), but it doesn't go below 0.2° Intervals. Maybe thats a limit that Sonoff set to save battery more agressively.
> 
> Edit: [Issue 21786](https://github.com/Koenkk/zigbee2mqtt/issues/21786) The manufacturer states "Accuracy 0.2°C", obviously it would have been more precise language if they stated "Accuracy" an "Resolution" independently.

the solution is simple

[image]

[image]

optimistic-optician, 2024-08-22:
@a1ex-ak Have you tested this? I have tried it the same way and couldn't get the sensor to report higher resolution than 0.2. Also, If my understanding is correct, the precision and calibration are applied in z2m, not configured on the sensor itself. I tried higher values but it has no effect as far as I can tell

a1ex-ak, 2024-08-22:
> @a1ex-ak Have you tested this? I have tried it the same way and couldn't get the sensor to report higher resolution than 0.2. Also, If my understanding is correct, the precision and calibration are [...]

Yes, these are the settings that work for me and work very well with z2m

optimistic-optician, 2024-09-04:
> Yes, these are the settings that work for me and work very well with z2m

I tried it, unfortunately, cannot confirm. My SNZB-02 always report 0,2°C steps. It seems to be a limitation of that device, at least with my firmware version.

a1ex-ak, 2024-09-04:
> I tried it, unfortunately, cannot confirm. My SNZB-02 always report 0,2°C steps. It seems to be a limitation of that device, at least with my firmware version.

SNZB-02D or SNZB-02P no SNZB-02

optimistic-optician, 2024-09-04:
> SNZB-02D or SNZB-02P no SNZB-02

SNZB-02D, i forgot the letter.
I have all three Variants in my System. SNZB-02 (no-letter) works best and reports two decimals. Their dynamic message rate algorithm works best in my experience. They stay quiet if the change is below 0.1°C and react quickly if the temperature changes.

SNZB-02D (With the display) only has one decimal and seems to have a minium resolution of 0.2°C.

SNZB-02P (the round one) has a larger battery and claims to have a more energy efficient chipset. They are a mixed bag. They provide 0.1°C resolution (better than "D"), but their dynamic reporting algorithm is not as good as the original version. Whenever the least significant digit flutters between two values (for instance 23.1°C and 23.2°C) they send new messages at the maximum configured rate, jumping between the two. My guess is that it only works with one decimal resolution internally so it cannot oversample and apply any smoothing. Unfortunately the only way I could quiet them down while retaining the 0.1°C resolution was to set the minimum report interval quite high, 180s in my case.
D and P Variants claim to be equipped with this Sensor: [Sensirion SHT40](https://sensirion.com/media/documents/33FD6951/662A593A/HT_DS_Datasheet_SHT4x.pdf) it offers 0.01°C resolution. It should be a very accurate, long term stable sensor.

TLDR: original was best, D and P Variants claim to have superior sensor, but have weird quirks.

sebeichholz, 2024-10-19:
Thank you @optimistic-optician for your detailed explanations / report! 
I have been using multiple SNZB-02 sensors, and now I bought some SNZB-02D and your comment explains exactly what I discovered by myself :-(

closed 2025-01-11 by github-actions[bot]

KevinMckk, 2025-02-03:
How to change **Sonoff SNZB-02** reporting interval:

1. You need to press the button on Sonoff SNZB-02 for 1 second to wake up the device.
2. Valid reporting interval changes in Zigbee2MQTT.
3. You might need to press the button again after home assistant says it sent the message, supposedly to acknowledge the change.

**Reminder values:**
max rep interval: 300 → 5 minutes, 900 → 15 minutes and so on
temperature - Min rep change: 10 → 0.1C, 20 → 0.2C and so on (for **SNZB-02D** see sensor precisions post)
humidity - Min rep change: 100 → 1%, 200 → 2% and so on

**Sensor precision post:**
See _**optimistic-optician**_ post above
https://github.com/Koenkk/zigbee2mqtt/issues/19626#issuecomment-2328410743

rct, 2025-06-01:
This (closed) issue has some of the most helpful info in the last few comments on these 3 sensors

Was wondering if the reporting interval control or precision limitations have improved with the 2.3.0 firmware for the 02D display sensor that was released this year.  

Can’t find much useful change log details

maia, 2025-06-02:
I'd be interested if setting the update interval to less than 5 minutes works for the Sonoff SNZB-02WD, which was released in May 2025 as a successor to the SNZB-02D.