Koenkk/zigbee2mqtt #19627: Onesti product lock slow to lock door

GaliBatu, 2023-11-08:
### What happened?

I have implemented the  Onesti Products easyCodeTouch_v1 zigbee module in Nimly Pro Touch  and communication works, unlocking door works flawlessly immediately after button is pressed. However, whenever I press lock after it has been unlocked the command fails and it hangs for a long time. Eventually the lock will change to locked state, probably at some timeout.

### What did you expect to happen?

Locking door should be equally snappy as unlocking

### How to reproduce it (minimal and precise)

Unlock via zigbee then press lock

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20230506

### Adapter

Sonoff Zigbee 3.0 P

### Debug log

groupID 0
Error 2023-11-08 18:42:58Publish 'set' 'state' to 'Main door' failed: 'Error: Command 0xf4ce3668a5305ee5/11 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13353 - 11 - 67 - 257 - 1 after 10000ms)'
Debug 2023-11-08 18:42:58Error: Command 0xf4ce3668a5305ee5/11 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13353 - 11 - 67 - 257 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)

Koenkk, 2023-11-08:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/32fc43f36c9a0189dccbef8cdef9cf31

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

GaliBatu, 2023-11-08:
Unfortunately, it behaves exactly the same. I get a long wait until timeout when the lock finally locks

Koenkk, 2023-11-09:
- Please check if the ext converter is loaded correctly, I updated the [converter](https://gist.github.com/Koenkk/32fc43f36c9a0189dccbef8cdef9cf31), the description of the device in the z2m frontend should end with `CUSTOM` now.
- If that is the case, provide me the herdsman debug log of the issue

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Brumhilde, 2023-11-16:
I have the same problem. Also I cannot set sound volume and lock does not report state as it should (when unlocking with fingerprint/code). 

Connect module fw 20220614.

closed 2024-06-14 by github-actions[bot]