Koenkk/zigbee2mqtt #19632: Pairing Moes MS-105B issues

gaetan-steijaert, 2023-11-09:
### What happened?

When the MS-105B is put in pairing mode, Z2M doesn't find it.

same issue as: https://github.com/Koenkk/zigbee2mqtt/issues/18744
but this issue was closed without comments

### What did you expect to happen?

New devices should be detected.

### How to reproduce it (minimal and precise)

Start Zigbee2MQTT
Click on "detect devices"
Bring Moes MS-105B into pairing mode
--> Zigbee2MQTT did not detect the device

### Zigbee2MQTT version

1.33.2-1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle

Chartreusito, 2023-12-15:
Have you tried moving your device closer to the coordinator and pairing only to it ?

gaetan-steijaert, 2023-12-19:
Yes, tried every trick in the book! I gave up and switched to other modules.

closed 2024-07-17 by github-actions[bot]