Koenkk/zigbee2mqtt #19633: Enbrighten 43080 brightness state does not update when changed from physical switch

jakelauer, 2023-11-09:
### What happened?

I can set the brightness for this switch programmatically just fine, and it works great. If I physically hold the switch buttons though, which is supposed to set the brightness, nothing happens. No messages show up in MQTT Explorer and state does not change in Zigbee2MQTT

### What did you expect to happen?

I expected to see the brightness state change.

### How to reproduce it (minimal and precise)

Unsure - I suppose just install the light and try to change the brightness via the switch?

### Zigbee2MQTT version

1.33.2 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 ZBDongle-E

jakelauer, 2023-11-09:
I noticed the brightness does update when requested manually, as noted in the documentation. Is this not broadcasted when it changes?

Koenkk, 2023-11-09:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/6b5e3c32115c64e7986d0bb63b0318fe

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

jakelauer, 2023-11-09:
@Koenkk Confirmed, that works!

Koenkk, 2023-11-10:
Integrated the fix, you can remove the ext converter after updating to the next release.

closed 2023-11-10 by Koenkk