Koenkk/zigbee2mqtt #19636: TUYA MOLO white label Heiman HS1EB does not generate action when battery low.

RobLeighton67, 2023-11-09:
### What happened?

This button was added locally as an additional fingerprint to the device in Heiman.js.

modelId: TS0211 - manufacturerName: _TZ1800_rrv4u5fr

Button stopped generating actions when pressed - battery = 0 also reported at this time, indicating low battery.

### What did you expect to happen?

I would expect the button action to be sent even if battery was low.

### How to reproduce it (minimal and precise)

Add device and er, wait for low battery whilst pressing - could be  a while!

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20221226

### Adapter

Sonoff USB Dongle 3.0 (CC2652 version)

RobLeighton67, 2023-11-09:
[missing action button press.txt](https://github.com/Koenkk/zigbee2mqtt/files/13307771/missing.action.button.press.txt)

From the debug log it can be seen that the device sends a Zone Status of 9 (setting bit 3 for battery low) - I suspect this means that _**fz.command_status_change_notification_action**_ uses 9 instead of 1 to index in the action lookup?

(Replacing the battery restores action messages and sets battery = 100, obviously)

closed 2024-06-07 by github-actions[bot]

hwde, 2026-02-01:
Somehow I bought the same button, still reported as unknown, but the button seems to work. With the following external converter I am able to get the "single" button event, but no others. And the battery is reported to be 100, but I assume it must be less ... with my so many tests ;)

```
import fz from 'zigbee-herdsman-converters/converters/fromZigbee';
import * as exposes from 'zigbee-herdsman-converters/lib/exposes';
import tuya from 'zigbee-herdsman-converters/lib/tuya';
import pkg from 'zigbee-herdsman-converters/lib/modernExtend';
const {battery} = pkg;

const e = exposes.presets;

export default {
    fingerprint: tuya.fingerprint("TS0211", ["_TZ1800_rrv4u5fr"]),
    model: "HS1EB",
    vendor: "Heiman",
    description: "Smart emergency button",
    extend: [battery()], 
    fromZigbee: [fz.command_status_change_notification_action, fz.ignore_basic_report],
    toZigbee: [],
    exposes: [e.action(["off", "single", "double", "hold"])],
};
```

Maybe the logs help to do better:

```
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext [254,12,69,202,176,7,185,50,81,254,255,101,17,44,0,0,183]
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --> parsed 12 - 2 - 5 - 202 - [176,7,185,50,81,254,255,101,17,44,0,0] - 183
[2026-02-01 19:41:11] debug:    zh:zstack:znp: <-- AREQ: ZDO - tcDeviceInd
[2026-02-01 19:41:11] debug:    zh:controller: Device '0x2c1165fffe5132b9' joined
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext []
[2026-02-01 19:41:11] debug:    zh:controller: Device '0x2c1165fffe5132b9' accepted by handler
[2026-02-01 19:41:11] debug:    zh:controller: New device '0x2c1165fffe5132b9' joined
[2026-02-01 19:41:11] debug:    zh:controller: Creating device '0x2c1165fffe5132b9'
[2026-02-01 19:41:11] debug:    zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2026-02-01 19:41:11] info:     zh:controller: Interview for '0x2c1165fffe5132b9' started
[2026-02-01 19:41:11] debug:    zh:controller:device: Interview - start device '0x2c1165fffe5132b9'
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:writer: --> frame [254,4,37,2,176,7,176,7,35]
[2026-02-01 19:41:11] info:     z2m: Device '0x2c1165fffe5132b9' joined
[2026-02-01 19:41:11] info:     z2m: Starting interview of '0x2c1165fffe5132b9'
[2026-02-01 19:41:11] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x2c1165fffe5132b9","ieee_address":"0x2c1165fffe5132b9"},"type":"device_joined"}'
[2026-02-01 19:41:11] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x2c1165fffe5132b9","ieee_address":"0x2c1165fffe5132b9","status":"started"},"type":"device_interview"}'
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: <-- [254,1,101,2,0,102]
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,2,0,102]
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 2 - [0] - 102
[2026-02-01 19:41:11] debug:    zh:zstack:znp: <-- SRSP: ZDO - nodeDescReq
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext []
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: <-- [254,13,69,193,176,7,176,7,185,50,81,254,255,101,17,44,128,138,254,18,69,130,176,7,0,176,7,2,64,128,11,18,82,82,0,0,44,82,0,0,112]
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext [254,13,69,193,176,7,176,7,185,50,81,254,255,101,17,44,128,138,254,18,69,130,176,7,0,176,7,2,64,128,11,18,82,82,0,0,44,82,0,0,112]
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --> parsed 13 - 2 - 5 - 193 - [176,7,176,7,185,50,81,254,255,101,17,44,128] - 138
[2026-02-01 19:41:11] debug:    zh:zstack:znp: <-- AREQ: ZDO - endDeviceAnnceInd
[2026-02-01 19:41:11] debug:    zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":1968,"eui64":"0x2c1165fffe5132b9","capabilities":{"alternatePANCoordinator":0,"deviceType":0,"powerSource":0,"rxOnWhenIdle":0,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
[2026-02-01 19:41:11] debug:    zh:controller: Device announce from '0x2c1165fffe5132b9:1968'
[2026-02-01 19:41:11] debug:    z2m: Device '0x2c1165fffe5132b9' announced itself
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext [254,18,69,130,176,7,0,176,7,2,64,128,11,18,82,82,0,0,44,82,0,0,112]
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --> parsed 18 - 2 - 5 - 130 - [176,7,0,176,7,2,64,128,11,18,82,82,0,0,44,82,0,0] - 112
[2026-02-01 19:41:11] debug:    zh:zstack:znp: <-- AREQ: ZDO - nodeDescRsp
[2026-02-01 19:41:11] debug:    zh:controller: Received ZDO response: clusterId=NODE_DESCRIPTOR_RESPONSE, status=SUCCESS, payload={"nwkAddress":1968,"logicalType":2,"apsFlags":0,"frequencyBand":8,"capabilities":{"alternatePANCoordinator":0,"deviceType":0,"powerSource":0,"rxOnWhenIdle":0,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1},"manufacturerCode":4619,"maxBufSize":82,"maxIncTxSize":82,"serverMask":{"primaryTrustCenter":0,"backupTrustCenter":0,"deprecated1":0,"deprecated2":0,"deprecated3":0,"deprecated4":0,"networkManager":0,"reserved1":0,"reserved2":0,"stackComplianceRevision":22},"maxOutTxSize":82,"deprecated1":0,"tlvs":[]}
[2026-02-01 19:41:11] debug:    zh:zstack:unpi:parser: --- parseNext []
[... cut ...]
[2026-02-01 19:41:13] debug:    zh:controller:database: Writing database to '/opt/zigbee2mqtt/data/database.db'
[2026-02-01 19:41:13] debug:    zh:zstack:unpi:parser: --- parseNext []
[2026-02-01 19:41:13] debug:    z2m: Received Zigbee message from '0x2c1165fffe5132b9', type 'raw', cluster 'genBasic', data '{"data":[13,11,18,38,241,200,84,83,48,50,49,49,32,32,32,32,32,32,32,32,32,32,68,2,190,0,0,0,1,2,4,1,255,21,1,0,0,255,255,255,255,255,255,255],"type":"Buffer"}' from endpoint 1 with groupID 0
[2026-02-01 19:41:13] debug:    z2m: Skipping message, still interviewing
[2026-02-01 19:41:2
```