Koenkk/zigbee2mqtt #19645: Corrupt Database

stefan-west1, 2023-11-09:
### What happened?

After a random amount of time, devices will randomly stop responding to requests. This happens over weeks, and one by one they all stop. I can delete and re-add the device, and re-interview it, but I cannot control it. The only fix I've found is to delete the db file and re-add all devices. Or restore the db file to one before the devices began to fail. I haven't yet been able to compare the two db files to see if there's any difference. However this has happened 3 or 4 times now.,

### Zigbee2MQTT version

1.33.2 commit: 9996c93

### Adapter firmware version

 0x26780700

### Adapter

 ConBee2/RaspBee2

lux73, 2023-11-12:
let me guess: z2m is running on an sd-card like Rasperry?

stefan-west1, 2023-12-01:
> let me guess: z2m is running on an sd-card like Rasperry?

Nope. It’s running in a docker container on a Dell R730XD in Unraid

stefan-west1, 2023-12-01:
I think I've found the issue, but don't know how to fix it. I've just compared 2 identical devices, one which works and one which doesn't. The one which doesn't work has an extra endpoint (endpoint 0). I tried manually going into the database.db file and removing the extra endpoint, but as soon as i restart zigbee2mqtt it adds it back.

closed 2024-06-29 by github-actions[bot]