Koenkk/zigbee2mqtt #19648: Some devices offline

philippk75, 2023-11-10:
### What happened?

Recently it has been happening more and more frequently that some devices in Home Assiatnt with Zigbee2Mqtt are offline or do not respond

### Zigbee2MQTT version

1.33.2-dev 

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Skyconnect Zigbee Adapter

kizio1000, 2023-11-10:
same here, all after latest update

jsb2092, 2023-11-13:
Also having the same issue

suspended, 2023-11-13:
Same issue here. After latest updates.

aletzi1, 2023-11-14:
Same issue here

Lyric41CR, 2023-11-14:
Idem pour moi sous 1.33.2 les devices se déco sans raison

matidellatorre, 2023-11-21:
Same issue with last update

helgek, 2023-11-23:
Having a similar issue, randomly one of my devices would not respond to commands coming from Z2M, results in a timeout error:https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1824743866 - coordinator is the highly praised SLZB-0 (based on stable chipset CC2652P  recommended by Z2M, not the experimental EFR32MG21 chipset).

closed 2024-06-21 by github-actions[bot]