Koenkk/zigbee2mqtt #19651: Missing very useful Power-on behavior for IKEA LED1924G9, LED2004G8

Afraskai, 2023-11-10:
### What happened?

Hello guys!

I hope im right in here to ask you for help. I moved from ZHA to zigbee2zqtt cause of it´s bigger device support and many more advantages.

In ZHA the Lamps [LED1924G9](https://www.zigbee2mqtt.io/devices/LED1924G9.html) and [LED2004G8](https://www.zigbee2mqtt.io/devices/LED2004G8.html#ikea-led2004g8) had an very useful feature to set the brightness color/color temperature after a power loss, independed of which brightness and color it had befor the power loss. 

The lamps theirselves have this feature and in the forum i saw zigbee2mqtt can do this also with the hue devices.

Thank you so much for your support!

### What did you expect to happen?

After somebody hits the power switch the brightness, color or color temperature should have an default.

Brightness from 1-100%
Colortemperature (LED2004G8)
Color (LED1924G9)

### How to reproduce it (minimal and precise)

Reset the power and the lamp starts with 1% brightness or off.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

Wielebny666, 2023-11-14:
Try sending state with transition = 1

Wireheadbe, 2023-11-14:
You could do this by setting the "Power-On" behavior, and then via Dev Console -> Cluster: LevelControl -> Attribute: OnLevel.
OnLevel = 255 = Last Known Brightness
OnLevel = 1 = Lowest Brightness
OnLevel = 254 = Highest Brightness

-> OnLevel is used every time the light is turned on.

Afraskai, 2023-11-18:
> You could do this by setting the "Power-On" behavior, and then via Dev Console -> Cluster: LevelControl -> Attribute: OnLevel. OnLevel = 255 = Last Known Brightness OnLevel = 1 = Lowest Brightness OnLevel = 254 = Highest Brightness
> 
> -> OnLevel is used every time the light is turned on.

Hi @Wireheadbe - thanks for this info! This worked!

closed 2023-11-18 by Afraskai