Koenkk/zigbee2mqtt #19662: TS0601_smart_human_presence_sensor_1 - flooding network with data

kizio1000, 2023-11-10:
### What happened?

After pairing human presence sensor ( https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor_1.html ) network started to be unreliable, quick check in log  shown that new device is flooding with distance readings that change all the time. 

### What did you expect to happen?

That constant distance measuring can be limited or switched off

### How to reproduce it (minimal and precise)

Pairing T165 human presence sensor and check logs 
( https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor_1.html )

### Zigbee2MQTT version

version: 1.33.2-1

### Adapter firmware version

Coordinator revision 7.1.1.0 build 273

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

Koenkk, 2023-11-10:
AFAIK TuYa device don't support to change the reporting frequency, so unfortunately we cannot fix this from the z2m side.

kizio1000, 2023-11-13:
> AFAIK TuYa device don't support to change the reporting frequency, so unfortunately we cannot fix this from the z2m side.

Thx for update. Will investigate options for tasmotizing it in this case. :)

closed 2023-11-13 by kizio1000

TheOriginal92, 2024-08-09:
> > AFAIK TuYa device don't support to change the reporting frequency, so unfortunately we cannot fix this from the z2m side.
> 
> Thx for update. Will investigate options for tasmotizing it in this case. :)

Sorry to bring this up again. Did you manage to fix the reporting issue?