Koenkk/zigbee2mqtt #19667: [New device support]: ONENUO Tuya Smoke Sensor 228WZ

follow-suit, 2023-11-11:
### Link

https://www.aliexpress.com/item/1005006161521708.html

### Database entry

{"id":58,"type":"EndDevice","ieeeAddr":"0xa4c1383840cc5def","nwkAddr":52998,"manufId":4417,"manufName":"_TZE204_kgaxpvxr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��,\u0019��,\u0007\u0000\u0000\u0000\u0000\u0011>��,g","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_kgaxpvxr","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1699695128013,"defaultSendRequestWhen":"immediate"}

### Comments

Please find the suggested converter below. The vendor as per product page is called "ONENUO" from which I was observing an increasing amount of TuYa ZigBee products over the last few weeks. So it would possibly make sense to add them as a new vendor. The only model name I could find was "228WZ", which is specified both on the product page and the packaging.

The smoke sensor can be configured with a sensitivity level (low, medium, high), which I mapped to enum values 0-2 (just a guess). Since the device does not report the DP 102 value, and my tests were inconclusive, I have no idea if this is working correctly, but at least I get no error in the logs:

> Received MQTT message on 'zigbee2mqtt/0xa4c1383840cc5def/set' with data '{"sensitivity":"low"}'
> Publishing 'set' 'sensitivity' to '0xa4c1383840cc5def'
> MQTT publish: topic 'zigbee2mqtt/0xa4c1383840cc5def', payload '{"battery":100,"last_seen":"2023-11-11T09:39:18+00:00","linkquality":128,"self_test":true,"self_test_result":"success","sensitivity":"low","silence":false,"smoke":false}'

Also, pressing the test button on the device will send a boolean value on DP 101 (also see schema definitions below) which I mapped to 'success' and 'error' of 'self_test_result'. Although exposing selfTest() would be a better fit from a data type perspective, I think selfTestResult() is a better match, since the DP does not indicate an ongoing test, but instead reports a one-time result. That is, I tried exposing selfTest(), and the value will remain 'true' indefinitely after pressing the test button, since no "end of test" is reported. Also, the DP is read-only. Maybe it would also make sense to create a new enum that only contains 'success' and 'error'.

The remaining DPs should be good to go:

- The detection of smoke (and later the absence thereof) are correctly reported
- Battery % is reported
- Silencing during the alarm works, but the silence state does not seem to be reported by the device, so it is never reset. Maybe it can be a be actively read, but I lack the knowledge on how to do this. Any hints are welcome.

DP schema definitions from TuYa IoT platform:
```
{
    "result": [
        {
            "schema": {
                "accessType": "ro",
                "code": "smoke_sensor_state",
                "valueDesc": "{\"range\":[\"alarm\",\"normal\",\"detecting\",\"unknown\"]}",
                "valueType": "Enum",
                "langDict": {
                    "normal": "Normal",
                    "alarm": "Smoke Alarm",
                    "detecting": "Detecting",
                    "unknown": "Unknown"
                },
                "name": "Smoke Detection State",
                "remark": "smoke_sensor_state"
            },
            "functionSet": {
                "accessType": "ro",
                "code": "smoke_sensor_state",
                "valueDesc": "{\"range\":[\"alarm\",\"normal\",\"detecting\",\"unknown\"]}",
                "valueType": "Enum",
                "langDict": {
                    "normal": "Normal",
                    "alarm": "Smoke Alarm",
                    "detecting": "Detecting",
                    "unknown": "Unknown"
                },
                "name": "Smoke Detection State",
                "remark": "",
                "mainDP": "smoke_sensor_state"
            }
        },
        {
            "schema": {
                "accessType": "ro",
                "code": "battery_percentage",
                "valueDesc": "{\"min\":0,\"max\":100,\"step\":1,\"unit\":\"%\",\"scale\":0}",
                "valueType": "Integer",
                "langDict": {
                    "unit": "%"
                },
[... 122 lines cut ...]
    vendor: 'TuYa',
    description: 'Smoke detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.smoke(), 
        e.battery(), 
        tuya.exposes.silence(), 
        tuya.exposes.selfTestResult(), 
        e.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('Sensitivity'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'smoke', tuya.valueConverter.trueFalse0],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw],
            [101, 'self_test_result', tuya.valueConverterBasic.lookup({'failure': false, 'success': true})],
            [102, 'sensitivity', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2)})],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-11-11:
looks good, could you make a PR to add support for this device?

follow-suit, 2023-11-11:
@Koenkk I will give it a try. Do you want it in TuYa or a new vendor?

Koenkk, 2023-11-11:
You can create a new file, regarding smoke_state, rename to smoke and make it a true/false value.

follow-suit, 2023-11-11:
> You can create a new file, regarding smoke_state, rename to smoke and make it a true/false value.

@Koenkk Are you sure? As per TuYa platform, we have 4 different possible values. Are 'detecting' and 'unknown' irrelevant? If so, should these be mapped to 'false', i.e. no smoke detected?

Koenkk, 2023-11-11:
I see, I propose to create a custom value converter which return multiple vales, a smoke and smoke_state (look in lib/tuya.ts for examples)

follow-suit, 2023-11-11:
> I see, I propose to create a custom value converter which return multiple vales, a smoke and smoke_state (look in lib/tuya.ts for examples)

Just to be sure, do you mean something like [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/29653044f43b6c54f6a6c7c6ebab5ec45c05045a/src/lib/legacy.ts#L1365)?

Koenkk, 2023-11-11:
No, like [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/29653044f43b6c54f6a6c7c6ebab5ec45c05045a/src/lib/tuya.ts#L513)

follow-suit, 2023-11-11:
> No, like [this](https://github.com/Koenkk/zigbee-herdsman-converters/blob/29653044f43b6c54f6a6c7c6ebab5ec45c05045a/src/lib/tuya.ts#L513)

The following seems to be working. Can the value converter stay there or should it be moved to some type of lib?

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_kgaxpvxr']),
    model: '288WZ',
    vendor: 'ONENUO',
    description: 'Smoke detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.smoke(),
        e.enum('smoke_state', ea.STATE, ['alarm', 'normal', 'detecting', 'unknown']).withDescription('Smoke state'),
        e.battery(),
        tuya.exposes.silence(),
        tuya.exposes.selfTestResult(),
        e.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('Sensitivity'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, null, function (lookup) {
                return {
                    from: function (v) {
                        const smokeState = lookup.from(v);
                        return {
                            'smoke': (smokeState === 'alarm'),
                            'smoke_state': smokeState
                        };
                    }
                }
            }(tuya.valueConverterBasic.lookup({ 'alarm': tuya.enum(0), 'normal': tuya.enum(1), 'detecting': tuya.enum(2), 'unknown': tuya.enum(3) }))],
            //[1, 'smoke_state', tuya.valueConverterBasic.lookup({'alarm': tuya.enum(0), 'normal': tuya.enum(1), 'detecting': tuya.enum(2), 'unknown': tuya.enum(3)})],
            [15, 'battery', tuya.valueConverter.raw],
            [16, 'silence', tuya.valueConverter.raw],
            [101, 'self_test_result', tuya.valueConverterBasic.lookup({ 'failure': false, 'success': true })],
            [102, 'sensitivity', tuya.valueConverterBasic.lookup({ 'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2) })],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-11-12:
It can stay there, but please remove the `lookup` parameter, and do:

```js
const lookup = { 'alarm': tuya.enum(0), 'normal': tuya.enum(1), 'detecting': tuya.enum(2), 'unknown': tuya.enum(3) }
const smokeState = lookup.from(v);
```

Next please make a PR 😄

follow-suit, 2023-11-12:
@Koenkk I just created one.
The reason why I pass the lookup to the closure is so that it is initialized only once as opposed to each time the value converter is called. I thought that performance would carry some weight when it comes to smoke detection.

Also, I don't understand your code snippet, as `lookup` does not have a `from()` function in your example. And wrapping the map in a call to `tuya.valueConverterBasic.lookup` would bring us back to the redundant initialization topic. I just left my suggestion as is and tried my best to translate it to TS.

Koenkk, 2023-11-13:
I've merged your PR, assuming this can be closed now, thanks!

closed 2023-11-13 by Koenkk

szatan81, 2024-01-18:
@follow-suit could you tell me how you managed to pair this device?