Koenkk/zigbee2mqtt #19670: [New device support]: MINI-Z1-2CH / 3CH / 4CH

misha1183, 2023-11-11:
### Link

https://aliexpress.ru/item/1005005960005800.html?spm=a2g0o.order_list.order_list_main.9.21efa396093tVs&gatewayAdapt=glo2rus&sku_id=12000035050899856

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138665bc7bf01","nwkAddr":33849,"manufId":4417,"manufName":"_TZ3000_lmlsduws","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}},"genBasic":{"attributes":{"65503":"�P�,\u0012","65506":56,"65508":1,"65534":0,"appVersion":72,"manufacturerName":"_TZ3000_lmlsduws","zclVersion":3,"modelId":"TS0002","powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2eab91","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2eab91","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":381610487},"lastSeen":1699709647725,"defaultSendRequestWhen":"immediate"}

### Comments

it works, but:
switch_type ???
power on behavior L1 ???
power on behavior L2 ???
thanks

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    fingerprint: [{modelID: 'TS0002', manufacturerName: '_TZ3000_lmlsduws'}],
    model: 'MINI-Z1-2CH',
    vendor: 'TuYa',
    description: '2 gang switch',
    toZigbee: [tz.on_off, tuya.tz.power_on_behavior_1],
    fromZigbee: [tz.on_off, tuya.tz.power_on_behavior_1],
//    extend:  tuya.extend.switch({endpoints: ['l1', 'l2']}),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power_on_behavior()],
    whiteLabel: [{vendor: 'Mini-Z1', model: 'MINI-Z1-2CH'}],
    endpoint: (device) => {
        return {l1: 1, l2: 2};
        },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await device.getEndpoint(1).read('genOnOff', ['onOff']);
        await device.getEndpoint(2).read('genOnOff', ['onOff']);
        },
};

module.exports = definition;
```

asfyra, 2023-12-28:
@misha1183 I used the following, which is actually a copy of [this work ](https://github.com/Koenkk/zigbee2mqtt/issues/20255) from @kirush0280 and managed to get all features working:

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    fingerprint: [{
            modelID: 'TS0002',
            manufacturerName: '_TZ3000_lmlsduws'
        }
    ],
    model: 'TS0002_switch_module',
    vendor: 'TuYa',
    description: '2 gang switch module',
    extend: tuya.extend.switch({
        switchType: true,
        endpoints: ['l1', 'l2']
    }),

    endpoint: (device) => {
        return {
            'l1': 1,
            'l2': 2
        };
    },
    meta: {
        multiEndpoint: true
    },
    configure: async(device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;
```

Koenkk, 2023-12-28:
@asfyra added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-28 by Koenkk

Jahislove, 2025-01-06:
despite the fact that it was added one year ago, my mini-Z1-2CH (same as first picture) is still detected as tuya TS0002_basic
-m2z v2.0.0

did i miss something ?
[image]