Koenkk/zigbee2mqtt #19676: SonOFF ZBMini - Exposes only Linkquality but no state.

JKSniper1, 2023-11-11:
### What happened?

SonOFF ZBMini: There are 2 exposes in Zigbee2MQTT while only one ENTITY in HA. 

While I can see STATE in Zigbee2MQTT -> device -> Exposes and also toggle it (and it works), and I can also see "State" as 
"{
    "linkquality": 21,
    "state": "OFF"
}"

In HA -> Settings -> Devices -> Core-Mosquitto -> device -> I can only see "linkquality," but no state. Therefore I can't use it for automations, see it's state, change it etc. 

### What did you expect to happen?

I expected to see both linkquality and state in HA as entities. 

### How to reproduce it (minimal and precise)

Normal pair ZBMini to Zigbee2MQTT. I've tried deleting it, re-pairing, different options etc. etc. - nothing worked. 

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20210708

### Adapter

SonOff dongle

JKSniper1, 2023-11-14:
Aha, cought some error in log. Maybe someone can help me? 

Logger: homeassistant.util.logging
Source: util/logging.py:156
First occurred: November 11, 2023 at 10:13:36 AM (39 occurrences)
Last logged: 9:11:18 AM

Exception in async_discover when dispatching 'mqtt_discovery_new_switch_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/0xe0798dfffeafc828/set', 'device': {'identifiers': ['zigbee2mqtt_0xe0798dfffeafc828'], 'manufacturer': 'SONOFF', 'model': 'Zigbee two way smart switch (ZBMINI)', 'name': '0xe0798dfffeafc828'}, 'name': None, 'object_id': '0xe0798dfffeafc828', 'origin': {'name': 'Zigbee2MQTT', 'sw': '1.33.2', 'url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/0xe0798dfffeafc828', 'unique_id': '0xe0798dfffeafc828_switch_zigbee2mqtt', 'value_template': '{{ value_json.state }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 316, in async_discover config: DiscoveryInfoType = discovery_schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: string value is None for dictionary value @ data['name']

Scoff123, 2024-01-01:
Hi, I've just added a ZBMINI to my Z2M successfully, and I have both the State and link quality showing correctly in my HA. I am running Z2M version 1.34.0 which is a newer version than you are running. Could be worth backing up and trying 1.34.0? I'm running a Sonoff ZBBridge / EZSP v8 with Tasmota firmware as my coordinator. The ZBMINI paired straight away and started publishing the state via Mosquitto. My ZBMINI has firmware build date 20200927.

closed 2024-07-30 by github-actions[bot]