Koenkk/zigbee2mqtt #19681: STS-IRM-251 Smartthings Motion = Battery level wrong.

TRusselo, 2023-11-12:
### What happened?

STS-IRM-251  motion sensor is run via 2 AAA batteries.
I used it on smartthings for years, 2AAA batteries last over 2 years.  battery reporting was fine on smartthings platform.

Moved sensor to zigbee2mqtt last year.  
Home assistant showed batteries were low. put in 2 new batteries.
2 or 3 days leter gone from 100% to 15%,  showing in my low battery list
put in new batteries
2 or 3 days leter gone from 100% to 15%,  showing in my low battery list
let them run to 0% few days later.
that was 6 months ago.

TLDR;
my sensor has been running at 0% battery for 6 months.

how can we fix this?
how can i help?

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

skyconnect

closed 2024-06-10 by github-actions[bot]

ryanwnzl, 2025-01-23:
> ### What happened?
> STS-IRM-251 motion sensor is run via 2477 battery. I used it on smartthings for years, 2477 batteries in this device usually last over 2 years. battery reporting was fine on smartthings platform.
> 
> Moved sensor to zigbee2mqtt last year. Home assistant showed batteries were low. put in a new battery 2 or 3 days leter gone from 100% to 15%, showing in my low battery list put in new battery 2 or 3 days leter gone from 100% to 15%, showing in my low battery list let it run to 0% few days later. that was 6 months ago. it still works.
> 
> TLDR; my sensor has been running fine at 0% battery for 6 months.
> 
> how can we fix this? how can i help?
> 
> ### What did you expect to happen?
> _No response_
> 
> ### How to reproduce it (minimal and precise)
> _No response_
> 
> ### Zigbee2MQTT version
> 1.33.2
> 
> ### Adapter firmware version
> 7.1.1.0 build 273
> 
> ### Adapter
> skyconnect
> 
> ### Debug log
> _No response_

TRusselo did this ever get resolved, I just setup an old STS-IRM-251 which seems to be working fine, but it has a new battery and is reading 20%

I am also running with SkyConnect on Z2M

felafel, 2025-02-20:
I am also having issues with this losing connection but I am not sure if it is the device.

I also have the SkyConnect with z2m.

TRusselo, 2025-02-22:
> I am also having issues with this losing connection but I am not sure if it is the device.
> 
> I also have the SkyConnect with z2m.

sounds like you have a different issue.