Koenkk/zigbee2mqtt #19682: z2m not start after update npm and z2m to 1.33.2

git-tail-d, 2023-11-12:
### What happened?

Hello,

i updated my Raspberry to the newest software versions and z2m not start with the error:

`npm start

> zigbee2mqtt@1.33.2 start
> node index.js

Building Zigbee2MQTT... (initial build), failed
Error: Command failed: npm run build
Error: ENOENT: no such file or directory, open '/opt/zigbee2mqtt/dist/.hash'
    at Object.open (node:internal/fs/sync:78:18)
    at Object.openSync (node:fs:565:17)
    at Object.writeFileSync (node:fs:2288:35)
    at writeHash (/opt/zigbee2mqtt/index.js:38:8)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)

    at ChildProcess.exithandler (node:child_process:422:12)
    at ChildProcess.emit (node:events:514:28)
    at maybeClose (node:internal/child_process:1105:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:305:5)
`
I tried to rollback z2m to 1.33.1 and 1.33.0 with the same errormessage.

In the older issues from 2020 was a solution to change the path to 

> '/opt/zigbee2mqtt/.dist/.hash'

But it also not work

Best Regards

### What did you expect to happen?

z2m start as normal

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

no access to the frontend. the latest know firmware

### Adapter

Conbee II

lux73, 2023-11-12:
npm should not trigger any Problem after upgrade - but nodejs Updates can...

try `npm rebuild` inside z2m Folder

ddcbe, 2023-11-12:
I have the same issue. It is triggered with the initial build: npm run build 
npm rebuild doesn't help in this case.

lux73, 2023-11-12:
quick & dirty: save your **/data** Folder inside **/opt/zigbee2mqtt**

do a complete fresh install and before starting copy back your **/data** Folder

please check if your file and folder rights are matching your desired User!

if using "zigbee" as User then

chown -R zigbee /opt/zigbee2mqtt - for User rights
chgrp -R zigbee /opt/zigbee2mqtt - for Group right

lux73, 2023-11-12:
even check your nodejs Version:

```
[10:56:57] root@apu:~# node -v
v18.17.1
```

it has to be minimum v16.x - **older v12.x will not work anymore**

git-tail-d, 2023-11-12:
thank you for the dast response.

npm -v
v20.9.0

> try npm rebuild inside z2m Folder

lux73, 2023-11-12:
`node` Version

not `npm` Version

what OS do you use?

git-tail-d, 2023-11-12:
Hello,

thank you for the advise. It work now again.
> node -v
>v20.9.0
>
> Raspberry Pi 4 Model B Rev 1.5 

What i have done:

>Copy the whole data folder
>deleted the folder /opt/zigbee2mqtt

>sudo mkdir /opt/zigbee2mqtt
>sudo chown -R ${USER}: /opt/zigbee2mqtt
>
># Clone Zigbee2MQTT repository
>git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
>
># Install dependencies (as user "pi")
>cd /opt/zigbee2mqtt
>npm ci --maxsockets 1
>
>backup the data folder
>sudo chown -R ${USER}: /opt/zigbee2mqtt/data
>
>npm run build
>
>npm start

Thank you for the advise

Best Regards

lux73, 2023-11-12:
nice, always happy to help

closed 2023-11-12 by git-tail-d

ddcbe, 2023-11-12:
For me, a reinstall did the trick. Thanks for the support!

git-tail-d, 2023-11-23:
Hello,

i have now carried out the steps described above for the second time after another update was available and I cannot switch to the dev branch either.

But this is only on my productive system on which other programs and systems are running. This does not happen on a test system on which only a home automation system with z2m is running.
I therefore assume that the combination with other program parts leads to this error.
For example, npm is also used by homebridge.

I read in another thread that it was sufficient to just use git pull and npm run build. But I had to delete the complete directory and reinstall z2m.