Koenkk/zigbee2mqtt #19690: 'Invalid option for' reported in the HA log for device AM43-0.45_40-ES-EB moes blinds motor

adekloet, 2023-11-12:
### What happened?

Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:129
Integration: MQTT (documentation, issues)
First occurred: November 11, 2023 at 17:59:18 (176 occurrences)
Last logged: 19:19:36

Invalid option for select.serre_rolgordijn_links1_set_upper_limit: 'True' (valid options: ['SET'])
Invalid option for select.serre_rolgordijn_links1_set_bottom_limit: 'True' (valid options: ['SET'])
Invalid option for select.serre_rolgordijn_rechts1_set_upper_limit: 'True' (valid options: ['SET'])
Invalid option for select.serre_rolgordijn_rechts1_set_bottom_limit: 'True' (valid options: ['SET'])

Device reporting: https://www.zigbee2mqtt.io/devices/AM43-0.45_40-ES-EB.html#moes-am43-0.45%252F40-es-eb

_TZE200_zah67ekd
TS0601

I have repaired the device and reset the upper and lower limits but the above logging persists. Do I have to set anything in Zigbee2Mqtt to fix this?

The blinds seem to be working as expected.

Thank you.

### What did you expect to happen?

No logging of these invalid options

### How to reproduce it (minimal and precise)

pair this device and set the upper and lower limit. (this is set on the device itself.

Zigbee2MQTT version
[1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
zStack3x0
Coordinator revision
20221226
Coordinator IEEE Address
0x00124b002298904d
Frontend version
0.6.142
Zigbee-herdsman-converters version
15.106.0
Zigbee-herdsman version
0.21.0

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20221226

### Adapter

Sleash

Serios, 2023-11-20:
I have exactly the same blinds drivers, and can confirm the issue. HA log is constantly spammed with this messages, although the drivers are working fine.

Serios, 2023-12-26:
A possible [workaround](https://github.com/Koenkk/zigbee2mqtt/issues/18788#issuecomment-1868461369_)

adekloet, 2023-12-26:
> A possible [workaround](https://github.com/Koenkk/zigbee2mqtt/issues/18788#issuecomment-1868461369)

Thanks! I like it since I don't use those entities at all. I going to give it a try.

Koenkk, 2023-12-26:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow and the next release on 1 January.

adekloet, 2023-12-26:
> Fixed!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow and the next release on 1 January.

Wow! Perfect Koen! Thanks.

closed 2024-07-25 by github-actions[bot]