Koenkk/zigbee2mqtt #19694: Bosch BTH-RA loses Zigbee connection. re-enter in the installation code helped.

burpy, 2023-11-12:
### What happened?

Bosch BTH-RA loses the Zigbee connection on the hardware is blinking orange and does not show the antenna symbol on the display. entering the installation code again in zigbee2mqtt helped. It flashed green and comes back. 

### What did you expect to happen?

Should stay connected. 

### How to reproduce it (minimal and precise)

Wait a few days. 

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

3.05.05

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230324151035

xdubx, 2023-11-24:
Have the same problem.

xdubx, 2023-12-16:
Yes. 
I do not enter the install code as you mentioned. The device joining after some time. You see the green led blinks up. 

The connection is good with over LQI 100. Its the same floor and have only a wall between. 

Maybe its the first old bug? That the coordinator don't answer on the ack? But I'm not familiar enough with the Zigbee protocol. I would probably have to generate a log and paste it here.

slugzero, 2023-12-16:
Thank you for checking. I opened pull request https://github.com/Koenkk/zigbee-herdsman/pull/830, and if my assumption is right, this should fix it.

xdubx, 2023-12-18:
> Thank you for checking. I opened pull request https://github.com/Koenkk/zigbee-herdsman/pull/830, and if my assumption is right, this should fix it.

Thx for the fix. I will test it. Then the new update is patched on my pi :)

xdubx, 2024-01-23:
did the pr is mirrored in the zigbee2mqtt repo?

slugzero, 2024-01-24:
Yes, it is included in zigbee2mqtt since [release 1.35.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0)

xdubx, 2024-01-28:
My trv's still disconnects. HA have the version 1.35.1. I try to get debug logs and paste it here. Thx for the information.

jomo02, 2024-03-11:
I have the same problem with 1.36.0 and got them only reconnect by entering the installation code again.

xdubx, 2024-03-19:
No i don't have the problem i must only activate the join button and wait some time or make an interaction on the trv itself.

xdubx, 2024-09-16:
iam currently at 1.36. still disconnects

jomo02, 2025-01-02:
I'm still having disconnects on 1.42.0

xdubx, 2025-01-02:
same over here

Frank1616, 2025-01-08:
I also have the same problem:
I have been using 3 Bosch thermostats for about 1.5 years.
Right from the start, I had the problem that the Zigbee connection was lost after a few weeks (or randomly even months). 
I have not been able to identify a systematic or a correlation.
The thermostats can then no longer be controlled, but the values are still displayed in Home Assistant.
Zigbee2MQTT does not report an error.

Zigbee2MQTT version: 1.42.0
Coordinator: zSTack3x0, Sonoff Zigbee 3.0 Dongle Plus-P
Coordinator version: 20221226
BOSCH BTH-RA firmware: 20241128 / 3.07.04 (latest)

slugzero, 2025-01-08:
@Frank1616 consider updating your coordinator firmware to a newer version, that should fix your issue. Anything newer than [20230410](https://github.com/Koenkk/Z-Stack-firmware/tree/f5a6270875cc13d32387524b54d460c9b6274841/coordinator/Z-Stack_3.x.0/bin) should work. If you decide to update to the latest firmware and the issue persists, I'd recommend to also try 20230410, because I am not 100% certain if my fix is also included in every newer firmware version. I am running my thermostats with that version and did not have any disconnects since.
@xdubx I'm afraid there is not much I can do to help with your issue. If your assumption is right that this is the old issue, the only thing you could try is also to switch to coordinator firmware 20230410 and check if that changes anything.

xdubx, 2025-01-08:
running the version 20230507. but i check if a newer version is available.

Frank1616, 2025-01-11:
@slugzero 
Thank you for your advice. I will do a coordinator firmware update in the next few days.
I will report if it has helped.

Frank1616, 2025-01-12:
Now running also coodinator version 20230507. At the moment everything seems to fine.
Also tried version 20240710 (latest stable). But all of my Busch-Jaeger light switches failed and I could not "re-pair" them.

closed 2025-03-21 by github-actions[bot]

jomo02, 2025-05-06:
It still happens for me in Zigbee2MQTT 2.2.1, Coordinator 20230507 and BTH-RA 3.07.06.

jomo02, 2025-05-06:
@burpy Have you always used the ZBDongle-E (seeing your EmberZNet coordinator)? And what has changed since your first post here?

burpy, 2025-05-06:
Yes the adapter is still the same -> SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230324151035
I use the dongle with 1m USB-cable on an Intel NUC Home Assistant.

I changed nothing. It worked from one day to the next. Maybe there was a zigbee2mqtt update fixing it but I don't know when this was. It works for me for many months or maybe years now.

Let me know I you need more details to compare our zigbee networks.

ParalaXEngineering, 2026-02-20:
Hello, did someone found some kind of issue? I have 8 of them, and now 1 is starting to act up, possibly a second one...