Koenkk/zigbee2mqtt #19696: White page (iOS) via Nginx Proxyy - no basic auth

boehser-enkel, 2023-11-13:
### What happened?

Getting a white page using Webkit Browser via Nginx Proxy Manager
No auth_basic configured
Even with auth_token configured 

Nginx:
`# ------------------------------------------------------------
# z2.max-wob.de, zm.max-wob.de
# ------------------------------------------------------------

server {
  set $forward_scheme http;
  set $server         "192.168.88.202";
  set $port           8383;

  listen 80;
listen [::]:80;

listen 443 ssl http2;
listen [::]:443 ssl http2;

  server_name z2.max-wob.de zm.max-wob.de;

  # Let's Encrypt SSL
  include conf.d/include/letsencrypt-acme-challenge.conf;
  include conf.d/include/ssl-ciphers.conf;
  ssl_certificate /etc/letsencrypt/live/npm-2/fullchain.pem;
  ssl_certificate_key /etc/letsencrypt/live/npm-2/privkey.pem;

  # Block Exploits
  include conf.d/include/block-exploits.conf;

    # Force SSL
    include conf.d/include/force-ssl.conf;

proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection $http_connection;
proxy_http_version 1.1;

  access_log /data/logs/proxy-host-16_access.log proxy;
  error_log /data/logs/proxy-host-16_error.log warn;

  location / {

    

    # Access Rules: 26 total
    
    allow 173.245.48.0/20;
    
    allow 103.21.244.0/22;
    
    allow 103.22.200.0/22;
    
    allow 103.31.4.0/22;
    
    allow 141.101.64.0/18;
    
    allow 108.162.192.0/18;
    
    allow 190.93.240.0/20;
    
    allow 188.114.96.0/20;
    
    allow 197.234.240.0/22;
    
    allow 198.41.128.0/17;
    
    allow 162.158.0.0/15;
    
    allow 104.16.0.0/13;
    
    allow 104.24.0.0/14;
    
    allow 172.64.0.0/13;
    
    allow 131.0.72.0/22;
    
    allow 2400:cb00::/32;
    
    allow 2606:4700::/32;
    
    allow 2803:f800::/32;
    
    allow 2405:b500::/32;
    
    allow 2405:8100::/32;
    
    allow 2a06:98c0::/29;
    
    allow 2c0f:f248::/32;
    
    allow 192.168.88.0/24;
    
    allow 2003:e0:171a:b600::/64;
    
    allow 172.18.0.0/16;
    
    allow 172.18.0.1/32;
    
    deny all;

    # Access checks must...
    
    satisfy all;
    

    
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;
    proxy_http_version 1.1;
    

    # Proxy!
    include conf.d/include/proxy.conf;
  }

  # Custom
  include /data/nginx/custom/server_proxy[.]conf;
}

`

### What did you expect to happen?

Zigbee2MQTT..

### How to reproduce it (minimal and precise)

Acess via Webkit browser

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20220219

### Adapter

Sonoff

lux73, 2023-11-15:
this config works for me several month:

```
server {
	listen							443 ssl http2;
	server_name						z2m.xxx.xx;

	include							header.conf;
	include							ssl_options.conf;

	ssl_certificate					/usr/local/etc/letsencrypt/live/z2m.xxx.xx/fullchain.pem;
	ssl_certificate_key				/usr/local/etc/letsencrypt/live/z2m.xxx.xx/privkey.pem;
	ssl_trusted_certificate			        /usr/local/etc/letsencrypt/live/z2m.xxx.xx/chain.pem;

	satisfy							all;

	allow							192.168.0.0/20;																					#allow own IP-Range
	allow							10.10.11.0/24;																					#allow OpenVPN
	deny							all;	

	auth_basic						"Restricted Area";
        auth_basic_user_file			/usr/local/etc/nginx/.htpasswd; 

	access_log						/var/log/nginx/z2m_access.log combined buffer=16k flush=5m if=$abnormal;
	error_log						/var/log/nginx/z2m_error.log error;

	client_max_body_size				32k;
	large_client_header_buffers		4	16k;

	location / {
		proxy_pass			http://192.168.12.230:8081/;
		proxy_set_header	Host $host;
		proxy_set_header	X-Real-IP $remote_addr;
		proxy_set_header	X-Forwarded-For $proxy_add_x_forwarded_for;
	}

	location /api {
		proxy_pass			http://192.168.12.230:8081/api;
		proxy_http_version	1.1;
		proxy_set_header	Host $host;
		proxy_set_header	Upgrade $http_upgrade;
		proxy_set_header	Connection "upgrade";
	}
}
```

and btw. your

```
allow 172.18.0.0/16;
 allow 172.18.0.1/32;
```

makes no sense - /16 includes all 172.18.0.1/32

boehser-enkel, 2023-11-16:
> include							header.conf;
> 	include							ssl_options.conf;

Tested that but also blank page. 
Needed to remove 
`	auth_basic						"Restricted Area";
        auth_basic_user_file			/usr/local/etc/nginx/.htpasswd; `

`	include							header.conf;
	include							ssl_options.conf;`

> makes no sense - /16 includes all 172.18.0.1/32

 I really do know but it just did not work without the second entry. Even though it's redundant.

lux73, 2023-11-16:
you sure there is a .htpasswd in the mentioned folder?

afaik the .htpasswd file is NOT plain text - make surevitbis properly encrypted

lux73, 2023-11-16:
Oh i See there is some mistake

I need to post both inludes to work - will post them in a few Hours 🙈

boehser-enkel, 2023-11-16:
> you sure there is a .htpasswd in the mentioned folder?
> 
> The .htpasswd file is NOT plain text - make sure it is properly encrypted

i don't want to do any additional auth

lux73, 2023-11-16:
header.conf
```
add_header X-Frame-Options "SAMEORIGIN";
add_header X-Content-Type-Options nosniff;
add_header X-XSS-Protection "1; mode=block";
```

ssl-options.conf

```
add_header Strict-Transport-Security "max-age=31536000; includeSubDomains; preload;" always;

ssl_protocols TLSv1.2 TLSv1.3;
ssl_session_cache shared:SSL:10m;
ssl_session_timeout 10m;
ssl_session_tickets off;

# Enable OCSP stapling (http://blog.mozilla.org/security/2013/07/29/ocsp-stapling-in-firefox)
ssl_stapling on;
ssl_stapling_verify on;
resolver MY_PRIVATE_DNS_SERVER_IP valid=300s;
resolver_timeout 5s;

ssl_dhparam /usr/local/etc/nginx/ssl/dhparam4096.pem;

ssl_prefer_server_ciphers on;
ssl_ciphers "ECDHE-RSA-AES256-GCM-SHA384:ECDHE-ECDSA-AES256-GCM-SHA384:ECDHE-ECDSA-CHACHA20-POLY1305:ECDHE-RSA-CHACHA20-POLY1305:DHE-RSA-AES256-GCM-SHA384:!aNULL:!eNULL:!EXPORT:!DES:!RC4:!3DES:!MD5:!PSK";
ssl_ecdh_curve secp521r1:secp384r1;
```

**you have to generate your own dhparameter via openssl** - stated in my ssl config as dhparam4096.pem

boehser-enkel, 2023-11-16:
> header.conf
>
> ```
> add_header X-Frame-Options "SAMEORIGIN";
> add_header X-Content-Type-Options nosniff;
> add_header X-XSS-Protection "1; mode=block";
> ```
>
> ssl-options.conf
>
> ```
> add_header [...]

But do i need that all for the problem?

lux73, 2023-11-16:
i don't know - all i ccould say: this configs postet works for me several month ^^

i would say say you can drop all the ssl stuff if don't need - but header.conf IS mandatory

boehser-enkel, 2023-11-17:
> i don't know - all i ccould say: this configs postet works for me several month ^^
> 
> i would say say you can drop all the ssl stuff if don't need - but header.conf IS mandatory

White page or loading error in any cases

lux73, 2023-11-17:
that's strange

post nginx proxy logfile after accessing - access.log & error.log pls

boehser-enkel, 2023-11-23:
> that's strange - do you use docker container virtualization or plain installation Services?
> 
> post nginx proxy logfile after accessing - access.log & error.log pls
> 
> /edit
> 
> i found a github repo that shares another nginx config -> https://github.com/linuxserver/reverse-proxy-confs/blob/master/zigbee2mqtt.subdomain.conf.sample

This one is missing /config/nginx/ssl.conf; & include /config/nginx/proxy.conf; & include /config/nginx/resolver.conf;

access Log:

```
[23/Nov/2023:06:48:12 +0000] - 200 200 - GET https *domain* "/" [Client 198.41.242.171] [Length 3430] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0" "-"
[23/Nov/2023:06:48:12 +0000] - 200 200 - GET https *domain* "/assets/index-94472864.css" [Client 172.71.160.48] [Length 27359] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0" "https://*domain*/"
[23/Nov/2023:06:48:12 +0000] - 304 304 - GET https *domain* "/assets/index-d19256aa.js" [Client 172.71.160.121] [Length 0] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0" "https://*domain*/"
[23/Nov/2023:06:48:12 +0000] - 304 304 - GET https *domain* "/assets/favicon-a8299d0b.ico" [Client 172.71.160.125] [Length 0] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0" "https://*domain*/"
[23/Nov/2023:06:48:17 +0000] - 200 200 - GET https *domain* "/" [Client 198.41.242.171] [Length 3430] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "-"
[23/Nov/2023:06:48:17 +0000] - 304 304 - GET https *domain* "/assets/index-94472864.css" [Client 172.71.160.48] [Length 0] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "https://*domain*/"
[23/Nov/2023:06:48:17 +0000] - 304 304 - GET https *domain* "/assets/index-d19256aa.js" [Client 172.71.160.121] [Length 0] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "https://*domain*/"
[23/Nov/2023:06:48:17 +0000] - 304 304 - GET https *domain* "/assets/index-d19256aa.js" [Client 172.71.160.121] [Length 0] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "https://*domain*/"
[23/Nov/2023:06:48:17 +0000] - 304 304 - GET https *domain* "/assets/favicon-a8299d0b.ico" [Client 172.71.160.125] [Length 0] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "https://*domain*/"
[23/Nov/2023:06:48:19 +0000] - 200 200 - GET https *domain* "/" [Client 198.41.242.171] [Length 3430] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "-"
[23/Nov/2023:06:48:19 +0000] - 200 200 - GET https *domain* "/" [Client 198.41.242.171] [Length 3430] [Gzip -] [Sent-to 192.168.88.202] "Mozilla/5.0 (iPhone; CPU iPhone OS 17_1_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.1.1 Mobile/15E148 Safari/604.1" "-"

```

error log: there is no error logged

closed 2024-06-21 by github-actions[bot]