Koenkk/zigbee2mqtt #19703: Hue iris uncontrollable after pairing. (works fine with ZHA and Deconz)

nelbs, 2023-11-13:
### What happened?

Recently, I migrated over 60 devices from deconz (Conbee II) to zigbee2MQTT (Sonoff 3.0 ZBDongle E). It required some effort, but almost every device is functioning well under the new setup. However, the Hue Iris is a pain in the ass. Although the pairing process went smoothly, the device became unavailable afterward, and I couldn't control it. I've tried unpairing and repairing it multiple times, and sometimes it becomes uncontrollable after a few hours, while other times it happens immediately after pairing.

The light is located a few meters from the coordinator and is surrounded by 15 routers. To eliminate the device as the source of the problem, I set up a ZHA network (with only 1 device) using my old Conbee II stick. After a successful pairing, the device remained controllable. I switched a few times between ZHA and Zigbee2MQTT and every time the problem occurs only with Z2MQTT.

When attempting to control the light, I receive the following message:

`error 2023-11-13 20:55:10Publish 'set' 'state' to '0x001788010333c5b8' failed: 'Error: Command 0x001788010333c5b8/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4068 - 11 - 177 - 6 - 11 after 10000ms)'`

Her are some details about the light
[image]

### How to reproduce it (minimal and precise)

- reset iris to factory settings
- pair device
- control it

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13341405/log.txt)

jensalbrecht81, 2023-12-05:
I'm having the same issue, I already changed from channel 11 to 25 but unfortunate the Hue Iris got disconnected after a while and never re-connects until I reset it using the Hue Dimmer.

closed 2024-07-10 by github-actions[bot]

ensonic, 2025-01-26:
I have the same issue. It happens after generating a map (aka scanning the network). If I unpower the Hue Iris and replug after a few days it works again.