Koenkk/zigbee2mqtt #19704: reate a CC2530 router

tux1980, 2023-11-13:
### What happened?

Hi Dear Team,

I did build 2 CC2530 Router as described over here:
https://www.zigbee2mqtt.io/advanced/zigbee/05_create_a_cc2530_router.html

The description worked like charme and after 1 hour I flashed the Devices with the Router .hex Firmware and everthing seemed fine. Both Devices are up and running.

But I cannot detect these devices in my Homeassistent Installation. I have already a Zigbee network with more than 50 devices (Which is the reason for the 2 routers). The Zigbee Coordinator (A ConBee II from Dresden electronic) is plugged into my Rasperry 4B USB Slot with my HomeAssistent Installation on top of the Raspberry 4B+.

Even after allowing the "zha Integration" to join Zigbee Devices automatically - no success:(. I can neither see the 2 new routers in the:

a) Zigbee Log

b) The visual Dashboard

c) in the Zigbee Device /Entity List.

You may have a suggestion how to slove that issue since I'm at my wits end.

Thanks
Wolfgang

### What did you expect to happen?

Homeassistent to discover the 2 new Zigbee Routers to extend the Zigbee Mesh Network.

### Zigbee2MQTT version

I'm using HomeAssistent with the zha Integration

### Adapter firmware version

Last Firmware, flashed it yesterday

### Adapter

CC2530

francisp2, 2023-11-14:
No idea. With Zigbee2MQTT you definitely see them

[image]

tux1980, 2023-11-14:
@francisp2 Ok, I do not have the Zigbee2MQTT add-on. Thought this might be working straight with the zha Integration? But trying to do so -  I cannot find the Zigbee2MQTT add-on, neither in the official HA repo nor in the HACS.

May you please explain how you did the setup in HomeAssistent? 

Thanks a lot
Wolfgang

francisp2, 2023-11-16:
I'm running a bare linux install of Zigbee2MQTT, not a homeassistant add-on.

DubhAd, 2023-11-16:
This is for issues with Zigbee2MQTT - for ZHA problems join the Home Assistant Discord server and ask there

Probably you'll find that ZHA doesn't create entities for routers

closed 2024-06-14 by github-actions[bot]