Koenkk/zigbee2mqtt #19705: [New device support]: Add SPM01-D2TZ-U01

dduransseau, 2023-11-13:
### Link

https://bituo-technik.com/wp-content/uploads/2022/11/SPM01-TW_Manual.pdf

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xbc026efffe7bc371","nwkAddr":54749,"manufId":4630,"manufName":"BITUO TECHNIK","powerSource":"Mains (single phase)","modelId":"SPM01X001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1281,"inClusterList":[0,3,1794,2820,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"appVersion":4,"hwVersion":1,"manufacturerName":"BITUO TECHNIK","modelId":"SPM01X001","locationDesc":"01.00.04","swBuildId":"003.00.03"}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":100,"acCurrentMultiplier":1,"acCurrentDivisor":100,"acPowerMultiplier":1,"acPowerDivisor":1000,"acFrequencyDivisor":100,"acFrequencyMultiplier":1,"acFrequency":4994,"totalActivePower":0,"totalReactivePower":0,"totalApparentPower":0,"rmsVoltage":23749,"rmsCurrent":5,"activePower":1,"reactivePower":0,"apparentPower":18,"powerFactor":14,"powerMultiplier":1,"powerDivisor":1000}},"seMetering":{"attributes":{"currentSummDelivered":[0,5],"currentSummReceived":[0,0],"divisor":1000,"multiplier":1}}},"binds":[{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd30f0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd30f0","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd30f0","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd30f0","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":1,"swBuildId":"003.00.03","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1699911885729,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to integrate the device SPM01-D2TZ-U01 (energy metering device with haElectricalMeasurement and seMetering cluster) but I face 2 issues:

First one is during startup where the service is fail to get acFrequency metric during 3 attempts, after it work:
```
Zigbee2MQTT:error 2023-11-13 18:24:59: Failed to configure '0xbc026efffe7bc371', attempt 1 (Error: ConfigureReporting 0xbc026efffe7bc371/1 haElectricalMeasurement([{"attribute":"acFrequency","minimumReportInterval":5,"maximumReportInterval":300,"reportableChange":10}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')
    at Endpoint.checkStatus (C:\Users\Damien\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:421:28)
    at Endpoint.configureReporting (C:\Users\Damien\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:762:22)
    at Object.acFrequency (C:\Users\Damien\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\reporting.ts:236:5)
    at Object.configure (C:\Users\Damien\zigbee2mqtt\data\extension\externally-loaded.js:45:9)
    at Configure.configure (C:\Users\Damien\zigbee2mqtt\lib\extension\configure.ts:117:13)
    at Immediate.<anonymous> (C:\Users\Damien\zigbee2mqtt\lib\extension\configure.ts:69:17))
```

Second even if power values seems correctly parsed the metric is exposed with value 0:
```
Zigbee2MQTT:debug 2023-11-13 17:48:01: Received Zigbee message from '0xbc026efffe7bc371', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"acFrequency":4992,"activePower":3,"apparentPower":29,"powerFactor":12,"reactivePower":0,"rmsCurrent":9,"rmsVoltage":23759,"totalActivePower":0,"totalApparentPower":0,"totalReactivePower":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-11-13 17:48:01: MQTT publish: topic 'zigbee2mqtt-dev/0xbc026efffe7bc371', payload '{"ac_frequency":50,"current":0.09,"energy":0.005,"linkquality":92,"power":0,"power_apparent":0,"power_factor":0.12,"power_reactive":0,"produced_energy":0,"voltage":237.59}'
```
This is my first try to integrate a device, so any help to fix my converter is greatly appreciated.

PS: The power factor seems inverted ( 100 - value) but I have to confirm that point

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;

const energy_metering = {
    cluster: 'seMetering',
    type: ['attributeReport'],

    convert: (model, msg, publish, options, meta) => {
        if (utils.hasAlreadyProcessedMessage(msg, model)) return;
        const payload = {};
        if (msg.data.hasOwnProperty('currentSummDelivered')) {
            const data = msg.data['currentSummDelivered'];
            const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
            payload.energy = value / 1000;
        }
        if (msg.data.hasOwnProperty('currentSummReceived')) {
            const data = msg.data['currentSummReceived'];
            const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
            payload.produced_energy = value / 1000;
        }
        return payload;
    }
}

const definition = {
    zigbeeModel: ['SPM01X001'],
    model: 'SPM01-D2TZ-U01',
    vendor: 'BITUO TECHNIK',
    description: 'Smart Energy Sensor',
    fromZigbee: [fz.electrical_measurement, energy_metering], // We will add this later 
    toZigbee: [],
    exposes: [e.power(), e.current(), e.voltage(), e.ac_frequency(), e.power_factor(), e.power_apparent(), e.power_reactive(), e.energy(), e.produced_energy()], // e.energy(), e.produced_energy()
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint, readFrequencyAttrs=true);
        await reporting.acFrequency(endpoint);
        await reporting.activePower(endpoint);
        await reporting.reactivePower(endpoint);
        await reporting.apparentPower(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.powerFactor(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.currentSummReceived(endpoint);
    },
};

module.exports = definition;
```

Naruto-kyun, 2023-12-17:
Hi,

This is my first converter too, and thanks to your work and some reverse engeeneering, this is how i get it working except power factor that seem divded by 100 (and cannot be configured in code: Koenkk/zigbee-herdsman-converters/blob/master/src/converters/fromZigbee.ts:801)

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['SPM01X001'],
    model: 'SPM01-D2TZ-U01',
    vendor: 'BITUO TECHNIK',
    description: 'Smart Energy Sensor',
    fromZigbee: [fz.electrical_measurement, fz.metering],
    toZigbee: [],
    exposes: [e.ac_frequency(), e.power(), e.power_reactive(), e.power_apparent(), e.current(), e.voltage(), e.power_factor(), e.energy(), e.produced_energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            divisor: 1000000, multiplier: 1
        });
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 100, 
            acCurrentMultiplier: 1, acCurrentDivisor: 100, 
            acPowerMultiplier: 1, acPowerDivisor: 1,

        });
        device.save();
    },
};

module.exports = definition;`

Naruto-kyun, 2023-12-17:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['SPM01X001'],
    model: 'SPM01-D2TZ-U01',
    vendor: 'BITUO TECHNIK',
    description: 'Smart Energy Sensor',
    fromZigbee: [fz.electrical_measurement, fz.metering],
    toZigbee: [],
    exposes: [e.ac_frequency(), e.power(), e.power_reactive(), e.power_apparent(), e.current(), e.voltage(), e.power_factor(), e.energy(), e.produced_energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        endpoint.saveClusterAttributeKeyValue('seMetering', {
            divisor: 1000000, multiplier: 1
        });
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 100, 
            acCurrentMultiplier: 1, acCurrentDivisor: 100, 
            acPowerMultiplier: -1, acPowerDivisor: 1,

        });
        device.save();
    },
};

module.exports = definition;

dduransseau, 2024-01-05:
Hi @Naruto-kyun thanks for your improvement, I just run a test and work fine except energy value.
Why you corrected the divisor, especially the minus sign ? Default value seems correct to me.
I'm investigating the power factor issue since I tested the module with ZHA and the value have the same issue but on raw data the value sent by the module seems correct:
`"0x0510": {
                "attribute_name": "power_factor",
                "value": 100
              },`

Naruto-kyun, 2024-01-05:
Hi @dduransseau, i change the factors to match what home assistant wants (otherwise the value doesn't match the unit)

Concerning the power factor this is hard coded in the file I mentioned earlier and it would require to change zigbee-herdsman-converters to fix the issue.

Actually I'm not even sure this issue is in the right repo.

dduransseau, 2024-01-05:
Understood for HA, however I'm not sure this fix should be handled by zigbee2mqtt instead of HA.
Regarding power factor I see the issue:
Following [Zigbee Cluster Library](https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf) p348 the power factor is represented as
> Contains the single phase or PhaseA, Power Factor ratio in 1/100ths.
So it's divided by 100 by zigbee-herdsman-converter to have a power factor between 0 and 1, which is the correct behavior. However the front-end display it as a percentage which is wrong but a common usage. So either the value should be divided by 100 and front-end should not display % or the value should be let as it is and percentage can be use. I'll open a dedicated issue to discuss that point.

Naruto-kyun, 2024-01-05:
> Understood for HA, however I'm not sure this fix should be handled by zigbee2mqtt instead of HA.

When i change the sigy, it actualy match what an other zigbee sensor do and what my zwave sensors do. But i'm not sure this the right way.

dduransseau, 2024-01-06:
@Naruto-kyun In HA you can set the metric (entity) to an individual device or a solar panel in the energy dashboard to define if it's consumption or production. 
There is some meter that support only one global metric like total_energy = (consumption-production) which consider production as negative power and consumption as positive. However for this meter that support both metric, to me it do not make sens to force a specific sign since each direction have dedicated metrics.
Can you share a meter that count energy with negative sign ?
Anyway I agree to force the power_multiplier to 1/-1 since the default one (1000) is wrong for W and VA.

dduransseau, 2024-01-07:
Device implemented in zigbee-herdsman-converters release 18.9.0

Naruto-kyun, 2024-01-07:
I've made the update.
This is what i said, now everithing is okay, except that now the power inverted.
Exemple with my electric hob (the update has been made a 14:00)
[image]

dduransseau, 2024-01-07:
I see, I'll have a look at HA to check how the direction should be manage. Do you have a reference of another meter that have the behavior you want (- for consumption and + for production) ?

Naruto-kyun, 2024-01-07:
I want + for consumption and - for production, but it gave me the other way.
And i know i mounted it with the good orientation since "Energy" increase and "Produced energy" stay at 0 in z2m
(I've done the error at first and "Produced energy" did increase but "Power" was positive)

You dont have this behavior?

dduransseau, 2024-01-07:
No, very weird I just checked the module is in good orientation but I correctly have + for consumption. Did not checked production (I can do it later this week). We agree that the pairing button is on "consumption side" and the module is on phase/live wire ?

Naruto-kyun, 2024-01-07:
https://bituo-technik.com/wp-content/uploads/2022/11/SPM01-TW_Manual.pdf
Since we have both the D2TZ variant your description seems good
(U2TZ would have been the other way)

Naruto-kyun, 2024-01-07:
Looking at your data (in 1st post): "energy":0.005,"produced_energy":0,
It doesnt seems like you have the issue...

Naruto-kyun, 2024-01-07:
I just found the "official" implementation: https://github.com/script0803/SPM0X-Z2M-Configuration/blob/main/SPM01_Z3_Local_01.js

dduransseau, 2024-01-07:
> I just found the "official" implementation: https://github.com/script0803/SPM0X-Z2M-Configuration/blob/main/SPM01_Z3_Local_01.js

Wow this implementation seems more complex than necessary. 
Anyway happy that you have accurate value now. Did you try the integration released in this morning ? It's working fine on my side, that would be nice if you can confirm everything is OK and the multiplier/divisor is correct.

Naruto-kyun, 2024-01-07:
Except "Energy" and "Produced Energy", all values seems correct.

I did merge their code and ours to get it working:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const { calibrateAndPrecisionRoundOptions } = require('zigbee-herdsman-converters/lib/utils');
const utils = require('zigbee-herdsman-converters/lib/utils');

const converters = {
    seMetering: {
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg, model)) return;
            const payload = {};
            const multiplier = msg.endpoint.getClusterAttributeValue('seMetering', 'multiplier');
            const divisor = msg.endpoint.getClusterAttributeValue('seMetering', 'divisor');
            const factor = multiplier && divisor ? multiplier / divisor : null;

            if (factor != null && (msg.data.hasOwnProperty('currentSummDelivered') ||
                msg.data.hasOwnProperty('currentSummReceived'))) {
                if (msg.data.hasOwnProperty('currentSummDelivered')) {
                    const data = msg.data['currentSummDelivered'];
                    const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
                    payload.energy = calibrateAndPrecisionRoundOptions(value * factor, options, 'energy');
                }
                if (msg.data.hasOwnProperty('currentSummReceived'))  {
                    const data = msg.data['currentSummReceived'];
                    const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
                    payload.produced_energy = calibrateAndPrecisionRoundOptions(value * factor, options, 'energy');
                }
            }
            return payload;
        }
    }
}

const definition = {
    zigbeeModel: ['SPM01X001'],
    model: 'SPM01X001',
    vendor: 'BITUO TECHNIK',
    description: 'Smart energy monitor for 1P+N system',
    fromZigbee: [fz.electrical_measurement, converters.seMetering],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
        await endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acPowerMultiplier: 1, acPowerDivisor: 1,
        });
        device.save();
    },
    exposes:  [
        e.ac_frequency(), e.power(), e.power_reactive(), e.power_apparent(), e.current(),
        e.voltage(), e.power_factor(), e.energy(), e.produced_energy()
    ],
};

module.exports = definition;
`

Naruto-kyun, 2024-01-13:
Also there is an other bug.
If a device has lost power, i to reconfigure it to get the good power scale back.

dduransseau, 2024-01-15:
Good point, I'll test to manage power directly into fromZigbee to bypass divisor.

Niamor71, 2024-01-30:
Hello there, trying your codes, I do not have any energy and produced_energy returned (returns null)...

`Zigbee2MQTT:info  2024-01-30 22:12:14: MQTT publish: topic 'zigbee2mqtt/BITUO Sensor', payload '{"ac_frequency":49.92,"current":9.03,"energy":null,"linkquality":216,"power":2106,"power_apparent":2129,"power_factor":0.98,"power_reactive":0,"produced_energy":null,"voltage":235.8}'`

Maybe I miss something....

dduransseau, 2024-02-02:
Hi, even after few minutes ? I might append that seMetering status are not send quickly if energy don't vary but after few minutes it's working fine (at home at least)

Niamor71, 2024-02-02:
Well.. no, energy values nevers popped.... What I did to make it work ; copy the code referenced by @Naruto-kyun  (https://github.com/script0803/SPM0X-Z2M-Configuration/blob/main/SPM01_Z3_Local_01.js) as it is... OK now I have some values and I get two new entities  : 
[image]
"hw version" and "locationdesc" with no values, but I don't care ! Mmm... power factor values seems quite strange to me...

Le-Syl21, 2024-03-18:
Hi,

I'm using Z2M edge on HASSIO with https://www.zigbee2mqtt.io/devices/SPM02V2.html

As you, I can't show enrgy things...

    "energy": null,
    "produced_energy": null
[image]
I will wait some hours and keep you in touch

tnesztler, 2024-03-29:
After switching to the "latest-dev" Docker image of Z2M (ie. the next version should be 1.36.1), I was able to see both "energy" and "produced energy". The fix is in a newer version of `zigbee_herdsman_converters` (from 3 weeks ago).
Thank you for the fix :)

Le-Syl21, 2024-04-01:
I've reinstall EDGE version today, not better :-/

GizzGool, 2024-04-02:
Hi,

Z2M 1.36.1-1 and not better :-/
[image]

Thanks in advance if anybody has a workaround ...

dduransseau, 2024-04-02:
I just had a look as this issue. I'm trying a fix that seems to work well at home even if I didn't had the issue here:
[custom_adapter](https://gist.githubusercontent.com/dduransseau/6232142b7a59280e76b705b512dffaf5/raw/44f8b445605c7fd4301a71874b1b513550828b78/SPM01X001.js)

GizzGool, 2024-04-05:
> I just had a look as this issue. I'm trying a fix that seems to work well at home even if I didn't had the issue here: [custom_adapter](https://gist.githubusercontent.com/dduransseau/6232142b7a59280e76b705b512dffaf5/raw/44f8b445605c7fd4301a71874b1b513550828b78/SPM01X001.js)

Thanks a lot for your help and answer @dduransseau 
But my device is SPM**02**X001 - **3** phases - (not SPM01X001 as in your fix).
should I just replace this reference in your file 🤔

dduransseau, 2024-04-05:
> But my device is SPM**02**X001 - **3** phases - (not SPM01X001 as in your fix).

You can try with [this](https://gist.githubusercontent.com/dduransseau/1631dd94e36c5d0c3ae3550094daf510/raw/36a37af55529722813a32fa45fd7426880e314da/SPM02X001.js)

GizzGool, 2024-04-07:
> You can try with [this](https://gist.githubusercontent.com/dduransseau/1631dd94e36c5d0c3ae3550094daf510/raw/36a37af55529722813a32fa45fd7426880e314da/SPM02X001.js)

Great @dduransseau , "Energy" and "Produced energy" appeared instantly 👍
However, after testing, I think their calculation is wrong with the current converter.
So I changed the divisor from 1000 to 100:

payload.energy = value / 100;
payload.produced_energy = value / 100;

Finally, do you think it is necessary to push this converter into the z2m base, given that the product SPM02X001 seems tobe already in it as it is already automatically detected but therefore without these 2 values?
It would not be bad to also have a nice image of the product 😁

dduransseau, 2024-04-07:
Your right I made a mixup between old integration where I hardcoded the divisor. Just updated the gist to use the divisor configured on the device, that should fixe the issue in a better way.

dduransseau, 2024-04-09:
Hi @script0803 I see you updated the SPM01 and SPM02 integration. I'm not sure of your fix, I'm only have a SPM01 and it do not support to be queried for energy. Did it raise an error for you ? What firmware version do you have ?

script0803, 2024-04-09:
Sorry guys, I just noticed this discussion. After the release of version 1.36.1, I found the error about energy and produced energy not being displayed. Now,  I fixed the error and tested it before I submitted it and it should be correct. 
Thank you for your support for our products. Thank you very much. If you encounter any problems, please @ me directly and I will do my best to help.

script0803, 2024-04-09:
In addition, we specify the parameters measured by our products in this way, where voltage and current are 0.00, active power/reactive power/apparent power is 0.000, and the unit is kW, but I found that all used in zigbee2mqtt seem to be w, so I Changed the ACPowerMultiplier and ACPowerDivisor in the file to use Watt as the unit.
Regarding electrical energy, we used 0.00, so we need to divide the last read data by 100.

GizzGool, 2024-04-09:
Hi @dduransseau and @script0803 !
Thank you very much for your response 👍

In addition, just commestic things, but is it possible that z2m detected these exposes with "phase **a**" at the end :
- Power (like Power phase b and Power phase c),
- Power reactive,
- Power apparent,
- Current,
- Voltage,
- Power factor.
Or with numbers 1, 2, and 3 like other energy meters?

Also, I noticer that entities Power, Power phase b and Power phase c (and others like courant, tension, ...) are présent in "Diagnostic" section and not in "Sensors" section like for exemple "Power apparent" ?

[image]

script0803, 2024-04-10:
Oh, I seem to have found a problem. It should be caused by the label added in the picture, but these definitions were not added by me. 
Therefore, I am not sure why the person who added this definition added diagnostic labels to parameters such as power current.
[image]

GizzGool, 2024-04-10:
Hello @script0803 

Thanks a lot for your feedbak !

> Regarding the issue of phase a, I have tried it before, but it was rejected by the administrator of zigbee2mqtt. I don’t know the specific reason.

Very strange indeed, I really don't see why not, knowing that many other products have the phases presented like this (either with letters or with numbers)! Do you have a way to resubmit?

> There is also a problem about the power display on diagnostic. I will further find out the reason and I will give you an answer later.

Good work ! Again do you have a way to resubmit with correction?

Have a nice day.

script0803, 2024-04-10:
Hi @GizzGool
I will try to submit again during my working hours tomorrow, please wait for my feedback, thank you！

GizzGool, 2024-05-16:
Hi @script0803,
I would like to come back to you because I have the impression that since your update #7350, now the instantaneous power on each phase is reported in W whereas it should be in kW. When I draw 3kW from a phase, it displays 3W (instead of 3000W).
If I use the converter from before, I obtain the correct values ​​(but without an icon for the device ;-).
Thank you in advance for your feedback.
Very good day

nicolinuxfr, 2024-06-07:
Hello 🙂 

I also have the SPM01-D2TZ-U01 connected to the latest stable version of Z2M (1.38.0 commit: fe048e6) and I have the same issues as the other : a wrong value for the power (in my case, it seems to need a multiplication by 1000, for instance I have 2,500 W instead of 2500 W) and no values at all for the energy.

What should I do @script0803  ? I thought the integration was fixed inside Z2M, but should I use a custom adapter ?

Thanks !

GizzGool, 2024-06-16:
Hi @script0803,
When do you think we will have the modification pushed for wrong power value issue.
Because it is too bad to have used a converter when this product is officially supported by z2m. 😉
Thanks in advance for your help.

nicolinuxfr, 2024-06-16:
Using the converter, I noticed another issue, this time with energy monitoring. The value it reports is way too low, I'm using it on an oven and the device reported values that were impossible.

I added [an integral](https://www.home-assistant.io/integrations/integration/) using the left method based on the power value and it seems to me it's OK, though I don't have another device to compare it to. 

What I'm sure is the default energy value is way off, it's on the bottom on this screenshot and I installed the device a week ago. There is no way my oven, which is often used, only consumed 0,8 kWh during this time.

[image]

jbouv55151, 2024-07-17:
Thank @script0803 for the updates. But, as @GizzGool and other users, we do have a problem with power, power_apparent et power_reactive which are 1000 to small.
Do you think it will be updated in Zigbee2MQtt soon ?
Is there a converter meantime ?

nicolinuxfr, 2024-07-17:
In the meantime, my solution may not be ideal, but it works perfectly, FYI.

[image]

Patricen1, 2025-01-03:
UP

nemesix29, 2025-01-19:
UP too

dduransseau, 2025-03-23:
Just to clarify the issue here. The power values are wrong because the device itself communicate a divisor of 1000 as discussed in [#8395](https://github.com/Koenkk/zigbee-herdsman-converters/issues/8395). Even if it's possible to force Z2M to multiply values by 1000 in converter, it would be a better to fix over firmware update.

Patricen1, 2025-03-24:
> Just to clarify the issue here. The power values are wrong because the device itself communicate a divisor of 1000 as discussed in [...]

Hello @dduransseau , just for workaround, how do you fix the issue within Z2M?

dduransseau, 2025-03-24:
It's possible to use an external converter to multiply power by 1000, the issue to implement directly in Z2M is if the device is fixed by a firmware update, to power value will be wrong again.

Patricen1, 2025-03-24:
> It's possible to use an external converter to multiply power by 1000, the issue to implement directly in Z2M is if the device is fixed by a firmware update, to power value will be wrong again.

Fully understood, as I do have no workaround for the moment can you please give me some clues regarding external converter?
Thx

dduransseau, 2025-03-24:
You could use this [converter](https://gist.github.com/dduransseau/6232142b7a59280e76b705b512dffaf5). I don't remember what's the issue with energy reporting but don't see issue on value reporting.

Patricen1, 2025-03-25:
> You could use this [converter](https://gist.github.com/dduransseau/6232142b7a59280e76b705b512dffaf5). I don't remember what's the issue with energy reporting but don't see issue on value reporting.

Thanks a lot, I do have this device for years but did never succeeded getting the right instantaneous power.
@script0803 can you initiate a PR ?
This looks like to be the same issue as [https://github.com/Koenkk/zigbee2mqtt/issues/22429](url)

Patricen1, 2025-03-25:
I did not realized it was a firmware related issue, thought it was a driver issue. So let's keep it running with external converter ! BTW I do have the same device, still in its box, but different brand -Zemismart-, I should check the operation !

dduransseau, 2025-03-25:
Zemismart is not exactly the same model since it's the Tuya version. Anyway it's correctly supported by z2m.

script0803, 2025-03-26:
Sorry for the wait, everyone. I’ve been busy with the development of ESP version Zigbee 3.0 devices and didn’t manage to follow up on this matter in time.
 Regarding the external converter, I tried to open a PR, https://github.com/Koenkk/zigbee-herdsman-converters/pull/8463but the admin’s response was that the function for "forcefully converting power divisor to 1" has too much duplicate code compared to existing functions, so my PR was rejected. 
Honestly, I’m not sure what to do now, because I can’t modify the original electrical_measurement: as it would affect all devices using that function.

script0803, 2025-03-27:
Hello everyone, I found a solution yesterday and submitted a PR, which was merged by the admin today. So please wait for the next Zigbee2MQTT version update.
 In addition, we’ve added new Zigbee 3.0 devices. The U00 is a product developed based on ESP32-H2, offering more features and stronger performance compared to the U01.

mrmaximas, 2025-03-27:
@script0803 hi! I have SPM01-D2TZ Tuya model _TZE200_bcusnqt8 with Zemismart whitelabel.  Is this your product? 
Maybe you have a firmware for this device that does not spam the network? 
I currently have ZigBee Module: V1.0.4 MCU Module: V3.0.3. Is it latest FW?

script0803, 2025-03-28:
Hi, @mrmaximas.
The SPM01-D2TZ is indeed our product. Zemismart is our sales partner. For the Tuya model _TZE200_bcusnqt8, the ZigBee Module: V1.0.4 is already the latest firmware for the SoC, and it is updated by Tuya. 

According to feedback from Zemismart, we have pushed specific MCU firmware for OTA to you. Due to the large number of DP points on our devices, Tuya’s SoC can cause interference with the Zigbee network. We are actively discussing this issue with Tuya. 

Until Tuya addresses the problem, we have reduced the reporting frequency to minimize the interference to the ZigBee network (the OTA firmware we pushed to you). 

Meanwhile, if you are using zigbee2mqtt, we recommend setting the debounce to 1s for each problematic device to effectively filter out duplicate information.

script0803, 2025-03-28:
After discussing with Tuya engineers, we initially believe the cause of this phenomenon is that Tuya’s Zigbee reports each attribute separately during the report process.
whereas our MCU, when communicating with Tuya’s SoC, only transmits one data packet at a time, containing the values of all DP points, while Tuya splits this during the report.
This leads to a large number of Zigbee messages being generated at the same time.
In our own U00/U01/U02 devices, since they comply with the Zigbee 3.0 standard protocol, each report only sends 1-2 packets, so this phenomenon does not occur.

mrmaximas, 2025-03-28:
@script0803 Thanks for the detailed information. It's nice to see a concerned manufacturer's representative here. Is there any way to flash normal no-spam firmware to SPM01-D2TZ via OTA? I don't plan to use these devices inside TUYA in the future, I keep TUYA hub only for updates and firmware captures.

[9 more replies]

closed 2025-07-11 by github-actions[bot]