Koenkk/zigbee2mqtt #19707: IKEA LED2201G8 can't be added

nbduck, 2023-11-14:
### What happened?

The IKEA LED2201G8 lightbulb can't be added to Zigbee2MQTT although it should be supported (https://www.zigbee2mqtt.io/devices/LED2201G8.html). I can detect it via touchlink and reset it. Even after resetting the bulb manually it won't automatically connect.

### What did you expect to happen?

After resetting the bulb will automatically connect.

### How to reproduce it (minimal and precise)

Reset the bulb, try to connect it.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 Dongle

lnorbi, 2023-11-19:
I have the same issue. Bulb properly reset, trying to connect, but nothing coming up on Z2M logs or anywhere else.

nbduck, 2023-11-19:
> I have the same issue. Bulb properly reset, trying to connect, but nothing coming up in Z2M logs or anywhere else.

Maybe it will work with the December release of zigbee2mqtt. I decided to wait because the lightbulb just has been added: https://github.com/Koenkk/zigbee2mqtt/issues/19118

dmitrybabeshko, 2023-11-20:
Hello,

Same problem after firmware of the bulb was updated.
Initially it worked. After the bulb firmware was updated it continued to work.
But when tried to remove device and pair again it always fails to complete the interview. 

**Adapter**
CC1352P-2 (CC1352P2_CC2652P_launchpad_coordinator_20230507.zip)

**Zigbee2MQTT**
zigbee2MQTT version 1.33.2-dev (commit #c87ac9c)
zigbee-herdsman (0.23.0)

**Bulb**
Firmware build date 20230922
Firmware version 3.0.10

**Logs**
```
Zigbee2MQTT:info  2023-11-20 15:44:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8c6fb9fffedf5e73","ieee_address":"0x8c6fb9fffedf5e73","status":"failed"},"type":"device_interview"}'
2023-11-20T14:44:20.031Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,78,32,115,94]
2023-11-20T14:44:20.032Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,78,32,115,94]
2023-11-20T14:44:20.033Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [223,254,255,185,111,140,0,0,68]
2023-11-20T14:44:20.033Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,78,32,115,94,223,254,255,185,111,140,0,0,68]
2023-11-20T14:44:20.033Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [78,32,115,94,223,254,255,185,111,140,0,0] - 68
2023-11-20T14:44:20.034Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":8270,"extaddr":"0x8c6fb9fffedf5e73","parentaddr":0}
2023-11-20T14:44:20.034Z zigbee-herdsman:controller:log Device '0x8c6fb9fffedf5e73' joined
2023-11-20T14:44:20.034Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-11-20T14:44:20.035Z zigbee-herdsman:controller:log Device '0x8c6fb9fffedf5e73' accepted by handler
2023-11-20T14:44:20.035Z zigbee-herdsman:controller:log Device '0x8c6fb9fffedf5e73' is already in database with different networkAddress, updating networkAddress
2023-11-20T14:44:20.035Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-11-20T14:44:20.044Z zigbee-herdsman:controller:log Interview '0x8c6fb9fffedf5e73' start
Zigbee2MQTT:info  2023-11-20 15:44:20: Starting interview of '0x8c6fb9fffedf5e73'
Zigbee2MQTT:info  2023-11-20 15:44:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8c6fb9fffedf5e73","ieee_address":"0x8c6fb9fffedf5e73","status":"started"},"type":"device_interview"}'
2023-11-20T14:44:20.094Z zigbee-herdsman:controller:device:log Interview - start device '0x8c6fb9fffedf5e73'
2023-11-20T14:44:20.095Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0x8c6fb9fffedf5e73', already got it
2023-11-20T14:44:20.095Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":8270,"nwkaddrofinterest":8270}
2023-11-20T14:44:20.095Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,78,32,78,32,36]
2023-11-20T14:44:20.101Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2023-11-20T14:44:20.102Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2023-11-20T14:44:20.102Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2023-11-20T14:44:20.102Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2023-11-20T14:44:20.102Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-11-20T14:44:30.104Z zigbee-herdsman:controller:device:log Interview - active endpoints request failed for '0x8c6fb9fffedf5e73', attempt 1
2023-11-20T14:44:30.105Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":8270,"nwkaddrofinterest":8270}
2023-11-20T14:44:30.106Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,78,32,78,32,36]
2023-11-20T14:44:30.110Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2023-11-20T14:44:30.110Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2023-11-20T14:44:30.111Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2023-11-20T14:44:30.112Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2023-11-20T14:44:30.113Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-11-20T14:44:40.116Z zigbee-herdsman:controller:device:log Interview - active endpoints request failed for '0x8c6fb9fffedf5e73', attempt 2
2023-11-20T14:44:40.117Z zigbee-herdsman:controller:device:log Interview - quirks check for 'TRADFRI bulb E27 WS globe 1055lm'-'IKEA of Sweden'-'Router'
2023-11-20T14:44:40.120Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2023-11-20T14:44:40.124Z zigbee-herdsman:controller:device:log Interview - failed for device '0x8c6fb9fffedf5e73' with error 'Error: Interview failed because can not get active endpoints ('0x8c6fb9fffedf5e73')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:636:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:488:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:579:17)'
2023-11-20T14:44:40.125Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-11-20T14:44:40.133Z zigbee-herdsman:controller:error Interview failed for '0x8c6fb9fffedf5e73 with error 'Error: Interview failed because can not get active endpoints ('0x8c6fb9fffedf5e73')'
Zigbee2MQTT:error 2023-11-20 15:44:40: Failed to interview '0x8c6fb9fffedf5e73', device has not successfully been paired
```

InnerBushman, 2023-12-26:
I can confirm. The bulb is discoverable via Touchlink but not a peep in the logs of Z2M.

Adapter - Sonoff_Zigbee_3.0_USB_Dongle_Plus
Zigbee2MQTT version - 1.34.0
Coordinator type - zStack3x0
Coordinator revision - 20221226

Did anyone had any luck with this bulb?
Is the issue identified on which side is the problem? Is it the firmware of the bulb or something with z2m?

Is there anyway I can monitor low level communications while the bulb is in pairing mode other than the logs of z2m?
I'm not a programmer but maybe I can somehow assist in diagnosing the issue?

lubiezunia, 2023-12-30:
I can confirm. It says that pairing was successful and after few seconds disconnects.

Adapter - Skyconnect
Zigbee2MQTT version - 1.34.0

InnerBushman, 2024-01-01:
That's different as in my case as it doesn't even try to connect with Sonoff.

roman-16, 2024-01-06:
I doesn't work for me either. I get no signs of pairing.

rwjack, 2024-01-16:
Is there anyone that actually managed to connect this bulb?

M4RKU5-C0D3, 2024-01-16:
My LED2201G8 just popped up after updating to the latest version 1.35.1 ✅ 

----
Zigbee2MQTT-Version: 1.35.1
Coordinator-Typ: zStack3x0 (SONOFF ZigBee 3.0 USB DONGLE PLUS)
Coordinator-Version: 20210708
Frontend-Version: 0.6.151
Zigbee-herdsman-converters version: 18.9.0
Zigbee-herdsman version: 0.30.0

rwjack, 2024-01-16:
I just updated to 1.35.1 - I'll give it a few more tries during the day.

I also had a bunch of issues trying to connect the Somrig and Rodret buttons, they just weren't sending initial signals (I can see the red light not flashing). I think this is a problem with Ikea devices and not related to Z2M.

M4RKU5-C0D3, 2024-01-16:
Maybe worth mentioning: I did a factory reset on the LED2201G8 over Touchlink before the update without success. But maybe this did the trick with 1.35.1 🤔

Also the manual stated that the LED2201G8 was already pre-paired with the bundled [E2001/E2002](https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002). Maybe this could be an initial issue requiring a factory reset.

vuminhkh, 2024-01-17:
Hello, it's not working for me neither with 1.35.1, I tried 5, 6 times factory reset with touchlink, perhaps it works better with the sonoff dongle plus ?. I have a skyconnect, below is my configuration :

`Zigbee2MQTT version
[1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [9eaaa0f](https://github.com/Koenkk/zigbee2mqtt/commit/9eaaa0f)
Coordinator type
EZSP v9
Coordinator revision
7.1.1.0 build 273
Coordinator IEEE Address
0x048727fffe22d557
Frontend version
0.6.151
Zigbee-herdsman-converters version
18.9.0
Zigbee-herdsman version
0.30.0`

Desiderius42, 2024-01-17:
I have basically the same configuration as [vuminhkh](https://github.com/vuminhkh)  I can touchlink scan/detect/identify and reset but other than that I cannot connect to the bulb :-( anything else I can try ?

vuminhkh, 2024-01-17:
@Desiderius42 yes, i'm having the same issue as you, touchlink ok, reset ok but the device is not paring up.  I've just tested with the sonoff dongle Plus, it does not work neither. The last test I think of is to update the firmware of the bulb with my gateway Ikea and then try to pair again. I'll keep you updated.

M4RKU5-C0D3, 2024-01-17:
Maybe this helps:
[image]

vuminhkh, 2024-01-17:
Hi, I updated to the latest firmware version for the bulb (same version as you @M4RKU5-C0D3 , it's still not working.
```
Zigbee2MQTT:info  2024-01-17 21:19:23: Start Touchlink identify of '0x60efabfffe070003' on channel 11
Zigbee2MQTT:info  2024-01-17 21:19:26: MQTT publish: topic 'zigbee2mqtts/bridge/response/touchlink/identify', payload '{"data":{"channel":11,"ieee_address":"0x60efabfffe070003"},"status":"ok","transaction":"mvy95-4"}'
Zigbee2MQTT:info  2024-01-17 21:19:30: Start Touchlink factory reset of '0x60efabfffe070003' on channel 11
Zigbee2MQTT:info  2024-01-17 21:19:35: Successfully factory reset device through Touchlink
Zigbee2MQTT:info  2024-01-17 21:19:35: MQTT publish: topic 'zigbee2mqtts/bridge/response/touchlink/factory_reset', payload '{"data":{"channel":11,"ieee_address":"0x60efabfffe070003"},"status":"ok","transaction":"mvy95-5"}'
```
After the factory reset, the bulb is not joining neither with Sonoff, nor Skyconnect, I have no idea left to try.

vuminhkh, 2024-01-17:
[image]
[image]

Very curious though, when I execute touchlink, I have an existing paired bulb to appear on the channel 15, the unpaired LED2201G8 appears on channel 11 (in this configuration, identify ok, reset ok but no interview), I rescan and sometimes my existing paired bulb appear on the channel 11, the LED2201G8 appears on channel 15 (in this configuration identify fail with timeout). How can it be possible ? I use the default channel (so it should be 11.) and added a lot of old IKEA devices successfully in the past.

vuminhkh, 2024-01-18:
Hello, just to let you know that on / off 6 times trick to factory reset works for me. Touchlink reset just does not work with this device.

Desiderius42, 2024-01-20:
very strannge I've tried the on/off 6 times trick but with no progress @vuminhkh can you tell us how you update the bulb ? it is from touchlink? 

`Version de Zigbee2MQTT
[1.35.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.1) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Type de coordinateur
EZSP v9
Révision du coordinateur
7.1.1.0 build 273
Adresse IEEE du Coordinateur
0xe0798dfffe0a910f
Version de l'interface
0.6.151
Zigbee-herdsman-converters version
18.9.0
Zigbee-herdsman version
0.30.0`

rwjack, 2024-01-20:
Same here, on/off 6 times basically does nothing. I can't tell if the brightness changes (enters pairing mode), because it is too bright to look at initially, but it doesn't seem to do anything.

I don't even see the light after the touchlink scan.

vuminhkh, 2024-01-20:
It's not easy the on/off 6 times trick, you should not let the bulb completely light up, switch it off right just before it lights up completely. You should install the bulb like in this video to turn on and off rapidly https://www.youtube.com/watch?v=mJm9YpPrGzk 
With a little bit of practice, I can factory reset very easily.  If the bulb does not blink then perhaps you did not do it right, because this factory reset is IKEA built in. I updated the bulb to the latest version with my gateway Tradfri (I did not resell it immediately after switching to Z2M just in case like this ;-) ) but I don't really think it has anything to do with the firmware version of the bulb because it still didn't work for me touchlink reset with the latest version.

rwjack, 2024-01-20:
Yeah I've seen that video and I can speed press my wall switch quite quickly, but something isn't adding up...

Can't believe how all these big companies are trying to be smart, pre-pair stuff, make factory resets so awkward, and all that on an open source standard. The exact opposite of smart.

> Hello, just to let you know that on / off 6 times trick to factory reset works for me. Touchlink reset just does not work with this device.

@vuminhkh, just to confirm, you are saying that only the reset works for you, but you still can't pair the bulb, or did you manage to connect the bulb to z2m as well?

vuminhkh, 2024-01-20:
I could pair the bulb with the factory reset.

vuminhkh, 2024-01-20:
The thing is in the video, they let the bulb lights up completely, I switch it off right when I see that it's going to light up, and wall switch is a bad idea, i spent hours on a wall switch to succeed like 2 times, i put the bulb then on a floor lamp, much more easily to switch on off and observe the bulb with this configuration.

vuminhkh, 2024-01-20:
I'm buying more bulbs this afternoon, i'll make a video for you, my family thought that I was crazy when they saw me stay on a wall switch for hours.

nbduck, 2024-01-21:
I finally made it work aswell. To pair the bulb I finally got a way that works 90%. 

1. Turn the light on.
2. Turn the light off and hold that state for 1-2 seconds.
3. Turn the light on and instantly off again. It should flash a very short time.
4. Turn it off and stay for 1-2 seconds.
5. Repeat 5 or 6 mor times.

After that the light should pulse and it has been resetted.

nilsburg, 2024-01-22:
> I finally made it work aswell. To pair the bulb I finally got a way that works 90%.
>
> 1. Turn the light on.
> 2. Turn the light off and hold that state for 1-2 seconds.
> 3. Turn the light on and [...]

Can confirm this works. Just tried and was able to pair de bulb with z2m succesfully. Thanks!

Desiderius42, 2024-01-22:
I can also confirm that @nbduck solution works.  I have to add that to simplify the process I have force accepting pairing until the end of the process. 

1. Turn the light on.
2. Turn the light off count to 2
3. Turn the light on and  off as soon as the light is on
4.  Repeat 5 or 6 more times step 2 and 3 
6.  turn on check that the light pulse and the the interview went ok.

Desiderius42, 2024-01-22:
I can also confirm that @nbduck solution works. I have to add that to simplify the process I have force accepting pairing until the end of the process.

Turn the light on.
Turn the light off count to 2
Turn the light on and off as soon as the light is on
Repeat 5 or 6 more times step 2 and 3
turn on check that the light pulse and the the interview went ok.

Thank you @vuminhkh for  finding the solution thank you all for finding a stable solution together.

rwjack, 2024-01-22:
This method does not work for me. Did you guys first update the bulb via the gateway? I didn't because I don't have one.

I tried Long off - Short on, Short off - Long on, short-short, long-long, couple of times, 5-6 times, 10+ times, nothing. Put on my shades, stared directly at the bulb, absolutely no change.

I'm returning this piece of garbage. Already wasted an accumulated hour or two of my life trying to connect a stupid bulb, which should be in pairing mode straight out the box.

Desiderius42, 2024-01-22:
@rwjack  whats a pity. 
It should be **long off very short on**.  you very close to making it.  ;-)

We are talking about the cheapest and most powerfull bulb we can find on the market. but it's true that until this morning it was a pain to put in place. 

Check at my proposal I have made it work twice now without problem.

Desiderius42, 2024-01-22:
I have created a pull request on the documentation https://github.com/Koenkk/zigbee2mqtt.io/pull/2512 

@Koenkk this could close this issue.

rwjack, 2024-01-22:
I also forgot to mention that I'm using 20 as my ZigBee channel, that could be an issue, but it shouldn't really.

I tried again, long off, very short on, repeat 5-6x, no avail.

closed 2024-01-22 by Koenkk

InnerBushman, 2024-01-24:
The bulb started to work after the update. Do we know what was the reason it did not pair?
I had no problems with factory reset in those bulbs but only this model didn't pair so far.
I don't want people to think they didn't pair it properly.
Can someone point out changes in the code / parameters which made this bulb work again?

imjuzcy, 2024-03-24:
FWIW, I did successfully pair it after lots of tries. Even though the reset by 6 toggles work, z2m just wouldn't detect the bulb at first. After a few tries, I decided to change the z2m Zigbee channel from 11 to 20, unplug the Zigbee dongle and replug it into another port, restarted the Home Assistant OS VM (in Proxmox), and enabled permit join by default. I had to repair everything, but this time the bulb paired after a reset of the bulb. Not sure how stable it'll be, we'll see.

Kraviecc, 2024-04-10:
> I can also confirm that @nbduck solution works. I have to add that to simplify the process I have force accepting pairing until the end of the process.
>
> Turn the light on. Turn the light off count [...]

For me, it was enough to repeat this solution 7 times (so I stopped turning OFF the bulb once it was toggled ON 7th time), of course keeping the rule of 2sec OFF, very quick ON. Thanks for the solution guys!

zoomtronicOR, 2024-05-13:
The lightbulb resets properly, but no interview in Z2M at all, tried more than 50 times, even put arduino and ssr to make reset circuit, https://www.coded-with-love.com/blog/ikea-tradfri-arduino-based-reset-station/ reset works every time, but no sign of pairing in z2m

precchia, 2024-10-22:
Funny... Issue closed and then people keep complaining it still does not work...
Can't pair my bulb either
I trived several times: 
Reset OK (light blinks)
Z2M starts pairing ain interview
And then device leaves the network.
Tried quit a few times, and with two different bulbs

I'm running Z2M 1.40.2 (from Home assistant Add-Ons)

What I got in the logs:

info 2024-10-22 22:31:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xd87a3bfffe0fbd22', payload '{"last_seen":"2024-10-22T22:31:42+02:00"}'
info 2024-10-22 22:31:43zh:controller: Interview for '0xd87a3bfffe0fbd22' started
info 2024-10-22 22:31:43z2m: Device '0xd87a3bfffe0fbd22' joined
info 2024-10-22 22:31:44z2m: Starting interview of '0xd87a3bfffe0fbd22'
info 2024-10-22 22:31:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd87a3bfffe0fbd22","ieee_address":"0xd87a3bfffe0fbd22"},"type":"device_joined"}'
info 2024-10-22 22:31:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd87a3bfffe0fbd22","ieee_address":"0xd87a3bfffe0fbd22","status":"started"},"type":"device_interview"}'
info 2024-10-22 22:31:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xd87a3bfffe0fbd22', payload '{"last_seen":"2024-10-22T22:31:48+02:00"}'
info 2024-10-22 22:31:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd87a3bfffe0fbd22","ieee_address":"0xd87a3bfffe0fbd22"},"type":"device_announce"}'
warning 2024-10-22 22:32:09z2m: Device '0xd87a3bfffe0fbd22' left the network
info 2024-10-22 22:32:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xd87a3bfffe0fbd22","ieee_address":"0xd87a3bfffe0fbd22"},"type":"device_leave"}'
error 2024-10-22 22:32:10zh:controller: Interview failed for '0xd87a3bfffe0fbd22 with error 'Error: DatabaseEntry with ID '34' does not exist'

wiwiki, 2024-11-27:
tried all, can't connect my two brand new Ikea lights, I can't even see them.

electronicm, 2024-12-19:
I had similar problems. The LED2201G8 could be connected with a successful interview, but was then no longer accessible. (no response from the device)
Even resetting the LED2201G8 several times and pairing it with the UZG-01 as a coordinator did not help.

Since there were several reports of problems connecting IKEA devices with third-party coordinators/routers, I tried pairing directly with an E2206 (INSPELNING smart plug) as a router in between.
Here the communication works without any problems. (Config and OTA)

So I think it's once again a manufacturer-specific thing to force people to use their own products.