Koenkk/zigbee2mqtt #19711: [New device support]: lumi.light.acn128 - Aqara Spotlight-T3

MiAutomations, 2023-11-14:
### Link

https://www.aqara.cn/Spotlight-T3_overview

### Database entry

{"id":64,"type":"Router","ieeeAddr":"0x54ef441000a7cf19","nwkAddr":37694,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn128","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.light.acn128","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20221123"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":27,"hwVersion":1,"dateCode":"20221123","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1699978643064,"defaultSendRequestWhen":"immediate"}

### Comments

Hello Team 

Can you please add this new device from the Aqara to your list of supported devices?

Once this is a light I try to get the Color Temperature  settings but without success ...  
[image]

Please let me know if you need any thing more from my side in order to get this device supported 

Thank you  

### Supported color modes

Color Temperature

Koenkk, 2023-11-14:
What is the model number of this device? I expect something like ZNTGMK11LM

MiAutomations, 2023-11-14:
> What is the model number of this device? I expect something like ZNTGMK11LM

The model number is: TDL01ML

Koenkk, 2023-11-15:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-11-15 by Koenkk