Koenkk/zigbee2mqtt #19713: TypeError: Cannot create property 'port' on boolean 'true'

FalkWinkler, 2023-11-14:
### What happened?

I have errors when using zigbee2mqqtt with Kubernetes.

When I use the properties of ZIGBEE2MQTT_CONFIG_FRONTEND, errors occur. It does not matter whether I use ZIGBEE2MQTT_CONFIG_FRONTEND_PORT or ZIGBEE2MQTT_CONFIG_FRONTEND_URL.

Everything works fine without these two properties. 

### How to reproduce it (minimal and precise)

`apiVersion: apps/v1
kind: StatefulSet
metadata:
  name: zigbee2mqtt
spec:
  selector:
    matchLabels:
      app: zigbee2mqtt
  serviceName: zigbee2mqtt
  template:
    metadata:
      labels:
        app: zigbee2mqtt
    spec:
      containers:
      - name: zigbee2mqtt
        image: ghcr.io/koenkk/zigbee2mqtt:latest
        resources:
          limits:
            memory: "128Mi"
            cpu: "500m"
        ports:
        - containerPort: 8080
        volumeMounts:
          - name: data
            mountPath: /data

        env:
          - name: ZIGBEE2MQTT_CONFIG_MQTT_BASE_TOPIC
            value: "zigbee2mqtt"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_INCLUDE_DEVICE_INFORMATION
            value: "true"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_KEEPALIVE
            value: "60"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_REJECT_UNAUTHORIZED
            value: "true"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_SERVER
            value: "mqtt://mosquitto.home-automation.svc.cluster.local"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_VERSION
            value: "5"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_USER
            value: "admin"
          - name: ZIGBEE2MQTT_CONFIG_MQTT_PASSWORD
            value: admin@123
          - name: ZIGBEE2MQTT_CONFIG_PERMIT_JOIN
            value: "false"
          - name: ZIGBEE2MQTT_CONFIG_SERIAL_PORT
            value: "tcp://ser2sock.home-automation.svc.cluster.local:10000"
          - name: ZIGBEE2MQTT_CONFIG_SERIAL_BAUDRATE
            value: "115200"
          - name: ZIGBEE2MQTT_CONFIG_SERIAL_DISABLE_LED
            value: "false"
          - name: ZIGBEE2MQTT_DATA
            value: /data
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_HOMEASSISTANT_DISCOVERY_TOPIC
            value: homeassistant
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_HOMEASSISTANT_LEGACY_ENTITY_ATTRIBUTES
            value: "false"
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_HOMEASSISTANT_LEGACY_TRIGGERS
            value: "false"
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_HOMEASSISTANT_STATUS_TOPIC
            value: homeassistant/status
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_LAST_SEEN
            value: ISO_8601
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_LEGACY_API
            value: "false"
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_LEGACY_AVAILABILITY_PAYLOAD
            value: "false"
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_LOG_LEVEL
            value: info
          - name: ZIGBEE2MQTT_CONFIG_ADVANCED_LOG_OUTPUT
            value: '["console"]'
          - name: ZIGBEE2MQTT_CONFIG_AVAILABILITY_ACTIVE_TIMEOUT
            value: "60"
          - name: ZIGBEE2MQTT_CONFIG_AVAILABILITY_PASSIVE_TIMEOUT
            value: "2000"
          - name: ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_LEGACY
            value: "false"
          - name: ZIGBEE2MQTT_CONFIG_DEVICE_OPTIONS_RETAIN
            value: "true"
          - name: ZIGBEE2MQTT_CONFIG_EXPERIMENTAL_NEW_API
            value: "true"
          - name: ZIGBEE2MQTT_CONFIG_HOMEASSISTANT
            value: "true"
          # - name: ZIGBEE2MQTT_CONFIG_FRONTEND_PORT
          #   value: "8080"
          # - name: ZIGBEE2MQTT_CONFIG_FRONTEND_URL
          #   value: "https://zigbee2mqtt.test"

  volumeClaimTemplates:
  - metadata:
      name: data
    spec:
      accessModes: [ "ReadWriteOnce" ]
      storageClassName: ceph-block
      resources:
        requests:
          storage: 1Gi`

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0

### Adapter

ser2sock tcp

### Debug log

main
Using '/data' as data directory
main
TypeError: Cannot create property 'port' on boolean 'true'
main
at /app/lib/util/settings.ts:411:41
main
at Array.forEach (<anonymous>)
main
at iterate (/app/lib/util/settings.ts:389:26)
main
at /app/lib/util/settings.ts:428:21
main
at Array.forEach (<anonymous>)
main
at iterate (/app/lib/util/settings.ts:389:26)
main
at /app/lib/util/settings.ts:428:21
main
at Array.forEach (<anonymous>)
main
at iterate (/app/lib/util/settings.ts:389:26)
main
at /app/lib/util/settings.ts:428:21

chaoslion, 2023-11-16:
Hey, 

it is a bit weird with the new default configuration.yaml. 
I stumbled on this aswell. Just invert the default `frontend: true` setting via

```bash
docker run -eZIGBEE2MQTT_CONFIG_FRONTEND=false -eZIGBEE2MQTT_CONFIG_FRONTEND_URL=https://ziggy ...
```

FalkWinkler, 2023-11-17:
That works, thank's.

closed 2023-11-17 by FalkWinkler