Koenkk/zigbee2mqtt #19715: TuYa TS0601_dimmer_1 does not support toggle command

krispek, 2023-11-14:
### What happened?

I have a [TuYa TS0601_dimmer_1](https://www.zigbee2mqtt.io/devices/TS0601_dimmer_1.html) dimmer, which should accept {"state": "TOGGLE"}. However, when I use this command, I got the following error:

````Publish 'set' 'state' to 'dimmer' failed: 'Error: Expected one of: ON, OFF, got: 'TOGGLE'' ````

Which implies that the toggle command does not work. Is this a bug, or a documentation issue? 

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

ikarisan, 2023-11-15:
You mean the "_TZE200_la2c2uo9"? 
I opened an issue in January 2023: https://github.com/Koenkk/zigbee2mqtt/issues/16334

Koenkk, 2023-11-15:
Thats because the TuYa api doesnt have such command.

ikarisan, 2023-11-15:
Hi @Koenkk !

That's very interesting. 
Is there a difference between dimmers and switches? I'm asking because my Tuya switch actuators (for example JR-ZDS01 "1 gang mini switch") support the TOGGLE command. My Tuya developer account has unfortunately expired, so I can no longer debug anything. But is this something that could be solved with an external converter? Or is my workaround via Nodered already the optimum?

Koenkk, 2023-11-15:
`JR-ZDS01` uses the standard Zigbee API , TS0601_dimmer_1 uses the TuYa api.

krispek, 2023-11-16:
Given that the Tuya-API does not support this, I would propose to change the documentation, because currently it does state that the toggle command is supported.

ikarisan, 2023-11-19:
@krispek I think I tried that out once too. But the problem was, that I had to wait for the first message from the dimmer device to get the current state. So, after a powerloss or update/restart of Z2M the state of the dimmer may be unknow/null and you cannot change the state.

closed 2024-06-17 by github-actions[bot]