Koenkk/zigbee2mqtt #19718: TS0505B_1_1 Do not disturb function

rodrigofragadf, 2023-11-14:
### What happened?

I have some TS0505B_1_1 lamps, before at exposes there was a do not disturb function and it worked well, but a few days ago this function disappeared and now I have the lights on after a power outage. How do I get the do not disturb function back?

### What did you expect to happen?

Bring back the do not disturb function

### How to reproduce it (minimal and precise)

Remove the lamp from the power supply and plug it back in

### Zigbee2MQTT version

 1.33.2

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

hibbeldi, 2023-11-20:
Same here with TS0505B_1_2 (but I'm not sure whether ist was there before? Not mentioned here https://www.zigbee2mqtt.io/devices/TS0505B_1_1.html nor https://www.zigbee2mqtt.io/devices/TS0505B_1_2.html )

rodrigofragadf, 2023-11-20:
> Same here with TS0505B_1_2 (but I'm not sure whether ist was there before? Not mentioned here https://www.zigbee2mqtt.io/devices/TS0505B_1_1.html nor https://www.zigbee2mqtt.io/devices/TS0505B_1_2.html )

Before the lamps were identified as another model, I don't remember exactly but I think it was the TS0501B. It's frustrating to have something that was already working well and suddenly stop working, this is an essential option for lamps.

rotem657, 2023-11-30:
Same on my side :(
It worked before and changed it's model.

riosdev, 2023-12-14:
Same problem here. The ones registered as TS0505B_1 still work, but if I re-pair them, they get registered as TS0505B_1_1.

rodrigofragadf, 2023-12-27:
Isn't there any way to get dnd back?

rotem657, 2024-01-13:
After updating Z2M, my devices changed to TS0505B_2_1, but the "do_not_disturb" mode option is still not available.
Please, any update?

Koenkk, 2024-01-15:
Added it back!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-15 by Koenkk

djcrawleravp, 2024-05-17:
Has anyone know where can I get the TS0505B_1_2 version? I'm still stucket in 1.33.1 because of this, my adaptive lights are not working, they light up in minimum and doesn't adapt or take too long to do so