Koenkk/zigbee2mqtt #19719: Aqara D1 3 gang not reporting state

yayeetye, 2023-11-15:
### What happened?

I have multiple Aqara D1 2gang and 3gang switches. with neutral. 
all the 3gang switches are not updating in z2m or in home assistant. i can press the switch on and off just fine and control them in home assistant fine.
but if i turn on the switch physically. it is not updating the state in z2m.
[image]
[image]

### What did you expect to happen?

I expect to see that the switch is switched on in z2m after i press them physically.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

 20230507

### Adapter

Sonoff Dongle P

yayeetye, 2023-11-15:
#14673 and #16524
could be same issue

MzA80, 2023-11-26:
I have a same issue here and can't find any solution at the moment. Appreaciate if Z2M can roll an update.

simclancy, 2023-11-30:
Yep have this issue as well

ASAujla, 2024-01-23:
Sorry, I have had the same problem since 2022.

leonly0224, 2024-02-05:
same problem, and the power report of this device is not working too

chun950yi, 2024-05-07:
I have the same issue

yayeetye, 2024-06-07:
issue still exist

djurny, 2024-12-05:
Hi there,
In our previous house we had several D1 (**no** neutral) switches, but only the 1 and 2 gang variants. I used most of them in their `decoupled` mode, where a press of the switch would not trigger a relay state change - this allowed me to put "smart lights" behind the switches. I would expect the 3 gang (**with** neutral) should also have this mode, if so, does the behavior change at all? Perhaps that new firmware will not send out switch presses if mode is set to `control_relay`?
Groetjes,