Koenkk/zigbee2mqtt #19721: [New device support]: Sonoff NS Panel PRO as a Router device

MiAutomations, 2023-11-15:
### Link

https://itead.cc/product/sonoff-nspanel-pro-smart-home-control-panel/

### Database entry

{"id":79,"type":"Router","ieeeAddr":"0xe0798dfffeb25098","nwkAddr":18228,"manufId":4098,"manufName":"SONOFF","powerSource":"Unknown","modelId":"NSPanelP-Router","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":7,"inClusterList":[0,3],"outClusterList":[0,3],"clusters":{"genBasic":{"attributes":{"modelId":"NSPanelP-Router","manufacturerName":"SONOFF"}}},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1700038992754,"defaultSendRequestWhen":"immediate"}

### Comments

Hello Team 

With the new firmware version available for the NSpanel PRO Sonoff Device The device have the possibility ack as zigbee router

I try to add this device as a Router to Z2M but appear as not supported 

I'm sharing the  database entry of this device in order if you can help me to get this device supported in the Z2M list os devices 

Please let me know if you need anything more from my side 

Thank you in advance for your support and best regards

Mike1969n, 2023-11-28:
I also wish this integration. I also can help.

leroyvandijk, 2023-12-22:
Would also like to see this device supported

pvklink, 2023-12-29:
I also like this device supported

Koenkk, 2023-12-30:
Added! Will be supported in the 1 January release.

closed 2023-12-30 by Koenkk

cobch, 2024-01-27:
Is it possible to connect it out of the box to z2m? Want to buy it but want to be sure it will work (-;

EnergieZ, 2024-03-28:
> Is it possible to connect it out of the box to z2m? Want to buy it but want to be sure it will work (-;

I have the same question ! I just buy it and install it, update to version 3.0, and configure it as a gateway. 
But now i'm stuck ! 
I've configure z2m with tcp and the ip of the nspannel, but it does not work

mihaimdinca, 2024-04-12:
I set the NSPanel Pro to router mode but not sure how to add it to ZigBee2MQTT. I do not see any option to start pairing mode on the router. It does not work just by pushing the all join button in z2m. Any suggestion would be appreciated. Thank you!

mihaimdinca, 2024-04-12:
Nevermind. I found it. Another problem though: the device drops from the network very often. No idea why. Any suggestions?

mihaimdinca, 2024-05-10:
Apparently, it works well now with the latest version of ZigBee2MQTT.

Guyvxt, 2024-05-14:
Apologies, my update script wasn't working and I was running an older version. Updated properly and now showing as supported. thanks

lediable, 2024-07-10:
> Nevermind. I found it. Another problem though: the device drops from the network very often. No idea why. Any suggestions?

Sorry it's been a few months but I really couldn't understand how it works

how did you use nspanel pro as a router ( or coordinator? ) on zigbee2mqtt?

What settings did you use? What port did you use? 

Thanks for your future reply

MiAutomations, 2024-07-10:
> Sorry it's been a few months but I really couldn't understand how it works
>
> how did you use nspanel pro as a router ( or coordinator? ) on zigbee2mqtt?
>
> What settings did you use? What port did you [...]

Hello 

You need to change the router or coordinator from the ewelink APP after this you go to your Z2M UI and click to add a new device and on the device  you selected the paring mode

lediable, 2024-07-10:
> Hello
>
> You need to change the router or coordinator from the ewelink APP after this you go to your Z2M UI and click to add a new device and on the device you selected the paring mode

yes that i have to choose coordinator or router i knew that. i just don't know what to put in the zigbee2mqtt configuration. without configuration it doesn't start, so i can't access this UI

I'm using home assistant

BaerHub, 2024-09-22:
im also unable to start zigbee2mqtt, in the serial config i dont know what port to add?

BaerHub, 2024-09-22:
> yes that i have to choose coordinator or router i knew that. i just don't know what to put in the zigbee2mqtt configuration. without configuration it doesn't start, so i can't access this UI
>
> I'm [...]

did you managed to get it working?

pvklink, 2025-01-10:
> Added! Will be supported in the 1 January release.

is this z2m 2.0 or 1,42

mihaimdinca, 2025-01-11:
Not sure I understand correctly. So to use the device as a router was possible for some time now, is it possible to use it as a coordinator as well?

pvklink, 2025-01-11:
yes, i have HASS running with 4 instances of Z2M each with an own coordinator!
they have each there own IP, port, and mqtt, base topic
IN hass you get 4 bridges with eacht there own devices. On the frontend you only see all devices as usual.

2 RPI connected zigbee sonoff stick with a each a docker z2m
2 NSPANELS with z2m running and mqtt ...

mihaimdinca, 2025-01-11:
So the z2m instance that uses the nspanel as coordinator is installed on the nspanel itself? Can you point me in the direction on how to do this? Is it like an Android app?

mihaimdinca, 2025-01-11:
I currently run the addon for home assistant with a single coordinator - Sonoff ZigBee Dongle P

pvklink, 2025-01-11:
> I currently run the addon for home assistant with a single coordinator - Sonoff ZigBee Dongle P

And yes i added some apps to my panel
the pro_tools
a new loader (so its rooted), 
homeassistant
z2m app via the latest beta pro tools

see 
https://github.com/seaky/nspanel_pro_tools_apk
https://github.com/seaky/nspanel_pro_tools_apk?tab=readme-ov-file#manual-2x-version
https://github.com/seaky/nspanel_pro_tools_apk/discussions/164

https://blakadder.com/nspanel-pro-sideload/
https://blakadder.com/nspanel-pro-secrets/
https://www.youtube.com/watch?v=c1Dqdz8yHDc

https://www.apkmirror.com/apk/home-assistant/home-assistant/

pvklink, 2025-01-11:
more coordinators arent needed, just playing arround, have 70 devices like to do shit stuff

mihaimdinca, 2025-01-11:
I have a two floor home and experiencing some problems regarding connectivity. I do have a sufficient number of routers but some may be of poor quality (woox and ikea bulbs and sonoff zbmini relays) or may not be working well with aqara end devices. Anyway I am considering options and one if them may be to use a coordinator for each floor.

mihaimdinca, 2025-01-11:
I guess, testing each will be the way forward.

pvklink, 2025-01-11:
they all use the same MQTT, but with a different base topic