Koenkk/zigbee2mqtt #19724: [New device support]: IKEA ORMANÄS

FakeTruth, 2023-11-15:
### Link

https://www.ikea.com/nl/nl/p/ormanas-led-strip-smart-gekleurd-en-wit-spectrum-40497395/

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x287681fffe6b415f","nwkAddr":33184,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"ORMANAS LED Strip","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{"lightingColorCtrl":{"attributes":{"colorTempPhysicalMin":250,"colorTempPhysicalMax":454,"colorCapabilities":31,"colorMode":1,"colorTemperature":370}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":1,"onTime":0}},"genIdentify":{"attributes":{"identifyTime":0}},"genOnOffSwitchCfg":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":106,"hwVersion":1,"dateCode":"20230512","swBuildId":"1.1.10","zclVersion":8,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1700062830540,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the new device page but got stuck at exposing on/off, brightness and color functionality.

Creating the javascript file for the new device marked it as 'supported' but I'm unable to turn it on/off, change brightness or colors.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ORMANAS LED Strip'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ormanas', // Vendor model number, look on the device for a model number
    vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
    description: 'ORMANAS LED Strip', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.brightness, fz.color_colortemp], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.light_brightness()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [250, 454]}),
};

module.exports = definition;
```

### Supported color modes

20 colors and white lights

### Color temperature range

{"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}

Koenkk, 2023-11-15:
Can you provide the debug log when turning on/off the device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

FakeTruth, 2023-11-15:
The ORMANÄS does not have a button, but this is what was logged when I plugged it in a power socket:
```
debug 2023-11-15 21:54:50: Device '0x287681fffe6b415f' reconnected
info  2023-11-15 21:54:50: MQTT publish: topic 'zigbee2mqtt_upstairs/0x287681fffe6b415f/availability', payload 'online'
debug 2023-11-15 21:54:50: Device '0x287681fffe6b415f' announced itself
info  2023-11-15 21:54:50: MQTT publish: topic 'zigbee2mqtt_upstairs/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe6b415f","ieee_address":"0x287681fffe6b415f"},"type":"device_announce"}'
info  2023-11-15 21:54:50: MQTT publish: topic 'zigbee2mqtt_upstairs/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x287681fffe6b415f"},"type":"device_announced"}'
```

After a while this was also logged:
```
debug 2023-11-15 21:58:08: Received Zigbee message from '0x287681fffe6b415f', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":16842768,"imageType":10244,"manufacturerCode":4476}' from endpoint 1 with groupID 0
warn  2023-11-15 21:58:08: Received message from unsupported device with Zigbee model 'ORMANAS LED Strip' and manufacturer name 'IKEA of Sweden'
warn  2023-11-15 21:58:08: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

Thanks for your help

Fleshi1981, 2023-11-16:
@FakeTruth, I can also add the device to Zigbee2mqtt but can't control ik via Home Assistant and then I mean only to turn it on or off. For changing the colours I understand that I needs to be a supported device. But can you control it via Home assistant?

Koenkk, 2023-11-16:
`Received message from unsupported device with Zigbee model` indicates that your ext converter has not been loaded, please check your configuration.

FakeTruth, 2023-11-16:
> @FakeTruth, I can also add the device to Zigbee2mqtt but can't control ik via Home Assistant and then I mean only to turn it on or off. For changing the colours I understand that I needs to be a supported device. But can you control it via Home assistant?

As an 'unsupported device' without an external converter the device exposes nothing for me, and doesn't even show up in Home Assistant:
[image]

> `Received message from unsupported device with Zigbee model` indicates that your ext converter has not been loaded, please check your configuration.
You're right, I had disabled the external converter at that time.

With the external converter as described above, I get this in the log after unplugging and plugging the LED strip:
`` 
debug 2023-11-16 20:19:25: Device '0x287681fffe6b415f' announced itself
info  2023-11-16 20:19:25: MQTT publish: topic 'zigbee2mqtt_upstairs/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe6b415f","ieee_address":"0x287681fffe6b415f"},"type":"device_announce"}'
info  2023-11-16 20:19:25: MQTT publish: topic 'zigbee2mqtt_upstairs/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x287681fffe6b415f"},"type":"device_announced"}'
debug 2023-11-16 20:19:27: Retrieving state of '0x287681fffe6b415f' after reconnect
```

And this is exposed:
![image](https://github.com/Koenkk/zigbee2mqtt/assets/5101779/8bb5f333-841d-416d-972b-db43ef2b68e0)

However all buttons result in errors:
```
debug 2023-11-16 20:22:30: Received MQTT message on 'zigbee2mqtt_upstairs/0x287681fffe6b415f/get' with data '{"state":""}'
error 2023-11-16 20:22:30: No converter available for 'state' ("")
debug 2023-11-16 20:22:30: Received MQTT message on 'zigbee2mqtt_upstairs/0x287681fffe6b415f/set' with data '{"state":"OFF"}'
error 2023-11-16 20:22:30: No converter available for 'state' ("OFF")
debug 2023-11-16 20:22:31: Received MQTT message on 'zigbee2mqtt_upstairs/0x287681fffe6b415f/set' with data '{"state":"ON"}'
error 2023-11-16 20:22:31: No converter available for 'state' ("ON")
debug 2023-11-16 20:22:31: Received MQTT message on 'zigbee2mqtt_upstairs/0x287681fffe6b415f/get' with data '{"brightness":""}'
error 2023-11-16 20:22:31: No converter available for 'brightness' ("")
debug 2023-11-16 20:22:32: Received MQTT message on 'zigbee2mqtt_upstairs/0x287681fffe6b415f/set' with data '{"brightness":81}'
error 2023-11-16 20:22:32: No converter available for 'brightness' (81)
```

Koenkk, 2023-11-16:
Remove this from your ext converter:

```
    fromZigbee: [fz.brightness, fz.color_colortemp], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.light_brightness()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
```

FakeTruth, 2023-11-16:
Ah yes, now it works! 🥳
[image]

But I cannot set the color yet.

I paired an Ikea Styrbar with it in an attempt to coax some more information out of the LED strip, but it doesn't seem to report its state when turned on/off or on color change.

Fleshi1981, 2023-11-17:
@FakeTruth, can you send me your latest ext converter file as an example? It will help me a lot 🙂

FakeTruth, 2023-11-17:
> @FakeTruth, can you send me your latest ext converter file as an example? It will help me a lot 🙂

Sure here it is, I renamed the model to L2112 as that's what I found on the LED controller:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ORMANAS LED Strip'],
    model: 'L2112',
    vendor: 'IKEA',
    description: 'ORMANAS LED Strip',
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [250, 454]}),
};
```

I also created a PR to add support out of the box: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6500

FakeTruth, 2023-11-18:
So I got basic functionality working, I can set the state of the LED strip. How can I get it to report its state when it changes? I assume it has something to do with 'reporting' as I added an entry on the reporting tab of the device to report `onOff` attribute with an interval of 1. This seems to update the state immediately when I turn it on/off with a remote, but I don't know if this is the best way to do it:
[image]

Fleshi1981, 2023-11-18:
@FakeTruth, sorry to bother you again. I can now control the LED strip in Home Assistant, thanks for that! But could you share your final ext converter file? Because I can't get it to work to be able to choose from any color.

FakeTruth, 2023-11-19:
> @FakeTruth, sorry to bother you again. I can now control the LED strip in Home Assistant, thanks for that! But could you share your final ext converter file? Because I can't get it to work to be able to choose from any color I can only choose between different white colours.

Yes no problem, here it is again: `/usr/share/zigbee2mqtt# cat data/ormanas.js`
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ORMANAS LED Strip'],
    model: 'L2112',
    vendor: 'IKEA',
    description: 'ORMANAS LED Strip',
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [250, 454]}),
};

module.exports = definition;
```

closed 2024-06-17 by github-actions[bot]