Koenkk/zigbee2mqtt #19726: [New device support]: Hue White & Color Ambiance Surimu square panel 30x30 929003598001

gerritjonker, 2023-11-15:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-surimu--vierkant--paneellamp/8720169159136

### Database entry

{"id":39,"type":"Router","ieeeAddr":"0x001788010d2129ab","nwkAddr":19070,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003598001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1700073958362,"defaultSendRequestWhen":"immediate"}

### Comments

I'm not familiar with Github pull requests. But this should also work. 
All the necessary files are attached. Light is the same as 929002966401 only smaller (30x30). Maybe change the description of 929002966401 to Hue White & Color Ambiance Surimu square panel 60x60?
[929003598001.md](https://github.com/Koenkk/zigbee2mqtt/files/13368853/929003598001.md)
[image]

### External converter

```shell
const philips = require('zigbee-herdsman-converters/lib/philips');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters//lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['929003598001'], 
    model: '929003598001', 
    vendor: 'Signify Netherlands B.V.', 
    description: 'Hue White & Color Ambiance Surimu square panel 30x30', 
    extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
};

module.exports = definition;
```

Koenkk, 2023-11-16:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gerritjonker, 2023-11-16:
Thanks Koen!

closed 2023-11-16 by Koenkk

Shobba, 2023-12-11:
Hi there!
I just added this and I cannot adjust the color, only the white temperature. The device def can support this.

gerritjonker, 2023-12-11:
No problems here. After the latest release I have removed the test files and everything is working as expected.

Shobba, 2023-12-11:
@gerritjonker can you adjust the color (blue/red etc)?

gerritjonker, 2023-12-11:
I can adjust the color, the brightness and even the special effect are working

Shobba, 2023-12-11:
Weird. Mine looks like this, so no color selection:
[image]

gerritjonker, 2023-12-11:
Mine looks exactly the same. Indeed no color selector. I tested only the predefined four colours and assumed it was working.

Shobba, 2023-12-11:
Okay, alright :) My PR got merged already, should be fixed in some upcoming update!