Koenkk/zigbee2mqtt #19731: [New device support]: _TZE200_vvmbj46n (Tuya Temperature and Humidity - TH05Z)

Jokanaan78, 2023-11-16:
### Link

https://fr.aliexpress.com/item/1005005923865974.html?gatewayAdapt=glo2fra

### Database entry

{"id":67,"type":"EndDevice","ieeeAddr":"0xa4c138a978038dbe","nwkAddr":21189,"manufId":4417,"manufName":"_TZE200_vvmbj46n","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"l  ,\u0011","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_vvmbj46n","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1700109974399,"defaultSendRequestWhen":"immediate"}

### Comments

I can make this work with the external converter (see below) but would rather have it checked by the pros and integrated into the platform

[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_vvmbj46n',
        },
    ],
    model: 'TH05Z',
    vendor: 'TuYa',
    description: 'Tuya Zigbee Temperature & Humidity Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.humidity(), e.battery(), tuya.exposes.batteryState(), tuya.exposes.temperatureUnit()],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
	[1, 'temperature', tuya.valueConverter.divideBy10],
	[2, 'humidity', tuya.valueConverter.raw],
	[3, 'battery_state', tuya.valueConverter.batteryState],
	[9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],

        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-11-16:
Looks good, could you make a PR?

sweepy-77, 2023-11-16:
I have unrecognized values with your code

`info 2023-11-16 18:50:56MQTT publish: topic 'zigbee2mqtt/0xa4c1384eb0d22c4f', payload '{"battery":null,"battery_state":null,"humidity":39,"linkquality":134,"temperature":26.3,"temperature_unit":"celsius"}'
debug 2023-11-16 18:50:56Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,134],"type":"Buffer"},"datatype":2,"dp":10}],"seq":63232}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:56Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":11}],"seq":63488}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:56Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,60],"type":"Buffer"},"datatype":2,"dp":12}],"seq":63744}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:56Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,20],"type":"Buffer"},"datatype":2,"dp":13}],"seq":64000}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:57Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,120],"type":"Buffer"},"datatype":2,"dp":17}],"seq":64256}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:57Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,120],"type":"Buffer"},"datatype":2,"dp":18}],"seq":64512}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:57Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,6],"type":"Buffer"},"datatype":2,"dp":19}],"seq":64768}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:57Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,6],"type":"Buffer"},"datatype":2,"dp":20}],"seq":65024}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:57Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":255}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:57Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":1}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:58Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":3}' from endpoint 1 with groupID 0
debug 2023-11-16 18:50:59Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":7}' from endpoint 1 with groupID 0
debug 2023-11-16 18:51:01Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":12}' from endpoint 1 with groupID 0
debug 2023-11-16 18:51:03Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":17}' from endpoint 1 with groupID 0
debug 2023-11-16 18:51:05Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":22}' from endpoint 1 with groupID 0
debug 2023-11-16 18:51:21Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9}],"seq":6656}' from endpoint 1 with groupID 0`

On the other hand with this code which comes from here  [(https://github.com/Koenkk/zigbee2mqtt/discussions/18827)]
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const definition = {
  fingerprint: [{
    modelID: 'TS0601',
    manufacturerName: '_TZE200_vvmbj46n'
  }],
  model: 'TSJS0601',
  vendor: 'TuYa',
  description: 'Temperature & humidity sensor with clock',
  fromZigbee: [legacy.fromZigbee.nous_lcd_temperature_humidity_sensor, fz.ignore_tuya_set_time],
  toZigbee: [legacy.toZigbee.nous_lcd_temperature_humidity_sensor],
  onEvent: tuya.onEventSetLocalTime,
  configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
  },
  exposes: [
    e.temperature(), e.humidity(), e.battery(),
[... cut ...]
debug 2023-11-16 19:02:41Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":30}' from endpoint 1 with groupID 0
debug 2023-11-16 19:02:43Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":35}' from endpoint 1 with groupID 0
debug 2023-11-16 19:02:45Received Zigbee message from '0xa4c1384eb0d22c4f', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":40}' from endpoint 1 with groupID 0`

oallain, 2023-11-20:
Hello,

I would love to see this model integrated with Zigbee2mqtt ❤️

Koenkk, 2023-11-21:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/78ac831ec1143edc92e8231eed73ebd6

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

oallain, 2023-11-25:
Hello @Koenkk 

I try it, and it dosn't work, my z2m don't start, so i comment line in my configuration.yaml.
I have : 
* create file ext_converter.js 
* set content form https://gist.githubusercontent.com/Koenkk/78ac831ec1143edc92e8231eed73ebd6/raw/5549c7abeebb1e79cf129eb2680ff8fabbf68e65/ext.js
* add in my configuration.yaml
```
external_converters:
  - ext_converter.js
```
* restart z2m

Try to start failled :(

sweepy-77, 2023-11-25:
try below

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_yjjdcqsq', '_TZE200_9yapgbuv', '_TZE204_upagmta9', '_TZE200_cirvgep4', '_TZE200_upagmta9', '_TZE200_vvmbj46n']),
    model: 'TS0601_temperature_humidity_sensor_2',
    vendor: 'TuYa',
    description: 'Temperature and humidity sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({queryOnDeviceAnnounce: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        // Required to get the device to start reporting
        await device.getEndpoint(1).command('manuSpecificTuya', 'dataQuery', {});
    },
    exposes: [e.temperature(), e.humidity(), e.battery(), tuya.exposes.temperatureUnit(),],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],
            [10, 'alarm_temperature_max', tuya.valueConverter.divideBy10],
            [11, 'alarm_temperature_min', tuya.valueConverter.divideBy10],
            [12, 'alarm_humidity_max', tuya.valueConverter.raw],
            [13, 'alarm_humidity_max', tuya.valueConverter.raw],
            [14, 'alarm_temp', tuya.valueConverter.trueFalse1],
            [15, 'alarm_humidity', tuya.valueConverter.trueFalse1],
            [17, 'interval_report_temp', tuya.valueConverter.raw],
            [18, 'interval_report_humidity', tuya.valueConverter.raw],
            [19, 'sensitivity_temp', tuya.valueConverter.raw],
            [20, 'sensitivity_humidity', tuya.valueConverter.raw],
        ],
    },
    whiteLabel: [
        tuya.whitelabel('TuYa', 'ZTH01', 'Temperature and humidity sensor', ['_TZE200_yjjdcqsq']),
        tuya.whitelabel('TuYa', 'ZTH02', 'Temperature and humidity sensor', ['_TZE200_9yapgbuv']),
        tuya.whitelabel('TuYa', 'ZTH05', 'Temperature and humidity sensor', ['_TZE204_upagmta9', '_TZE200_upagmta9']),
        tuya.whitelabel('TuYa', 'ZTH05Z', 'Temperature and humidity sensor', ['_TZE200_vvmbj46n']),
        tuya.whitelabel('TuYa', 'ZTH08-E', 'Temperature and humidity sensor', ['_TZE200_cirvgep4']),
    ],
};

module.exports = definition;`

oallain, 2023-11-26:
Hello @Koenkk and @sweepy-77 

Thanks for your help.
I've seen that z2m's built-in editor displays errors.
On both versions of your script, I get the following error:
```
Request 'zigbee2mqtt/bridge/request/extension/save' failed with error: 'tuya.onEvent is not a function'
```

sweepy-77, 2023-11-26:
maybe a story of indentation
look here:

[https://gist.github.com/sweepy-77/772f25eb2efca35689a5309377d0da93#file-zth05z-js](url)

oallain, 2023-11-26:
I have a new error : 

`Request 'zigbee2mqtt/bridge/request/extension/save' failed with error: 'ConstructorClass is not a constructor'`

saxmanio85, 2023-11-28:
This is working for me apart from the time sent to the device is one hour out. Is there something I can do to correct the time?

pubFbx, 2023-11-29:
Hello, I wrongly thought it was complicated to add an "unsupported device", but in the end when the ".js" is available, it's very simple.  Thank you very much for this work.
One question: is it complicated / possible to replace the "Z" icon displayed in Z2M for these devices?

saxmanio85, 2023-11-29:
So I've just tried the following (from here, I didnt write this: https://github.com/Koenkk/zigbee2mqtt/discussions/18827#discussioncomment-7150347 )

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');

const definition = {
  fingerprint: [{
    modelID: 'TS0601',
    manufacturerName: '_TZE200_vvmbj46n'
  }],
  model: 'JM-TRH-ZGB-V1',
  vendor: 'TuYa',
  description: 'Temperature & humidity sensor with clock',
  fromZigbee: [legacy.fromZigbee.nous_lcd_temperature_humidity_sensor, fz.ignore_tuya_set_time],
  toZigbee: [legacy.toZigbee.nous_lcd_temperature_humidity_sensor],
  onEvent: tuya.onEventSetLocalTime,
  configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
  },
  exposes: [
    e.temperature(), e.humidity(), e.battery(),
    e.numeric('temperature_report_interval', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(60)
    .withValueStep(1)
    .withDescription('Temperature Report interval'),
    e.numeric('humidity_report_interval', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(60)
    .withValueStep(1)
    .withDescription('Humidity Report interval'),
    e.enum('temperature_unit_convert', ea.STATE_SET, ['celsius', 'fahrenheit']).withDescription(
      'Current display unit'),
    e.enum('temperature_alarm', ea.STATE, ['canceled', 'lower_alarm', 'upper_alarm'])
    .withDescription('Temperature alarm status'),
    e.numeric('max_temperature', ea.STATE_SET).withUnit('°C').withValueMin(-20).withValueMax(60)
    .withDescription('Alarm temperature max'),
    e.numeric('min_temperature', ea.STATE_SET).withUnit('°C').withValueMin(-20).withValueMax(60)
    .withDescription('Alarm temperature min'),
    e.enum('humidity_alarm', ea.STATE, ['canceled', 'lower_alarm', 'upper_alarm'])
    .withDescription('Humidity alarm status'),
    e.numeric('max_humidity', ea.STATE_SET).withUnit('%').withValueMin(0).withValueMax(100)
    .withDescription('Alarm humidity max'),
    e.numeric('min_humidity', ea.STATE_SET).withUnit('%').withValueMin(0).withValueMax(100)
    .withDescription('Alarm humidity min'),
  ],
};

module.exports = definition;
```

oallain, 2023-12-01:
Many thanks !
It's work :) 

```json
{
    "battery": 39,
    "humidity": 63,
    "humidity_alarm": "upper_alarm",
    "humidity_report_interval": 120,
    "humidity_sensitivity": 6,
    "last_seen": "2023-12-01T18:50:06+01:00",
    "linkquality": 140,
    "max_humidity": 60,
    "max_temperature": 39,
    "min_humidity": 20,
    "min_temperature": 0,
    "temperature": 22,
    "temperature_alarm": "canceled",
    "temperature_report_interval": 120,
    "temperature_sensitivity": 0.6,
    "temperature_unit_convert": "celsius"
}
```

baderas, 2023-12-25:
Can confirm that the external converter that [saxmanio85](https://github.com/saxmanio85) posted works fine for me.

sweepy-77, 2023-12-25:
Can you modify the alarms with this code.

with the code below you can modify the alarm by turning on the backlight and have alerts on the minimum and maximum

https://gist.github.com/sweepy-77/772f25eb2efca35689a5309377d0da93

grunthos503, 2023-12-30:
The code provided by @sweepy-77 at https://gist.github.com/sweepy-77/772f25eb2efca35689a5309377d0da93 is working for me.

Koenkk, 2023-12-31:
@sweepy-77 could you make a PR for this?

sweepy-77, 2024-01-01:
I've never done a PR and I don't see how to do it, I don't know if I'll succeed.

Koenkk, 2024-01-02:
@sweepy-77 it can be done by clicking the pencil/edit icon [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts)

kboeckx, 2024-01-03:
I succesfully added the device: Zigbee model 'TS0601' and manufacturer name '_TZE200_vvmbj46n'
But I'm not able to set the "temperature_report_interval" and "humidity_report_interval". In the log I see 

`Received MQTT message on 'zigbee2mqtt/0xa4c13885784ea0aa/set' with data '{"temperature_report_interval":10}'`

and

`Publishing 'set' 'temperature_report_interval' to '0xa4c13885784ea0aa'`

But nothing is changed on the device itself.
What need to be done so I can change the report interval ?

sweepy-77, 2024-01-03:
There is a button on the top of the device for lighting. Once turned on I send the order for the change

kboeckx, 2024-01-04:
> There is a button on the top of the device for lighting. Once turned on I send the order for the change

That did the trick ! Thx

sweepy-77, 2024-01-07:
@Koenkk I followed your link and created the PR but I don't see it. 
I also saw that the ZTH05Z was added with version 1.35.0 with the model JM-TRH-ZGB-V1 [#6705](https://github.com/Koenkk/zigbee-herdsman-converters/issues/6705) but the commands to change the alarms and the interval report do not work.

Koenkk, 2024-01-08:
@sweepy-77 I didn't see a PR indeed, there are some more docs [here](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request)

lukast008, 2024-01-18:
I have an issue with this device. I have 4 of them and from time to time they randomly stop refreshing the temperature and humidity. When I try to reconfigure it in Z2M I receive following logs:
```
Debug 2024-01-18 09:52:05Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Termometr salon","transaction":"pxb8p-1"}'
Info 2024-01-18 09:52:05Configuring 'Termometr salon'
Error 2024-01-18 09:53:06Failed to configure 'Termometr salon', attempt 3 (Error: Bind 0xa4c138229d9cae94/1 genBasic from '0x00212effff09a2bd/1' failed (waiting for response TIMEOUT) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:776:19) at runNextTicks (node:internal/process/task_queues:60:5) at processTimers (node:internal/timers:509:9) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:545:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:5013:13) at Configure.configure (/app/lib/extension/configure.ts:120:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))
Info 2024-01-18 09:53:06MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Termometr salon"},"error":"Failed to configure (Bind 0xa4c138229d9cae94/1 genBasic from '0x00212effff09a2bd/1' failed (waiting for response TIMEOUT))","status":"error","transaction":"pxb8p-1"}'
```

For another currently working device the same action in logs looks like:
```
Debug 2024-01-18 09:54:45Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Termometr gabinet","transaction":"pxb8p-2"}'
Info 2024-01-18 09:54:45Configuring 'Termometr gabinet'
Info 2024-01-18 09:54:45Successfully configured 'Termometr gabinet'
Info 2024-01-18 09:54:45MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Termometr gabinet"},"status":"ok","transaction":"pxb8p-2"}'
```

CBDesignS, 2024-01-19:
I have this fresh off the boat from china and it was discoverd/interviewed and setup somewhat correctly apart from there was no control over the reporting times. I then used the convertor posted by sweepy-77 and it works better than the auto interviewed setup but both versions show a battery % reporting over the top by about 100. so brand new batteries gave 200% and now a week later it is down to 195%. Any ideas how to correct or convert the value to the correct one ?  
Please bear in mind this is my first venture into zigbee2mqtt and I am a very new newbie.

sorryusernameisalreadytaken, 2024-03-11:
More than a month after the pr and release. What is the current state, does work well or are there any issues?

robertp007, 2024-04-22:
I now also have this thermometer. Unfortunately, the temperature sensitivity and the temperature period report cannot be comprehensibly adjusted. I need an accuracy of 0.1 degrees and a message when the temperature changes by +- 0.1 degrees. How can I set these values?

CBDesignS, 2024-04-24:
I removed the custom convertor and reverted back to letting z2m find the device, it does and now and the device reports battery at 116% on a fresh set of 3 x AAA cells and once they discharge down the unit stops sending data but shows online at 34% charge. 
using the custom convertor The image shown in z2m used to show the correct picture of the device (at the top in the first post on this thread) but now using the discovered device it shows the generic picture with basic black & white Lcd display. 
Does the old Lcd version only have 2 cells to power it and the newer have 3 giving the incorrect battery data ?

robertp007, 2024-05-02:
I have had a similar experience too. Only when I use the custom sensor description from sweepy-77 does the setting of temperature sensitivity and interval work moderatly.
However, the temperature sensitivity does not allow a setting lower than 3. According to the enclosed user sheet, values below 1.0 should also be possible here.
The automatic sensor integration via Z2M is not yet error-free and really usable.

closed 2024-11-29 by github-actions[bot]

b4rRa, 2025-01-08:
Got 2 of them (_TZE200_vvmbj46n) Looks like they are still not implemented in zigbee2mqtt v2

Showing wrong device icon:
[image]

and no options for reporting.

wesleyRaposo, 2025-01-08:
I just installed the device and several sensors are showing errors.

[image]

[image]

[image]

[image]

wesleyRaposo, 2025-01-13:
> @Koenkk Same us guys above, the device appears as another device and all the settings are wrong. Also on 2.0.0.

I had created an automation that solved the problem.
Every time the reporting entities had a value greater than 100, the automation automatically reduced it to 60 (the value I set as the default).
The problem I identified was that whoever programmed the device recognition in Zigbee2QTT made a mistake in identifying the properties "Temperature periodic report" and "Humidity periodic report".
According to my device's manual, these properties are to be defined in "minutes" and come out with the factory default value of "120".
But the programmer set these properties to "percentage" and defined the range of values ​​between 0 and 100. This is the root of the problem.
When I installed the device, several problems occurred because I set these parameters to 50 or 60 but, a few minutes later, they returned to 120 by themselves. That's when I created the automation and the problem was solved.
But a few days later I downloaded the Zigbee2MQTT 2.0.0-2 update. They did a major overhaul and these parameters no longer have corresponding sensors and the automation stopped working.
On the other hand, the value set in them is now stable (it is not changing on its own). So, in a way, even though the parameters have the wrong type (it was supposed to be minutes and allow values ​​greater than 100), the error stopped happening.

[image]