Koenkk/zigbee2mqtt #19732: [Feature request]: set friendly_name without editing configuration.yaml

pohly, 2023-11-16:
### Is your feature request related to a problem? Please describe

My current workflow when adding a new device is:
- enable all to join
- wait for the interview to complete and the new device appearing in the configuration.yaml
- overwrite the friendly_name value
- log into the machine where zigbee2mqtt is running
- restart it

### Describe the solution you'd like

It would be simpler if the web interface allowed to modify the "friendly name" under "settings". Currently (= [1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2)) the field is shown, it just cannot be changed.

### Describe alternatives you've considered

A "quit" or "restart" button in the web frontend would also help a bit.

### Additional context

I run zigbee2mqtt as a systemd service. If it quits, it will get restarted automatically.

AntaresFR, 2023-11-16:
You can change it in the Web GUI in the device list, the blue icon on the right side in the line of the device.
You can also restart Z2M from the Webgui. If restart is requiered by a setting change, a button will appear on top right corner. if you want to manually restart and no button, you can do it in "Settings > Tools".

pohly, 2023-11-16:
How did I manage to miss that?! :hot_face: 

Thanks for pointing it out.

closed 2023-11-16 by pohly